raspberrypi/rpicam-apps #274: [BUG]  meson's dependency() fails to find installed Gstreamer files

jlpoolen, 2022-03-16:
I have a 64 bit Gentoo Linux, Linux GenPi64 5.10.92-v8-p4 #1 SMP PREEMPT Thu Mar 10 12:18:30 UTC 2022 aarch64 GNU/Linux, running on a Raspberry Pi 4.  (The image also runs on the Raspberry Pi Zero 2 W, but I'm developer on the more powerful RPi4.)
I have compiled and installed a recent. within the last two weeks, high watermark of gstreamer v. 20.

Here is a screenshot demonstrating gstreamer installed and runs.
[image]

When I build libcamera, the meson dependency check fails to find both gstreamer.pc files and consequently libcamera's Gstreamer is disabled.  I am trying to run gstreamer using libcamera and believe I need the Gstreamer support.

[libcamera_meson_reconfigure.txt](https://github.com/raspberrypi/libcamera-apps/files/8259077/libcamera_meson_reconfigure.txt)

Specifically, the points of failure are at meson's dependency function:

    GenPi64 /usr/local/src/libcamera # head src/gstreamer/meson.build
    # SPDX-License-Identifier: CC0-1.0

    glib_dep = dependency('glib-2.0', required : get_option('gstreamer'))

    gst_dep_version = '>=1.14.0'
    gstvideo_dep = dependency('gstreamer-video-1.0', version : gst_dep_version,
                              required : get_option('gstreamer'))
    gstallocator_dep = dependency('gstreamer-allocators-1.0', version : gst_dep_version,
                                  required : get_option('gstreamer'))

    GenPi64 /usr/local/src/libcamera #

Line 11 of src/gstreamer/meson.build calls the found() method on both gstvideo_dep and gstallocator_dep objects and both return false when they should return true.

So, the problem is that pkconfig is not finding the .pc files.

**The gstreamer layout on the Gentoo system**

The pkconfig files are present in /usr/local/lib64/pkgconfig:

    GenPi64 /home/jlpoole/mesontut1 # ls -la /usr/local/lib64/pkgconfig/gstreamer-allocators-1.0.pc
    -rw-r--r-- 1 root root 552 Mar 13 01:08 /usr/local/lib64/pkgconfig/gstreamer-allocators-1.0.pc
    GenPi64 /home/jlpoole/mesontut1 # ls -la /usr/local/lib64/pkgconfig/gstreamer-video-1.0.pc
    -rw-r--r-- 1 root root 612 Mar 13 01:08 /usr/local/lib64/pkgconfig/gstreamer-video-1.0.pc
    GenPi64 /home/jlpoole/mesontut1 #

The compiled run-time executables are in /usr/local/bin:

    GenPi64 /usr/local/src/libcamera # ls -la /usr/local/bin
    total 3132
    drwxr-xr-x 1 root root    938 Mar 13 01:16 .
    drwxr-xr-x 1 root root     74 Mar 13 01:06 ..
    -rwxr-xr-x 1 root root  44976 Mar 13 01:15 fribidi
    -rwxr-xr-x 1 root root 241280 Mar 13 01:16 ges-launch-1.0
    -rwxr-xr-x 1 root root  53048 Mar 13 01:15 gst-device-monitor-1.0
    -rwxr-xr-x 1 root root  80008 Mar 13 01:15 gst-discoverer-1.0
    -rwxr-xr-x 1 root root 183312 Mar 13 01:15 gst-inspect-1.0
    -rwxr-xr-x 1 root root 100248 Mar 13 01:15 gst-launch-1.0
    -rwxr-xr-x 1 root root 135584 Mar 13 01:15 gst-play-1.0
    -rwxr-xr-x 1 root root  95096 Mar 13 01:15 gst-stats-1.0
    -rwxr-xr-x 1 root root  30072 Mar 11 10:24 gst-tester-1.0
    -rwxr-xr-x 1 root root  76256 Mar 13 01:16 gst-transcoder-1.0
    -rwxr-xr-x 1 root root  39464 Mar 13 01:15 gst-typefind-1.0
    -rwxr-xr-x 1 root root  75464 Mar 13 01:16 gst-validate-1.0
    -rwxr-xr-x 1 root root 142936 Mar 13 01:16 gst-validate-images-check-1.0
    -rwxr-xr-x 1 root root   1949 Mar 11 10:19 gst-validate-launcher
    -rwxr-xr-x 1 root root  29632 Mar 13 01:16 gst-validate-media-check-1.0
    -rwxr-xr-x 1 root root  27072 Mar 13 01:16 gst-validate-rtsp-server-1.0
    -rwxr-xr-x 1 root root  53672 Mar 13 01:16 gst-validate-transcoding-1.0
    -rwxr-xr-x 1 root root 169552 Mar 13 01:15 hb-ot-shape-closure
    -rwxr-xr-x 1 root root 193736 Mar 13 01:15 hb-shape
    -rwxr-xr-x 1 root root 201176 Mar 13 01:15 hb-subset
    -rwxr-xr-x 1 root root 323568 Mar 13 01:15 hb-view
    -rwxr-xr-x 1 root root  34048 Mar 13 01:16 json-glib-format
    -rwxr-xr-x 1 root root  24464 Mar 13 01:16 json-glib-validate
    -rw-r--r-- 1 root root      0 Jan 17 00:52 .keep
    -rwxr-xr-x 1 root root  56936 Mar 13 01:15 orc-bugreport
    -rwxr-xr-x 1 root root 152488 Mar 13 01:15 orcc
    -rwxr-xr-x 1 root root  32264 Mar 13 01:15 pango-list
    -rwxr-xr-x 1 root root  30096 Mar 13 01:15 pango-segmentation
    -rwxr-xr-x 1 root root 149920 Mar 13 01:15 pango-view
    -rwxr-xr-x 1 root root  98152 Mar 13 01:16 playout
    -rwxr-xr-x 1 root root 150640 Mar 12 06:35 stunbdc
    -rwxr-xr-x 1 root root 126240 Mar 12 06:35 stund
    GenPi64 /usr/local/src/libcamera #

In case one wonders about "Run-time dependency", I tried this variant and got the same result: failed to find gstreamer:

    GenPi64 /usr/local/src/libcamera # export PKG_CONFIG_PATH=/usr/local/bin
    GenPi64 /usr/local/src/libcamera # date; echo $PKG_CONFIG_PATH; meson --reconfigure build
    Wed Mar 16 05:01:11 AM UTC 2022
    /usr/local/bin
    ...
        GStreamer support        : NO

To simulate the problem in this environment, I created a simple meson project of "hello world" in c.  I then created a simple meson file and added a dependency of "gstreamer-1.0":

    GenPi64 /usr/local/src/libcamera # cat /home/jlpoole/mesontut1/meson.build
    project('tutorial', 'c')
    #
    # add a dependency
    #
    gstreamdep = dependency('gstreamer-1.0')
    executable('demo', 'main.c')
    GenPi64 /usr/local/src/libcamera #

The project failed.  So I set my environment to point to the location where Gentoo has the configuration files: 

    GenPi64 /home/jlpoole/mesontut1 # echo $PKG_CONFIG_PATH
    /usr/local/lib64/pkgconfig
    GenPi64 /home/jlpoole/mesontut1 #
    
Then I ran "meson build" on my test tutorial, and this time my sample project successfully was built.  Thinking that $PKG_CONFIG_PATH might solve the problem, I returned to libcamera and tried to build again and unfortunately the meson functions appear not to be finding gstreamer.

I realize this may be a bug in meson, but since libcamera relies on meson to find packages, I'm raising it here.  I also me be missing something very fundamental about libcamera's implementation of meson.

naushir, 2022-03-16:
Unfortunately this is not something we would be able to help with directly.  You probably want to ping this to either Gentoo Distribution maintainers or possibly the libcamera development list at https://lists.libcamera.org/listinfo.

jlpoolen, 2022-03-16:
Can you confirm that gstvideo_dep.found() returns true on your build?

kbingham, 2022-03-16:
I would expect this to be a configuration issue in gentoo here. We wouldn't put in specific overrides for libcamera - so this should be solved by either meson or gentoo.

I can confirm I can build with gstreamer support on a bullseye install on RPi.

kbingham, 2022-03-16:
Can you confirm how you set your PKG_CONFIG_PATH when reconfiguring meson?
I believe distros have ways to add extra search paths to the default pkg-config path - but that will be gentoo specific

closed 2022-03-16 by jlpoolen