raspberrypi/rpicam-apps #276: libcamera-still hangs with timelapse output to STDOUT on 64 bit RPi OS

knutegit, 2022-03-16:
RPi OS 64 bit, Pi 4B 2GB

libcamera-still --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80 --output -

[1:12:00.364873573] [19737] ERROR V4L2 v4l2_videodevice.cpp:1179 /dev/video15[14:cap]: Unable to request 1 buffers: Cannot allocate memory
[1:12:00.364972031] [19737] ERROR RPI raspberrypi.cpp:986 Failed to allocate buffers

Problem exists in files from repository and in git files compiled on the Pi.

Is the libcamera-apps-timelapse-fix.zip file appropriate for 64 bit?  I tried it with and without, neither worked.

naushir, 2022-03-16:
Can you run your command through the ``camera-bug-report`` tool and attach the bug report file to this issue please?

knutegit, 2022-03-17:
Sorry to spin you up on this, it's not reproducible today.  I downloaded and recompiled libcamera, libepoxy and libcamera-apps today.  libcamera-still ran with no problems.  I deleted all the files in /usr/local/bin and still no problems.  I made no changes to the Pi since yesterday so I don't know what caused it to hang then and not now.

closed 2022-03-17 by knutegit