raspberrypi/rpicam-apps #277: [BUG] Jpeg output artifacts on every second image with OV5647

Sharsie, 2022-03-17:
**Describe the bug**
I am getting weird artifacts at the top of every second photo with jpeg output. Raw output seems to be unaffected. 

[libcamera-still output](https://imgur.com/a/rJ506Kz)
commands are run without any arguments (except -o), one after another. It does not really matter what is the delay between the commands, even if I run the second command 15 minutes later, it produces the same artifacts.

[raspistill output](https://imgur.com/a/apiZW9Y)
Raspistill produces images without artifacts (although green tinted) - again without any args

To better demonstrate this really happens at every second photo, here is the result of raw+jpeg output
[for index in {1..14}; do libcamera-still -r -o /mnt/fake-storage/mount-point/one-off/libcamera-still-$index.jpg; done](https://imgur.com/chpcp8d)

**Bug report**
[camera-bug-report -o ... -c "libcamera-still -o ..."](https://gist.github.com/Sharsie/2b08c901e56762a7cdef3638056bcb5a)

**Additional context**
Happens with OV5647:
- Waveshare IR-CUT kamera (B)
- Waveshare RPi Zero V1.3 kamera

Does not happen with Sony IMX219 CCD
- Raspberry Pi NoIR kamera V2

naushir, 2022-03-18:
Thanks for the report.  This seems like a strange bug.  I've tried your command on my Pi 4 + ov5647 setup, and cannot reproduce this.  I will try to get my hands on a Zero W board to run this on to exactly match your setup.  In the meantime, could you try the following if you are able to?

1) Capture an uncompressed YUV output file instead of JPEG: ``` libcamera-still -o test.yuv -r -e yuv420```.  Please attach both the RAW and YUV files.
2) Can you run on a Pi 3B+/4 board if you have one available and see if you get the same result?
3) If you reboot between runs, do you still get the corruption on every other image?

Sharsie, 2022-03-18:
Thanks for the hints
1. here are the yuv + raw images taken one after another (I can't open it atm, so not sure if yuv contains the artifacts)
https://github.com/Sharsie/libcamera-apps-bug-report/commit/0aa169f2602e106d7216fc7c0606f3aab3d639b5
2. I'll try to get my hands on one of these, will report later
3. First image after reboot always contains artifacts: https://github.com/Sharsie/libcamera-apps-bug-report/commit/9333e131ee5539ae339d4236b397480b28fdb822

naushir, 2022-03-18:
Thanks for the update!  The YUV images do have the corruption as well, so it's nothing to do with the image encoding.

I've just run this on a Zero W board and I cannot reproduce this here.  It's also unlikely that the sensor is faulty as your RAW files are ok.  I'm leaning towards a platform issue, so if you could try on another platform that would be very useful.

naushir, 2022-03-29:
@Sharsie any updates on this one?

Sharsie, 2022-03-29:
I brough one of the cameras home, but unfortunatelly it's the one with [baked in mini CSI connector](https://www.waveshare.com/rpi-zero-v1.3-camera.htm) and I don't have the adapter at hand for rpi 3/4
So I'll pickup the other one this week, will report back once I have it

Sharsie, 2022-03-31:
@naushir Tested on raspberry pi 3 A+, no artifacts on the images. The OS is exactly the same as I pulled the sd card from Zero W and tried the exact same setup on 3A+.

naushir, 2022-04-01:
If you used the same sensors in the 3A+, it does suggest a possible hardware issue with your Zero W board.

Sharsie, 2022-04-01:
Yea, that sounds like my luck, I bought two zero Ws and two OV5647, zero Ws produce artifacts and OV5647 are affected by #271 
It's probably a HW fault with the combination of the two, as it is produced on both boards and with both cameras.
Thanks for trying to help though, into the bin they go! :)

closed 2022-04-01 by Sharsie

naushir, 2022-04-01:
Before you throw the boards in the bin, would you be able to try a few things for me to see if they help your situation?

- Try a different power supply connected to the Zero W.
- add ``force_turbo=1`` to your ``/boot/config.txt`` file.

Let me know the results if you are able to try those things.

Sharsie, 2022-04-01:
I tried 3 different PSUs, no change, `force_turbo=1` yields no change either, I can also try `force_turbo=1` on different PSUs if that helps during the weekend

folkertvanheusden, 2022-04-06:
I have this problem as well. On a raspberry pi 3+ and on a zero w. Both have a version 1.3 camera.
See https://forums.raspberrypi.com/viewtopic.php?p=1987929#p1987929

So 4 cameras tested, 2 rpi's.