raspberrypi/rpicam-apps #278: [BUG] Zero2W+HQ Cam+Bullseye OS Lite: libcamera-hello stalls

jgrmpf, 2022-03-22:
**Describe the bug**
Using Zero2W with latest Raspi OS 64bit OS Lite (March 21, 2022) and a HQ Camera attached lets "libcamera-hello" executed on the console just stall with no preview being displayed. A "libcamera-jpeg -o test.jpg" also never stops and does not store a picture.

Output on console:
```
pi@raspberrypi:~ $ libcamera-hello 
Made DRM preview window
[0:04:07.421520495] [779]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e
[0:04:07.476413584] [781]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:04:07.476519729] [781]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:04:07.476585458] [781] ERROR CameraSensor camera_sensor.cpp:551 'imx477 10-001a': Camera sensor does not support test pattern modes.
[0:04:07.537481541] [781]  INFO RPI raspberrypi.cpp:1317 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:04:07.539076222] [779]  INFO Camera camera.cpp:1028 configuring streams: (0) 2028x1520-YUV420
[0:04:07.539632782] [781]  INFO RPI raspberrypi.cpp:747 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:04:07.614005315] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: Saturation = 1.000000
[0:04:07.616121347] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: Contrast = 1.000000
[0:04:07.616178118] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: Brightness = 0.000000
[0:04:07.616220305] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: AwbMode = 0
[0:04:07.616268638] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: Sharpness = 1.000000
[0:04:07.616311815] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: ExposureValue = 0.000000
[0:04:07.616381398] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: AeExposureMode = 0
[0:04:07.616419159] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:04:07.616466346] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: AeMeteringMode = 0
[0:04:07.616503012] [785]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: NoiseReductionMode = 3
```

**Bug report**
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/8323141/bug.txt)

**Additional context**
HW is Raspberry Pi Zero 2 W.
OS is Raspberry Pi OS 64bit Lite installed by Imager on SD card.
Camera is HQ Camera with 15 cm ribbon cable.
libcamera package is libcampera-apps-lite shipped with the Raspi OS Lite.
uname -a: ```Linux raspberrypi 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux```

naushir, 2022-03-22:
@jgrmpf thanks for the report.  Would you be able to provide the console output with the following command please:
```
LIBCAMERA_LOG_LEVELS=*:0 libcamera-hello
```

jgrmpf, 2022-03-22:
Sure. Please find it here:
[libcam2.log](https://github.com/raspberrypi/libcamera-apps/files/8324228/libcam2.log)

kbingham, 2022-03-22:
I think it's this one if you want to try it out locally by applying the patches: 
  - https://patchwork.libcamera.org/project/libcamera/list/?series=2983

jgrmpf, 2022-03-23:
Thanks for the quick response to the issue! I re-checked the ribbon cable connection, but no change to the behavior. Unfortunately I don't have another camera at hand. I will try to get one and test also with an RPI2.

jgrmpf, 2022-03-23:
I switched the cables and then it actually worked!
Ok, then sorry for creating a BUG report, and thanks for pointing me to the issue :grimacing:

closed 2022-03-23 by jgrmpf