raspberrypi/rpicam-apps #281: [BUG] imx219 binned 2x 8bits mode produces corrupted frames

geehalel, 2022-03-29:
**Describe the bug**
Trying to use the binned 2x mode (1640x1232) with the imx219 sensor in 8 bits mode produces frames with upper half containing parts/blocks of the original image, and bottom half made of zeroes. You can see samples in [this forum post](https://forums.raspberrypi.com/viewtopic.php?t=332103#p1987886). Use the following command to reproduce the issue:

```
libcamera-jpeg -n --mode 1640:1232:8:U --width 1640 --height 1232 -o test8bits.jpg
```

**Bug report**
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/8372761/bug.txt)

**Additional context**
Every other raw formats produce correct frames in 8 and 10 bits, particularly the 640x480 format which is also binned. I also use another application (`indi_v4l2_ccd` which reads frames from `/dev/video0`) and get the same result.
As stated in the forum post, the 640x480 format defined in `imx219.c` uses analog/special binning 2x mode (registers 0x174/5 set to 0x3), whereas the 1640x1232 format uses digital/normal binning (registers 0x174/5 set to 0x1). It seems that digital/normal binning in 8 bits is not supported in the sensor.
Thanks for your work, it helps a lot when beginning with libcamera!

davidplowman, 2022-03-30:
I've had a little play around with this. So far as I can tell, the 1640x1232 10-bit mode only works with "normal" binning, whereas the 1640x1232 8-bit only works with "special" binning. Yet the 640x480 10-bit mode does work with "special" binning. I can't find any discussion of this in the datasheet so, frankly, it's all a bit of a mystery.

The question is what we should do about this. One of the problems is that we often work with existing drivers and we have no way of testing whether the functionality that was there before still works (because we don't have the hardware used for the original development), and indeed we can't tell what was actually tested and working even before we touch the driver. The driver seems to assume that the "frame size" and "8/10-bit format" registers are orthogonal, but the evidence seems to suggest the contrary.

I guess one could fix this by breaking that assumption, and tying the binning mode registers to the output format registers, but at the moment I'm a bit in the "is this worth doing?" kind of place. Can you say anything more about your use-case, is there a specific reason to prefer an 8-bit over a 10-bit mode?

geehalel, 2022-03-30:
Thanks for having looked into this so quickly.
My first concern was to check if my setup was not faulty in some manner, You confirm me that this is not the case, and adds another point concerning 10bits binning in analog mode. I did not think about it. There are some hidden features in this sensor.
I am making an experimental [INDI]l(https://indilib.org) driver for libcamera and it exposes every formats/sizes returned by libcamera. I was simply testing every combinations and came across this one. 8bits bayer may eventually be useful while streaming for focusing, but 640x480 is enough for that purpose, or any non bayer format anyway.
It is not worth refactoring the imx219 code, I may simply disable the frame size in bayer 8 bits mode, or not use it. The point is that now we know, thanks to your investigation.
I'll close the issue if you agree, or let you do so.
Many thanks again for your effort.

closed 2022-03-30 by davidplowman

6by9, 2022-03-30:
For reference, https://patchwork.linuxtv.org/project/linux-media/patch/1583846229-6799-3-git-send-email-prabhakar.mahadev-lad.rj@bp.renesas.com/
Part of a series that includes mods to support IMX219 on the Renesas rcar mipi-csi receiver, so that's the most likely device.