raspberrypi/rpicam-apps #284: [BUG] cannot set exif properties after the fact on raw DNGs produced by libcamera-still

vincentwoo, 2022-04-06:
**Describe the bug**
After saving camera output from the Raspberry PI HQ camera board as a raw dng with libcamera-still, I cannot seem to use exiftool to modify its EXIF data.

**Bug report**
output of `camera-bug-report -t 10 -o bug.txt -c "libcamera-still -n -r -o test.dng"` [attached](https://github.com/raspberrypi/libcamera-apps/files/8422660/camera-bug-report.txt).

Afterwards, attempting to do the following fails:

```
$exiftool -datetimeoriginal="2022:04:05 17:28:00" test.dng
Error: [minor] IFD1 pointer references previous ExifIFD directory - test.dng
    0 image files updated
    1 files weren't updated due to errors
```

It would be very helpful if libcamera-still automatically set DateTimeOriginal on DNGs!

vincentwoo, 2022-04-06:
Additionally, I find that stripping all EXIF data first seems to get past the error, implying there is something odd with the EXIF data libcamera-still writes by default (assumes exiftool is doing The Right Thing):

```
$ exiftool -all= test.dng
Warning: [minor] Can't delete IFD0 from DNG - test.dng
    1 image files updated
$ exiftool -DateTimeOriginal="2022:04:05 17:28:00" test.dng
    1 image files updated
```

davidplowman, 2022-04-07:
I've had a look at this. To be honest I find it all quite puzzling.

I believe the real problem is that there's a "bogus" IFD1 at the end of the dng file. Try `exiftool -v test.dng`. It seems to be a "duplication" of whichever the last sub-IFD was that was created. In the current code it's the EXIF IFD, but if you swap the creation of the EXIF IFD and the full image sub-IFD round, then it complains about that one instead.

When you "do something" to the file that "fixes" it, I think it's just "unsharing" these two IFDs, and the different "Strip Offsets" is a bit of a red herring. You can also "fix" the file with `exiftool -m -DateTimeOriginal="2022:04:05 17:28:00" test.dng`.

As usual, finding examples or documentation that are actually helpful is very difficult. [This](https://stackoverflow.com/questions/11959617/in-a-tiff-create-a-sub-ifd-with-thumbnail-libtiff) was about as good as I could find, and I don't see anything in the libcamera-apps code that's clearly wrong. So I don't understand why this "extra" IFD1 appears at the end. If anyone knows, I'd love to hear about it.

Anyway, besides "fixing" the file with `exiftool`, I've added a nasty-looking bodge that appears to make the problem go away, but I'd really prefer to have a better understanding. You can find it [here](https://github.com/raspberrypi/libcamera-apps/pull/285), I don't know if you're in a position to recompile libcamera-apps and give it a try.

vincentwoo, 2022-04-08:
I have my hacky workaround in production right now, so I don't want to risk a recompile/test, but I think its highly likely your fix works if you tested it and it seems to! I have absolutely no knowledge of how EXIF is supposed to work myself, and I very much appreciate your attention on this issue.

naushir, 2022-04-11:
Fixes now merged.  Feel free to close this issue once you have tested the latest changes.

davidplowman, 2022-04-20:
I think I'm going to close this one now, but please open another issue if anything pops up. Thanks!

closed 2022-04-20 by davidplowman