raspberrypi/rpicam-apps #286: [BUG] CircularBuffer::Available underflows when the buffer is empty

CarlLusty, 2022-04-09:
**Describe the bug**
CircularBuffer::Available underflows when the buffer is empty, returning -1 as a size_t.

This is a problem in CircularOutput::outputBuffer. The size of the buffer as reported by CircularBuffer::Available is very large, regardless of the buffers actual size. This means that 'std::runtime_error("circular buffer too small")' is never called, even if the buffer is too small. This can result in outputBuffer trying to Write to memory outside of the actual buffer and causing a crash.

**Bug report**
```
camera-bug-report -t 0 -o report.txt -c "libcamera-raw -t 3000 --circular 1 -o test.h264 -v --width 1920 --height 1080"
```
[report.txt](https://github.com/raspberrypi/libcamera-apps/files/8456212/report.txt)

davidplowman, 2022-04-09:
Oops! Thanks for reporting and diagnosing the problem. Patch [here](https://github.com/raspberrypi/libcamera-apps/pull/287) ready for merging.

CarlLusty, 2022-04-09:
Uhhh, I'm not sure that is what you really want.
Setting Available() to 0 when the buffer is empty will result in CircularOutput::outputBuffer always reporting 'circular buffer too small', right?

CarlLusty, 2022-04-09:
Wouldn't returning size_ or more likely size_ - 1 make more sense?

davidplowman, 2022-04-09:
Indeed, doing too many things at once!

naushir, 2022-04-11:
Fixes now merged.  Feel free to close this issue once you have tested the latest changes.

closed 2022-04-11 by CarlLusty