raspberrypi/rpicam-apps #291: [BUG] libcamera-apps stopped working

janvrska, 2022-04-24:
**Describe the bug**
Libcamera-still just waiting.

Today I just rebooted my raspberry and after that libcamera-apps stopped working. I'm not aware that I would update anything that might break it.

I used unchanged libcamera-apps from "Raspberry Pi OS with desktop, April 4th 2022" . So I tried to build it myself, and it didn't help. Then I updated kernel and it worked for a few minutes, but not anymore. Then I rebuild it on updated kernel and there is report from your tool.

I'm afraid there may be a hardware problem

**Bug report**
[report.txt](https://github.com/raspberrypi/libcamera-apps/files/8550548/report.txt)

naushir, 2022-04-25:
This possibly looks like a hardware/connectivity issue.  Can you ensure your camera connector cable is connected securely on both ends?  Alternatively, please try another camera board if you have another available to use.

janvrska, 2022-04-25:
Don't know why, but today it start working again. I Did nothing from yesterday and yes cable is connected securely. I checked it yesterday. Some ghosts in machine probably. But anyway, thank you for your time.

closed 2022-04-25 by janvrska