raspberrypi/rpicam-apps #292: libcamera-still always chooses maximum resolution RAW data.

john-ardu, 2022-04-27:
**Describe the bug**
When I use libcamera-still to take pictures, even I use the `--width`, `--height` options to specify the image resolution, libcamera-still still chooses the maximum resolution as the original data, it seems that libcamera-still chooses to crop on the maximum resolution Cut to the specified resolution.
In fact, if you don't care about memory issues, it's fine to do this, but when we use a camera with a larger resolution or shoot on a device with a smaller memory, even specifying a small resolution can't avoid the memory allocation problem.

**Bug report**
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/8570121/bug.txt)

**Additional context**
[image]

naushir, 2022-04-27:
``--width`` and ``--height`` only specify the output resolution of the final image.  You need to specify the ``--mode`` command line parameter to explicitly select a particular camera mode if you do not want the largest sensor mode.  See [here](https://www.raspberrypi.com/documentation/accessories/camera.html#common-command-line-options) for specific useage.

However, you ought to have enough memory even on a Pi Zero device.  Are you seen any captures running out of memory or is this speculative?

davidplowman, 2022-04-28:
Just to add that the behaviour - choosing the full resolution raw mode - is deliberate as this copies the behaviour or `raspistill` which we are (mostly) trying to emulate. In fact, the initial versions didn't do this, and users complained that it's not like `raspistill`!

In your particular case, `libcamera-still -o test.jpg --width 2028 --height 1520 --mode 2028:1520` should do what you want.

john-ardu, 2022-04-29:
Thanks for your quick reply,
I missed this option.

> However, you ought to have enough memory even on a Pi Zero device to use the largest sensor resolution. Are you seen any captures running out of memory or is this speculative?

It is enough for V2 or HQ, but not enough for Arducam 64mp.

closed 2022-04-29 by john-ardu