raspberrypi/rpicam-apps #293: [BUG] No post processing stage found for "annotate_cv"

davidjkim120, 2022-05-10:
**Describe the bug**
Getting error: 
```
No post processing stage found for "annotate_cv"
```
Command ran:
```
libcamera-vid --nopreview -t 0 --framerate 30 --width 1920 --height 1080 -b 3000000 --post-process-file /home/pi/BirdScripts/libcamera-postprocess.json
```

Contents of postprocess json:
```
{
    "annotate_cv" :
    {
		"text" : "Frame: %frame fps: %fps exp: %exp",
		"fg" : 255,
		"bg" : 0,
		"scale" : 1.0,
		"thickness" : 2,
		"alpha" : 0.3
    }
}

```

**Bug report**
camera-bug-report -t 1 -o bug.txt -c "libcamera-vid --nopreview -t 0 --framerate 30 --width 1920 --height 1080 -b 3000000 --post-process-file /home/pi/BirdScripts/libcamera-postprocess.json"

```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 09-05-2022 (16:59:25)
Command: /usr/local/bin/camera-bug-report -t 1 -o bug.txt -c libcamera-vid --nopreview -t 0 --framerate 30 --width 1920 --height 1080 -b 3000000 --post-process-file /home/pi/BirdScripts/libcamera-postprocess.json

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

[... 989 lines cut ...]
libcamera build: v0.0.0+3431-c4602118

--------------------------------------------------------------------------------
libcamera-vid --nopreview -t 0 --framerate 30 --width 1920 --height 1080 -b 3000000 --post-process-file /home/pi/BirdScripts/libcamera-postprocess.json
--------------------------------------------------------------------------------
[0:29:49.310012321] [30894]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3431-c4602118
[0:29:49.335441674] [30896]  WARN RPI raspberrypi.cpp:1233 Mismatch between Unicam and CamHelper for embedded data usage!
[0:29:49.336570961] [30896] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009a090a but control is not exposed by device /dev/v4l-subdev0
[0:29:49.336776551] [30896]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
No post processing stage found for "annotate_cv"
[0:29:49.338151965] [30894]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUV420
[0:29:49.339021736] [30896]  INFO RPI raspberrypi.cpp:751 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
[0:29:49.384565321] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: Saturation = 1.000000
[0:29:49.384660339] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: Contrast = 1.000000
[0:29:49.384683283] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: Brightness = 0.000000
[0:29:49.384703079] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: AwbMode = 0
[0:29:49.384722653] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: Sharpness = 1.000000
[0:29:49.384742246] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: ExposureValue = 0.000000
[0:29:49.385152649] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: AeExposureMode = 0
[0:29:49.385184705] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:29:49.385214186] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: AeMeteringMode = 0
[0:29:49.385672237] [30903]  INFO IPARPI raspberrypi.cpp:666 Request ctrl: NoiseReductionMode = 1
Error: ***** TIMEOUT *****

```

davidplowman, 2022-05-10:
Hi, we don't normally build and link the OpenC V-based post-processing stages because not all systems have OpenCV installed. Have you tried to build them for yourself? [This](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps-without-rebuilding-libcamera) would be the place to start.

naushir, 2022-05-20:
Closing this now...

closed 2022-05-20 by naushir