raspberrypi/rpicam-apps #295: How to create the deb file

supertobi, 2022-05-20:
It would be nice to have the scripts to build a deb file for libcamera-apps in this repository or add the instructions how to do it to this documentation:
https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-libcamera-apps

naushir, 2022-05-23:
Could I ask what you intend to do with these packages. We already have the libcamera apps packaged up of-course, so you could use those directly?

supertobi, 2022-05-23:
> Could I ask what you intend to do with these packages.

Yes, I have the 16MP autofocus camera from arducam.com. ArduCam provides deb files with an patched version of libcamera-apps and the patches. Sadly this version still has the bug with the DNGs: https://github.com/raspberrypi/libcamera-apps/pull/285
Now I forked libcamera-apps merged the patches from ArduCam by hand with the latest version: https://github.com/supertobi/libcamera-apps

It compiles and works fine but I like it more if software is installed via an clean DEB file. As you already have the libcamera apps packaged it would be nice if you could share you receipt how to do it.

naushir, 2022-05-23:
I don't know anything about packaging, perhaps @XECDesign can help.  However, any scripts we might have are specifically for this repo, and we won't be able to provide support for 3rd party packages.

supertobi, 2022-05-23:
1. I've just created a PR with the patches, perhaps its easier to merge them into the main repository.

>However, any scripts we might have are specifically for this repo, and we won't be able to provide support for 3rd party packages.

2. That would be OK, every tiny piece would be better then nothing.

XECDesign, 2022-05-23:
All the standard Debian packaging documentation applies.

https://www.debian.org/doc/manuals/maint-guide/

Enable the source repo in `raspi.list`, run `sudo apt update` and then `apt source libcamera-apps` will fetch everything you might need.

naushir, 2022-05-26:
Assuming this is all resolved now.  Please reopen if you have any further questions.

closed 2022-05-26 by naushir