raspberrypi/rpicam-apps #297: [BUG] ExposureTime / ShutterSpeed

Bra1nsen, 2022-05-27:
How set minimum ExposureTime / maximum ShutterSpeed.

**Dear Developers**, I working on a SkyCam for SolarPowerPredictions. I created some examples..

`libcamera-jpeg -o shutterspeedxy.jpg --shutter xy`

[image]

Ironically there comes a point where the exposure increases again instead of continuing to decrease.

However I just would like to reduce the exposure time further? How can I set the maximum possible, do i need to add more settings?

Thank you!

Bra1nsen, 2022-05-27:
[image]

the longer the exposure time, the brighter the image..

davidplowman, 2022-05-27:
Hi, could you please tell us what camera you are using. Could you then execute the following commands
```
libcamera-jpeg -v --gain 1 --shutter 1000 -o shutter_1000.jpg
libcamera-jpeg -v --gain 1 --shutter 100 -o shutter_100.jpg
```
Each capture will output a log, and near the end you'll see it reporting the "Exposure time:" and the "Ag" and "Dg" (analogue and digital gain values). Could you post the values that you see there? They can also be confirmed by looking in the EXIF data (with a tool like `exiftool`, for example) for the "Exposure Time" and "ISO" fields. Thanks.

Bra1nsen, 2022-05-27:
Raspberry HQ Cam (IMX477)

`libcamera-jpeg -v --gain 1 --shutter 1000 -o shutter_1000.jpg`

`Still capture image received
Exposure time: 1000
Ag 1 Dg 1 Total 1
Thumbnail dimensions are 320 x 240
Thumbnail size 8759
JPEG size is 1547788
EXIF data len 266
Closing Libcamera application(frames displayed 27, dropped 118)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
`

`libcamera-jpeg -v --gain 1 --shutter 100 -o shutter_100.jpg`

`Still capture image received
Exposure time: 571
Ag 1 Dg 1 Total 1
Thumbnail dimensions are 320 x 240
Thumbnail size 7761
JPEG size is 1508091
EXIF data len 266
Closing Libcamera application(frames displayed 12, dropped 133)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
`

ok 571 seems to be the minimum

davidplowman, 2022-05-27:
That sounds about right for an imx477. You can probably go a bit lower with a different camera mode (try adding `--mode 2028:1540` to the commands). For me those images are really very dark - how are yours?

Bra1nsen, 2022-05-27:
[image]
 thats a image i made in april with allsky software 0,1ms exposure time, iam wondering why i cant get the same results..

davidplowman, 2022-05-27:
How is the lens's aperture set? That has quite a big effect too.

davidplowman, 2022-05-27:
Do you have the jpeg from the old setup so that you can check what the EXIF claims?

Bra1nsen, 2022-05-27:
unfortunately i didn't save the pics but i will build the same setup again in parallel, once i have the results (middle next week) i will post them. I'm curious about the results. ..

davidplowman, 2022-05-30:
It does seem that the legacy camera stack lets you use shorter exposures, I'm suspicious that the minimum exposure number in the Linux kernel driver is not correct. We'll take a look.

Bra1nsen, 2022-05-30:
libcamera:

[image]

old setup with camera stack:

[image]

[image]

Note: There are **not exactly the same lenses**, but comparable. I know this is not an easy question to answer, but how long do you think it will take to fix it? Did you already find the cause?

Bra1nsen, 2022-05-31:
I'm super grateful that people like you care about this. end of next week would be outstanding. If I can support you in any way, let me know. i have two cameras on the roof, one with libcam and the other with stack legacy.

Bra1nsen, 2022-06-14:
hey david how are you doing? Any Updates to the exposure difficulty? 

happy greetings
Paul

davidplowman, 2022-06-14:
Hi, yes I think that change should be available in the latest development kernel. You can get that by running `sudo rpi-update`. (Usual warning: this is an untested kernel. Back up anything that you really don't want to lose!)

Bra1nsen, 2022-06-14:
on the scale from 1-10 - how safe is this david?

davidplowman, 2022-06-15:
Normally I don't notice any problems at all. I think people say there are quite often things that aren't right, but in fairly out of the way places that you'd be unlucky to notice. Very occasionally I've actually noticed problems, but that's pretty unusual. I don't think I've ever lost my entire image so that I can't recover it, but the danger exists. Back up your bitcoin wallet somewhere else!

Do you have a spare SD card where you could download a fresh Bullseye, do `sudo rpi-update` and try the camera? That would seem a pretty low risk way of verifying the likely behaviour.

Bra1nsen, 2022-06-26:
great thank you it works fine! What did you set as minimal exposure time for the imx477?

Bra1nsen, 2022-06-26:
[image]

LEFT
`libcamera-jpeg -o ddd.jpg --shutter 1 --gain 1 `

RIGHT 
```
 camera.set_controls({"ExposureTime": 1, "AnalogueGain": 1})
 camera.capture_file('ldr_01.jpg')
```

shouldn't these be exactly the same pictures?

davidplowman, 2022-06-27:
Hi, I believe the minimum exposure time at full resolution should be 114us, but the EXIF in the jpeg should have the correct value.

As regards using Picamera2, libcamera has quite a long pipeline of frames between exposure being changed and the new value taking effect. In fact even the image sensor on its own takes several frames; it can't do it immediately. However, if you set the values you want _before_ starting the camera then the very first frame will have the values you requested. Does that help?

Bra1nsen, 2022-06-27:
114us - interesting. how did you come up with that?

"very first frame will have the values you requested"

how do i write this in code? It's still about exposure time series with Python.

happy greetings

davidplowman, 2022-06-27:
> 114us - interesting. how did you come up with that?

This is what you get if you set the minimum number of exposure lines to 4 (sensors normally count the exposure internally in units of the time it takes to scan out a row of pixels), which matches what the legacy driver seemed to be doing.

> 
> "very first frame will have the values you requested"
> 
> how do i write this in code? It's still about exposure time series with Python.

Something like this:
```
# Stop the camera if it's running, and configure it into the capture mode. Then:
picam2.set_controls({"ExposureTime": 1, "AnalogueGain": 1.0})
picam2.start()
picam2.capture_file("test.jpg")  # this image will have minimum exposure
```
To do this without stopping the camera, you would have to add a loop that monitors the image metadata, and spots when you see the values you asked for.
> 
> happy greetings

Bra1nsen, 2022-07-01:
hey david 

Is there a way to display all camera settings that were set for an image (e.g. exposure time)? 

My HDRmerge program (radiance-online) gives me an error message with the camera response function. I finally want to merge the exposure time series...

It works on legacy stack... so I suspect there may be discrepancies between the images.

"Most cameras play tricks with the response curve, white balance, and other capture characteristics that undermine any attempt to get a consistent HDR result. Just a true "manual mode" is suited to HDR capture"

exif ldr_02 legacy
[image]

exif ldr_02 libcamera
[image]

naushir, 2022-07-01:
> Is there a way to display all camera settings that were set for an image (e.g. exposure time)?

There was a recent PR merged that may help you - https://github.com/raspberrypi/libcamera-apps/pull/307.  It adds a ``--metadata`` command line argument to ``libcamera-still`` that outputs the request metadata from libcamera.  It has the same values (and more) that are provided in the exif of the capture

Bra1nsen, 2022-07-01:
ok is there an analog option in camera.set_controls() ? I use a python script to generate multiple images with different exposure time/shutter speed values. greets

davidplowman, 2022-07-01:
Not quite sure what's going on there. The "exif ldr_02 libcamera" window doesn't look like it's showing the output for a file produced by libcamera-still or Picamera2. Both of those would normally have some make/model info, as well as exposure time and ISO. Can you say anything more about how you made those images?

Bra1nsen, 2022-07-02:
> Not quite sure what's going on there. The "exif ldr_02 libcamera" window doesn't look like it's showing the output for a file produced by libcamera-still or Picamera2. Both of those would normally [...]

```
from picamera2 import *
from time import sleep
from fractions import Fraction

with picamera2.Picamera2() as camera:
	
		config = camera.still_configuration()
		camera.configure(config)
     
		camera.start()
		camera.set_controls({"ExposureTime": 3000, "AnalogueGain": 1})
		camera.capture_file('ttx.jpg')
		print(camera.capture_metadata())
		camera.stop()
```

exiftool ttx.jpg

[image]

Bra1nsen, 2022-07-02:
> Yes, there is a mechanism in picamera2. Have a look at the examples [here ](https://github.com/raspberrypi/picamera2/blob/main/examples/metadata.py)and [here](https://github.com/raspberrypi/picamera2/blob/main/examples/metadata_with_image.py) to obtain the request metadata. This will be the same key/value pair as output by the `--metadata` flag.

```
from picamera2 import *
from time import sleep
from fractions import Fraction

with picamera2.Picamera2() as camera:
	
		config = camera.still_configuration()
		camera.configure(config)
     
		camera.start()
		camera.set_controls({"ExposureTime": 300, "AnalogueGain": 1})
		camera.capture_file('ttx1.jpg')
		print(camera.capture_metadata())
		
		camera.stop()
		camera.start()

		camera.set_controls({"ExposureTime": 60000, "AnalogueGain": 1})
		camera.capture_file('ttx2.jpg')
		print(camera.capture_metadata())
		camera.stop()
```
[image]

Why is there no difference between 0,3ms and 6ms exposure time?

davidplowman, 2022-07-02:
Hi, don't really have much time to help at weekends, but two quick things.

1. You need to call set_controls before start, otherwise there is a many frame delay for the values to take effect.
2. Not sure why you're not seeing many exif fields. If you have the source check out the save method in request.py. You can see it creating exif fields in there, I don't understand why you aren't seeing them.

naushir, 2022-07-05:
Is there any further work needed for this?  If not I will close this issue down.

Bra1nsen, 2022-07-06:
Yes, the original problem was solved, thanks

closed 2022-07-06 by Bra1nsen

davidplowman, 2022-07-06:
Hi, it sounds like some kind of version mismatch, perhaps you've updated libcamera but your Picamera2 doesn't match? Can you just explain how you updated libcamera (`sudo apt install python3-libcamera`?) and how you updated Picamera2 (either with `pip install picamera2 --upgrade`, or have you checked out the sources)? Thanks.

Bra1nsen, 2022-07-06:
all i did was `sudo apt-get update`

davidplowman, 2022-07-06:
That's a bit strange. But I think that particular error is fairly typical of your libcamera being newer than your picamera2. You could try updating both to the latest release, but note that there is one breaking API change. Where you were calling `camera.still_configuration()` you would now need to call `camera.create_still_configuration()` (see [here](https://forums.raspberrypi.com/viewtopic.php?t=337013)).

Bra1nsen, 2022-07-06:
awesome david it worked after ` pip install picamera2 --upgrade`