raspberrypi/rpicam-apps #300: Documentation on imx477 capabilities needed

EricClaeys, 2022-06-15:
**Describe the bug**
What are the minimum, maximum, and default values for all the HQ camera's capabilities when using `libcamera` (like exposure, sharpness, brightness, etc.).  I have seen bits and pieces of this information, but not everything.  I want to put them into the [Allsky ](https://github.com/thomasjacquin/allsky) program so it can validate user input.

Does this information exist?  If not, can it be documented?

It would be REALLY nice if there was a command that output this information in a machine-readable way, so I could effectively query the camera.  ZWO's library has that capability.

naushir, 2022-06-22:
Once the libcamera changes are in, you can see an example of how to get control capabilities in https://github.com/raspberrypi/libcamera-apps/pull/305.

naushir, 2022-06-29:
All the relevant changes have now been merged into libcamera and libcamera-apps.  We will kick of the process to get a release into apt, or you can pull the sources and rebuild the libraries to test them out earlier.