raspberrypi/rpicam-apps #302: Still stream effective frame rate halved

Schaggo, 2022-06-18:
Hi,

I am trying to capture still images as fast as possible with an app based on libcamera-still.

Somehow, the effective frame rate is half of the set frame rate (for example set frame rate = 2, but only every full second a buffer is filled with data). This happens until I reach a limit (~9 fps) where the program can’t keep up with the data and the frame rate is no longer constant.

Do you have an idea what happens here, and how I might achieve set fps = effective fps?

Thank you for your time!

naushir, 2022-06-21:
One likely issue that may cause this is the lack of buffering.  How many buffers and requests have you made available for your application to use?  You will likely need 2 or more to keep up with the rate you request.

naushir, 2022-06-21:
The number of frame buffers used is not controlled by ``max_frame_buffers`` in ``config.txt``.  Please revert that back to the original value.

To control the number of buffers used in the camera, you need to set the buffer count like libcamera-apps does [here](https://github.com/raspberrypi/libcamera-apps/blob/2f1136f872732f0c8ba8c43df7c475c1e8be516a/core/libcamera_app.cpp#L255).  You must also ensure you have enough requests circulating through your application to keep the pipeline handler from dropping frames.

closed 2022-06-21 by Schaggo

Schaggo, 2022-07-20:
I managed to capture raw images at full res. But, now I have the problem that the effective framerate is stuck at 16.666 fps. 

I set the frame time via 
`int64_t frame_time = 1000000 / options_->framerate; // in us
				controls_.set(controls::FrameDurationLimits, { frame_time, frame_time });`

and assigned 4 via 
`else if ((flags & FLAG_STILL_BUFFER_MASK) == FLAG_STILL_QUAD_BUFFER)
		configuration_->at(0).bufferCount = 4;`

If I set the frame rate higher than 8 (until 20) the effective frame rate is at 16.6.
Any idea what's going on?

naushir, 2022-07-20:
Can you provide the command line you are using and what sensor is connected?

Schaggo, 2022-07-20:
Sure thing: 
I'm using the imx519 with the following constructed command:
`libcamera-still --raw -t 0 -s -v -o
    /home/pi/images/offsetsynch_.dng
    --shutter 10000 --framerate 10
    --ev 0 --awbgains 1.6,1.5
    --gain 1 --denoise off  --exposure sport --mode 4656:3496:10 &`

naushir, 2022-07-20:
I presume the imx519 is an Arducam part?  If so you will have to go through their support channels as this is not an official Raspberry Pi camera.