raspberrypi/rpicam-apps #303: [BUG] Memory leak in DRM preview

asmirnou, 2022-06-19:
**Describe the bug**

Raspberry Pi 4 with Lite OS and HQ camera, HDMI monitor plugged in.
libcamera-still in timelapse mode having captured 9-10 images fails to allocate more frame buffers. 
With ` --nopreview` option images are captured continuously. 

```
libcamera-still --timeout 0 --timelapse 2000 -o test.jpg
```

```
...
Still capture image received
[0:20:03.496245571] [3271]  INFO Camera camera.cpp:1029 configuring streams: (0) 2028x1520-YUV420
[0:20:03.516958039] [3272]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:20:04.303260783] [3271]  INFO Camera camera.cpp:1029 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[0:20:04.305372158] [3272]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received
[0:20:05.695079489] [3271]  INFO Camera camera.cpp:1029 configuring streams: (0) 2028x1520-YUV420
[0:20:05.716344063] [3272]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:20:06.503059414] [3271]  INFO Camera camera.cpp:1029 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[0:20:06.505150535] [3272]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[0:20:06.574413779] [3272] ERROR V4L2 v4l2_videodevice.cpp:1211 /dev/video14[16:cap]: Unable to request 1 buffers: Cannot allocate memory
[0:20:06.574476441] [3272] ERROR RPI raspberrypi.cpp:1008 Failed to allocate buffers
```

**Bug report**

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/8935127/bug.txt)

**Additional context**

`--keypress` option instead of `--timelapse` leads to the same effect.

The bug may affect other apps (vid, detect), basically where DRM preview resets periodically.

Reducing size of preview using `--viewfinder-width 640 --viewfinder-height 480` captures images a little longer.

naushir, 2022-06-21:
Hmmm, I can reproduce this as well. @davidplowman do you have any ideas about why the DRM preview may fail to release some buffers?

davidplowman, 2022-06-21:
Hmm, yes I can reproduce this quite easily too. I am a bit suspicious of those bo_handle things that are flying about. I have no idea what they are, but maybe they need releasing or closing in some way. Let me do some snooping.

asmirnou, 2022-06-21:
Thank you for quick fix! I confirm it resolves the issue.