raspberrypi/rpicam-apps #306: Access to sensor temperature requested

EricClaeys, 2022-06-23:
I realize that most people using the HQ camera probably don't care what the sensor temperature is, but it's very useful for people taking long exposures, like those of us using the camera in an allsky setup at night, Knowing the temperature allows us to subtract "dark" images (images taken at various temperatures with the camera covered so the only thing in the image is dark noise and hot pixels).  This significantly decreases the noise in the images.

Astronomy cameras like ZWO and QHY report the sensor temperature and it would be very useful if the HQ camera did as well, especially if we could get to it via our "allsky" software.

6by9, 2022-06-23:
Duplicate of https://github.com/raspberrypi/libcamera/issues/19

The IMX477 does have an on-sensor temperature sensor.

There is no V4L2 control for reading back temperature though. There was a driver submitted [fairly recently](https://lore.kernel.org/linux-media/20220415111845.27130-3-benjamin.mugnier@foss.st.com/) that tried adding it, but it started a discussion about whether it should be a V4L2 control, or to go through the hwmon API.
I queried it as a [point for discussion at an upcoming Media Summit](https://www.spinics.net/lists/linux-media/msg212772.html), and the one response views a V4L2 control as their preferred solution.

The problem is that if we adopt a V4L2_CID_ value before it gets merged to mainline, then it may change as mainline add additional controls, and that breaks existing apps. If we have agreement with mainline, then I can get the V4L2_CID_ assigned, and we can then add it to imx477 easily.

naushir, 2022-06-23:
I'll look into enabling the temperature sensor on the imx477.  As @6by9 said, there is no V4L2 control to return the value back, but it does get return out in embedded data.  So if we can agree with the libcamera dev team to add a temperature control, that may be enough to start with.

6by9, 2022-06-23:
> but it does get return out in embedded data.

Only libcamera gets to see the embedded data contents, so that pushes it back into libcamera (rather than V4L2) and makes adding a new control easier.
It needs the driver to enable the temperature sensor, and libcamera can then pull it out.

naushir, 2022-06-23:
I've prototyped something that seems to work well enough.  I will post the patches to the libcamera mailing list and our kernel driver.  The latter change is trivial, but the former will need review from the libcamera team - @kbingham 😄

kbingham, 2022-06-23:
I already know I like the idea of getting the temperature in the metadata for completed requests ;-)

naushir, 2022-06-23:
I'd do anything to avoid me making kernel patches ;-)

EricClaeys, 2022-06-23:
@naushir, @kbingham, @6by9,  You guys are great.  I put a request in and within hours you reply that you know what needs to be done and you're working on it.  THANKS!

Our "allsky" software calls `libcamera-still` via the `system()` call in a C++ program so all we get back is an exit code.  Would it be possible to have `libcamera-still` write all the metadata (e.g., temperature, actual exposure, actual gain, actual WBR, actual WBB, and other settings) to a file that's specified on the command line?  e.g., `libcamera-still --metadata /tmp/metadata.txt [other arguments...]`.
If it wrote the data in a simple `SETTING_NAME=value` format it would be super easy to read and easy for humans to parse.
Using the "verbose" option to write to stderr is pretty clunky since many other things are written to stderr and it's not in a format that's easy to parse.

Thought?

EricClaeys, 2022-06-23:
Just curious...
I noticed the patch clamps the temperature to between -20 and 80.  Most chips can go lower than -20; not sure about above 80, and they report the temperature with at least 1 digit precision.  Does the imx477 (or the libcamera/Linux software) force it to be an integer between -20 and 80 ?
The "allsky" software rounds floating point temperatures to an integer, and dark current below -20 C is usually not much different than at 20 C, so this really has no impact, but I was just curious.

EricClaeys, 2022-06-23:
@kbingham Kieran, as someone who will use this information, documenting the highlighted line is important.  Are there any other times a lack of reported temperature would happen?

> +        The SensorTemperature control will only be returned in metadata if a themal sensor is present.

> Hrm - I don't know if we 'need' to say this last bit, but it seems correct.  Presumably there are other metadata fields that are optional to provide and are only in the metadata if they are available.

naushir, 2022-06-24:
> @naushir, @kbingham, @6by9, You guys are great. I put a request in and within hours you reply that you know what needs to be done and you're working on it. THANKS!
>
> Our "allsky" software calls [...]

To start with, I will put the sensor temperature as a field in the exif of the image file (DNG/JPEG).  This ties in with how we report other properties (e.g. gain, exposure, black level, etc.).  I don't know anything about the allsky software, but I assume it already reads exif data to extract some of these values.  So it ought to be trivial to add some code to extract the sensor temperature.  Do let me know if this is not correct though!

naushir, 2022-06-24:
> **UPDATE**: I saw in another patch that the temperature is a float, so ignore that part of the post below.
> 
> Just curious... I noticed the patch clamps the temperature to between -20 and 80. Most chips can go lower than -20; not sure about above 80, and they report the temperature with at least 1 digit precision. Does the imx477 (or the libcamera/Linux software) force it to be an integer between -20 and 80 ? The "allsky" software rounds floating point temperatures to an integer, and dark current below -20 C is usually not much different than at 20 C, so this really has no impact, but I was just curious.

The -20C to 80C integer range comes from the thermal sensor limitation.  I'm expect other devices to have different ranges and precision.

naushir, 2022-06-24:
> To start with, I will put the sensor temperature as a field in the exif of the image file (DNG/JPEG). This ties in with how we report other properties (e.g. gain, exposure, black level, etc.). I [...]

After a brief look into this, there is no standard exif tag for sensor temperature.  Annoyingly there is a tag for ambient temperature but I'd rather not use that!  Adding non-standard tags is quite painful and prone to breakages depending on what libraries are used to read/write tags, so for now perhaps the best approach is indeed to write out metadata to a txt file.

EricClaeys, 2022-06-25:
There may be additional metadata that won't have standard exif tags, so it's probably better long-term to write what has standard tags to exif and optionally write all metadata to a file.
Will the metadata include things like auto white balance settings, auto exposure, auto gain, and whatever other values can be automatically set?

naushir, 2022-06-29:
> Will the metadata include things like auto white balance settings, auto exposure, auto gain, and whatever other values can be automatically set?

The metadata will include everything the libcamera library has returned out with the request.  This is an example file generated with the change in https://github.com/raspberrypi/libcamera-apps/pull/307:

```
ExposureTime=30000
SensorTimestamp=82044065837000
ScalerCrop=(0, 0)/4056x3040
SensorBlackLevels=[ 4096, 4096, 4096, 4096 ]
ColourTemperature=4483
DigitalGain=1.002235
ColourGains=[ 2.742903, 1.774089 ]
AeLocked=true
SensorTemperature=26.000000
Lux=148.950699
FrameDuration=100000
ColourCorrectionMatrix=[ 2.016614, -0.904894, -0.111727, -0.389223, 1.812109, -0.422884, -0.104301, -0.571240, 1.675540 ]
AnalogueGain=3.413333
FocusFoM=5079
```

naushir, 2022-06-29:
All the relevant changes have now been merged into libcamera and libcamera-apps.  We will kick of the process to get a release into apt, or you can pull the sources and rebuild the libraries to test them out earlier.

closed 2022-06-29 by naushir

naushir, 2022-06-30:
> @naushir, some questions on the metadata:
> 
> `ScalerCrop=(0, 0)/4056x3040` Is this the actual size of the image and the starting x,y coordinates if cropped?

Yes, these give you the crop in the sensor image together with its size.

> 
> `ColourGains=[ 2.742903, 1.774089 ]` is that red, blue or blue, red ?

The ordering is Red - Blue.

> 
> `AeLocked=true` is this Auto Exposure Locked, i.e., whether or not auto exposure was on? If so, is there an equivalent for auto gain?

This control essentially tells you if the AE algorithm has converged to a stable target.  It includes shutter speed and gain considered together.

> 
> `Lux=148.950699` the average brightness of the image, from 0 to 255 ? How is this value determined?

This is the estimated scene lux value, so the range is only lower bounded to 0.  It is determined based on calibration performed during IQ tuning.  You can find the reference parameters in the tuning file under the ``rpi.lux`` section.