raspberrypi/rpicam-apps #309: Exif data missing

wcbonner, 2022-06-30:
These items were available in raspistill, but are not embedded by libcamera-still. The dates with those standardized names are especially useful when using the --timelapse feature and working with images later.

```
Exif Version                    : 0220
Date/Time Original              : 2022:06:26 19:15:48
Create Date                     : 2022:06:26 19:15:48
```

These items were embedded in raspistill and not by libcamera-still. I've not been using them and am less concerned about their absence, but if they are available, it would be nice to have them for completeness.

```
X Resolution                    : 72
Y Resolution                    : 72
Resolution Unit                 : inches
Y Cb Cr Positioning             : Centered
F Number                        : 2.9
Exposure Program                : Aperture-priority AE
Components Configuration        : Y, Cb, Cr, -
Shutter Speed Value             : 1/30
Aperture Value                  : 2.9
Brightness Value                : 2.56
Max Aperture Value              : 2.9
Metering Mode                   : Center-weighted average
Flash                           : No Flash
Focal Length                    : 3.6 mm
Maker Note Unknown Text         : (Binary data 316 bytes, use -b option to extract)
Flashpix Version                : 0100
Color Space                     : sRGB
Exif Image Width                : 1920
Exif Image Height               : 1080
Interoperability Index          : R98 - DCF basic file (sRGB)
Exposure Mode                   : Auto
White Balance                   : Auto
Aperture                        : 2.9
Focal Length                    : 3.6 mm
Light Value                     : 7.3
```

davidplowman, 2022-06-30:
Hi, thanks for the questions.

So we add EXIF_TAG_DATE_TIME already, but apparently exiftool reports that as the "Modify Date". I'm happy to add EXIF_TAG_DATE_TIME_ORIGINAL ("Date/Time Original") and EXIF_TAG_DATE_TIME_DIGITIZED ("Create Date") too (https://github.com/raspberrypi/libcamera-apps/pull/311)

I'm nervous about adding EXIF_TAG_EXIF_VERSION because I'm not sure I want to promise any particular level of conformance, simply because I have no idea whether the software does indeed conform or not.

As regards the other ones - some are certainly doable, others might require information that we possibly don't have without doing more work elsewhere, so mostly I'd like to take a wait-and-see approach on those.

naushir, 2022-07-04:
Patches have been merged, and the change will be available in the next release.  Closing for now.

closed 2022-07-04 by naushir