raspberrypi/rpicam-apps #314: [BUG] libcamera-still crashes

roelfrenkema, 2022-07-08:
*--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 08-07-2022 (20:29:17)
Command: /usr/local/bin/camera-bug-report -t 5 -o bug.txt -c libcamera-still -t 1000 -o test.jpg

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
camera01

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03112
Serial		: 10000000134ef266
Model		: Raspberry Pi 4 Model B Rev 1.2

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=a2d140d5-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

** cat /boot/config.txt **
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
[... 951 lines cut ...]
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Mar 24 2022 13:20:19 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start_x)

** libcamera-still --version **
libcamera-apps build: 1cdc057d2db9-intree 28-06-2022 (05:12:42)
libcamera build: v0.0.0+3572-e13ccdd5

--------------------------------------------------------------------------------
libcamera-still -t 1000 -o test.jpg
--------------------------------------------------------------------------------
libEGL warning: DRI3: failed to query the version
libEGL warning: DRI2: failed to authenticate
X Error of failed request:  BadRequest (invalid request code or no such operation)
  Major opcode of failed request:  155 ()
  Minor opcode of failed request:  1
  Serial number of failed request:  16
  Current serial number in output stream:  16

naushir, 2022-07-10:
It looks like you might have ``dtoverlay=vc4-fkms-v3d`` in your ``config.txt`` file.  Please remove that line.  Also, you seem to be manually setting the cma heap size.  Unless you have a very specific reason, you should not be doing this.

naushir, 2022-07-10:
You also seem to be using the arducam camera.  So I would suggest going through their support channels.  They have modified our libcamera library software, so we cannot provide support here.

closed 2022-07-10 by naushir