raspberrypi/rpicam-apps #315: [BUG] libcamera-apps-lite package upgrade pulls in X11 libraries

rlee287, 2022-07-09:
**Describe the bug**
Attempting to upgrade the `libcamera-apps-lite` package upgrade pulls in X11 libraries, even though the package description states that "This version excludes features which require X11 and Qt". This is show in the commands+outputs below.

If this is not the appropriate place to report this packaging error, please point me in the correct direction.

**Bug report**

```
$ sudo apt update
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease                          
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease                  
Hit:4 https://download.docker.com/linux/debian bullseye InRelease              
Hit:5 http://archive.raspberrypi.org/debian bullseye InRelease                 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
$ apt list --upgradable 
Listing... Done
libcamera-apps-lite/stable 0~git20220630+765bb46-2 arm64 [upgradable from: 0~git20220228+0dc5ea0-1]
libcamera0/stable 0~git20220705+f30ad033-1 arm64 [upgradable from: 0~git20220510+22656360-1]
$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  fontconfig libaacs0 libaom0 libass9 libasyncns0 libavc1394-0 libavcodec58
  libavdevice58 libavfilter7 libavformat58 libavutil56 libbdplus0 libblas3
  libbluray2 libbs2b0 libcaca0 libcairo-gobject2 libcairo2 libcdio-cdda2
  libcdio-paranoia2 libcdio19 libchromaprint1 libcodec2-0.9 libdatrie1
  libdav1d4 libdc1394-25 libdrm-amdgpu1 libdrm-nouveau2 libdrm-radeon1
  libepoxy0 libfftw3-double3 libflite1 libfribidi0 libgbm1 libgdk-pixbuf-2.0-0
  libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgfortran5 libgl1
  libgl1-mesa-dri libglapi-mesa libglvnd0 libglx-mesa0 libglx0 libgme0
  libgraphite2-3 libgsm1 libharfbuzz0b libiec61883-0 libjack-jackd2-0
  liblapack3 liblilv-0-0 libllvm11 libmp3lame0 libmpg123-0 libmysofa1 libnorm1
  libnuma1 libopenal-data libopenal1 libopenjp2-7 libopenmpt0 libopus0
  libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpgm-5.3-0
  libpixman-1-0 libpocketsphinx3 libpostproc55 libpulse0 librabbitmq4
  libraw1394-11 librsvg2-2 librsvg2-common librubberband2 libsdl2-2.0-0
  libsensors-config libsensors5 libserd-0-0 libshine3 libsnappy1v5 libsndfile1
  libsndio7.0 libsodium23 libsord-0-0 libsoxr0 libspeex1 libsphinxbase3
  libsratom-0-0 libsrt1.4-gnutls libssh-gcrypt-4 libswresample3 libswscale5
  libthai-data libthai0 libtheora0 libtwolame0 libudfread0 libva-drm2
  libva-x11-2 libva2 libvdpau-va-gl1 libvdpau1 libvidstab1.1 libvorbisenc2
  libvorbisfile3 libvpx6 libvulkan1 libwavpack1 libwayland-client0
  libwayland-cursor0 libwayland-egl1 libwayland-server0 libwebpmux3
  libx11-xcb1 libx264-160 libx265-192 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0
  libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shape0 libxcb-shm0
  libxcb-sync1 libxcb-xfixes0 libxcursor1 libxdamage1 libxfixes3 libxi6
  libxinerama1 libxkbcommon0 libxrandr2 libxrender1 libxshmfence1 libxss1
  libxv1 libxvidcore4 libxxf86vm1 libz3-4 libzmq5 libzvbi-common libzvbi0
  mesa-va-drivers mesa-vdpau-drivers mesa-vulkan-drivers ocl-icd-libopencl1
  pocketsphinx-en-us shared-mime-info va-driver-all vdpau-driver-all
  x11-common
The following packages will be upgraded:
  libcamera-apps-lite libcamera0
2 upgraded, 154 newly installed, 0 to remove and 0 not upgraded.
Need to get 113 MB of archives.
After this operation, 328 MB of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.
$ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

Scratchcat1, 2022-07-09:
The dependency chain appears to be `libcamera-apps-lite` -> `libavcodec58` -> `libcairo2` -> X11 libraries.

The need for libavcodec was added in [this commit](https://github.com/raspberrypi/libcamera-apps/commit/f22e148a71f15a7844ac32bcec310fb0834332ef).

naushir, 2022-07-10:
With https://github.com/raspberrypi/libcamera-apps/pull/313 merged, libav will not be linked for the lite release in the latest packages.

closed 2022-07-10 by naushir