raspberrypi/rpicam-apps #319: libcamera-still produces a huge number of INFO messages to stderr with --verbose 0 set

wcbonner, 2022-07-16:
I'm running libcamera-still as part of a service, and it is very sending lots of information messages that I don't find useful. 

It seems that there should be a way to make it less verbose than the default. 

--verbose 1 might be appropriate for the default verbosity, while --verbose 0 should only print ERROR messages. 

My current solution is to redirect stderr to /dev/null, but that means I don't see the actual error messages.

naushir, 2022-07-17:
By default, there should not be any per-frame logging from libcamera-still - unless you use the ``-v`` command line option.  Can you share some of this log to have a look at please?

You could also disable libcamera logging by setting the ``LIBCAMERA_LOG_LEVELS=*:4`` environment variable.

wcbonner, 2022-07-17:
Thanks. Knowing about the environment variable reduces output significantly, but not completely. Can you point me to a description of what environment variables are recognized?

Initially I was seeing this:
```
wim@WimPiZeroW-Hope:~ $ libcamera-still --nopreview --thumb none --width 1920 --height 1080 --timeout 59580000 --timelapse 60000 --output /media/WimPiZeroW-Hope/DCIM/20220717/0717%04d.jpg --framestart 284 --version --verbose 0
libcamera-apps build: 35266e84043f 07-07-2022 (17:54:26)
libcamera build: v0.0.0+3700-f30ad033
wim@WimPiZeroW-Hope:~ $ libcamera-still --nopreview --thumb none --width 1920 --height 1080 --timeout 59580000 --timelapse 60000 --output /media/WimPiZeroW-Hope/DCIM/20220717/0717%04d.jpg --framestart 284 --verbose 0
[16:05:41.315855257] [31740]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[16:05:41.542365221] [31741]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[16:05:41.547318524] [31740]  INFO Camera camera.cpp:1029 configuring streams: (0) 1296x728-YUV420
[16:05:41.551613319] [31741]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
[16:06:41.755427156] [31740]  INFO Camera camera.cpp:1029 configuring streams: (0) 1920x1080-YUV420 (1) 2592x1944-SGBRG10_CSI2P
[16:06:41.764914315] [31741]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
Still capture image received
[16:06:42.684325325] [31740]  INFO Camera camera.cpp:1029 configuring streams: (0) 1296x728-YUV420
[16:06:42.696380586] [31741]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
[16:07:41.766498084] [31740]  INFO Camera camera.cpp:1029 configuring streams: (0) 1920x1080-YUV420 (1) 2592x1944-SGBRG10_CSI2P
[16:07:41.778117642] [31741]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
Still capture image received
[16:07:42.672106173] [31740]  INFO Camera camera.cpp:1029 configuring streams: (0) 1296x728-YUV420
[16:07:42.684000857] [31741]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
```

After setting the environment variable I'm at least down to just the single line per image capture, but that should still not output IMO. 

```
wim@WimPiZeroW-Hope:~ $ LIBCAMERA_LOG_LEVELS=*:4
wim@WimPiZeroW-Hope:~ $ export LIBCAMERA_LOG_LEVELS
wim@WimPiZeroW-Hope:~ $ libcamera-still --nopreview --thumb none --width 1920 --height 1080 --timeout 125000 --timelapse 60000 --output /media/WimPiZeroW-Hope/DCIM/20220717/0717%04d.jpg --framestart 288 --verbose 0
Still capture image received
Still capture image received
```

wcbonner, 2022-07-17:
I should mention that I've tried without the verbose option at all, with -v 0, and --verbose 0.

naushir, 2022-07-18:
> I should mention that I've tried without the verbose option at all, with -v 0, and --verbose 0.

``--verbose`` is a boolean switch, so is only used to set enable or disabled.  You can ignore the ``0`` after the command line parameter.

naushir, 2022-07-18:
> Thanks. Knowing about the environment variable reduces output significantly, but not completely. Can you point me to a description of what environment variables are recognized?

You can have a look at the libcamera documentation [here](https://git.libcamera.org/libcamera/libcamera.git/tree/Documentation/environment_variables.rst) that lists the environment variables used.

> Initially I was seeing this:
>
> ```
> wim@WimPiZeroW-Hope:~ $ libcamera-still --nopreview --thumb none --width 1920 --height 1080 --timeout 59580000 --timelapse 60000 --output [...]

I would be reluctant to remove that message as in general users would expect an indication of capture.  One suggestion would be to use invert-match grep to remove the capture line:

```
libcamera-still ... 2>&1 | grep -v "Still capture image received"...
```
ought to do the trick.

wcbonner, 2022-07-18:
> I would be reluctant to remove that message as in general users would expect an indication of capture. One suggestion would be to use invert-match grep to remove the capture line:
>
> ``` [...]

raspistill was quiet unless the --verbose option was specified. My argument is that libcamera-still should have similar lack of non-error output unless the --verbose option is specified.

I'm running this as a service and spawning an extra process of grep adds significant complications. All of this is running for a long time, and I'm trying to avoid putting unnecessary messages into the system logs.

will-v-pi, 2022-07-19:
#320 should fix this issue - the command you're running should produce no output on verbosity level 0 once that is merged