raspberrypi/rpicam-apps #324: libcamera-apps gives out error when mutter is disabled

JeffZZM, 2022-07-22:
Hi.
I'm using 2GB version of Pi 4, running on linux version 11 bullseye.
I disabled [mutter] due to it being incompatible with xrdp for remote desktop.
After disabling it, when running libcamera-hello or libcamera-vid, the following error occurs:
```
user1@raspberrypi:~ $ libcamera-hello
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[2:16:51.959344985] [17964]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033 
[2:16:51.995711522] [17977]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[2:16:51.996309405] [17977]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[2:16:52.000570507] [17964]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[2:16:52.001035798] [17977]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 19
Aborted
```
Once mutter is re-enabled the problem is fixed. 
note: the libcamera-still can work and will store a photo.

Best Regards

naushir, 2022-07-25:
You can use the ``--qt-preview`` option for the viewfinder to work around this.

JeffZZM, 2022-07-26:
> You can use the `--qt-preview` option for the viewfinder to work around this.

Thanks for your help! 
Also do you know which graphical interface that`s being used is requiring mutter to operate?

naushir, 2022-07-27:
> Also do you know which graphical interface that`s being used is requiring mutter to operate?

This is the compositor used by Wayland - I don't know of anything specific applications that requires mutter to operate.

closed 2022-07-27 by naushir