raspberrypi/rpicam-apps #325: Error 2 -wfatal-errors

victoryred, 2022-07-24:
While building the libcamera-apps, executing make -j4 runs fine until it gets to jpeg.cpp.
At line 477 in 'void create_exif_data(), I get the following error: invalid type argument of unary '*'.
the line is LOG(2, "Exposure time: "<< *exposure_time);

The compliation termines due to -wfatal-errors Error 2.
Of course an Error 1 is generated by the linker for jpeg.cpp.o (This goes away when Error 2 gets fixed).

Anything I can do to help myself?

aubourg, 2022-07-24:
Just got the error. It's from a recent commit.

git checkout 35f56d9ffd2bd2e7a2b2a15e33c07990c40dc37e

will get you back to a working source code until it gets fixed.

naushir, 2022-07-25:
Can you share your compiler version please?  This does not happen on my compilations with gcc v10.2.1 or clang 12.

naushir, 2022-07-25:
Ah, sorry I apologies - I understand what is going on.

There has been a recent API breaking change to the libcamera library, see [here](https://git.libcamera.org/libcamera/jmondi/libcamera.git/commit/?h=jmondi/pinephone&id=1c4d4801850559d6f919eef5c2ffbaf7675dbc46).  This API breakage has been fixed in libcamera-apps with 32c9ead4e9cdc12ba5e8a9b4ebbb7fdadaed726e.  If you are building libcamera-apps on or after this commit, you will need to build/install the latest code for the libcamera libraries as well, otherwise you will get the above build errors.

naushir, 2022-07-26:
Resolving now.

closed 2022-07-26 by naushir