raspberrypi/rpicam-apps #326: Make fails on GCC-12 Manjaro-Arm-Rpi aarch64 build [BUG]

m3cinc, 2022-07-25:
**Describe the bug**
Make fails to complete on GCC-12 

**Bug report**
[marc@scan3d64 build]$ uname -a
Linux scan3d64 5.15.52-1-MANJARO-ARM-RPI #1 SMP PREEMPT Tue Jul 5 12:49:52 UTC 2022 aarch64 GNU/Linux

[marc@scan3d64 build]$ cd
[marc@scan3d64 ~]$ git clone https://github.com/raspberrypi/libcamera-apps.git
Cloning into 'libcamera-apps'...
remote: Enumerating objects: 1548, done.
remote: Counting objects: 100% (774/774), done.
remote: Compressing objects: 100% (336/336), done.
remote: Total 1548 (delta 532), reused 501 (delta 438), pack-reused 774
Receiving objects: 100% (1548/1548), 465.90 KiB | 2.03 MiB/s, done.
Resolving deltas: 100% (1090/1090), done.
[marc@scan3d64 ~]$ cd libcamera-apps
[marc@scan3d64 libcamera-apps]$ mkdir build
[marc@scan3d64 libcamera-apps]$ cd build
[marc@scan3d64 build]$ cmake .. -DENABLE_DRM=1 -DENABLE_X11=1 -DENABLE_QT=1 -DENABLE_OPENCV=1 -DENABLE_TFLITE=0
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/sbin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No previous build - default to Release build
-- Platform: 
-- Found PkgConfig: /usr/sbin/pkg-config (found version "1.8.0") 
-- Checking for module 'libcamera'
--   Found libcamera, version 0.0.0
-- libcamera library found:
--     version: 0.0.0
--     libraries: /usr/lib/libcamera.so;/usr/lib/libcamera-base.so
--     include path: /usr/include/libcamera
-- Found Boost: /usr/lib/cmake/Boost-1.79.0/BoostConfig.cmake (found version "1.79.0") found components: program_options 
-- Checking for libavcodec
-- libavcodec found:
--     libraries: avcodec;avdevice;avformat;swresample
-- LIBDRM_LINK_LIBRARIES=/usr/lib/libdrm.so
-- LIBDRM display mode enabled
-- X11_LINK_LIBRARIES=/usr/lib/libX11.so
-- EPOXY_LINK_LIBRARIES=/usr/lib/libepoxy.so
-- EGL display mode enabled
-- QTCORE_LINK_LIBRARIES=/usr/lib/libQt5Core.so
-- QTCORE_INCLUDE_DIRS=/usr/include/qt/QtCore;/usr/include/qt
-- QT display mode enabled
-- Checking for OpenCV
-- OpenCV library found:
--     version: 4.6.0
--     libraries: -lopencv_core;-lopencv_imgproc;-lopencv_objdetect
--     include path: /usr/include/opencv4
-- OpenCV support is included
-- TFLite support not being included
-- Configuring done
-- Generating done
-- Build files have been written to: /home/marc/libcamera-apps/build
[marc@scan3d64 build]$ make -j1
[  2%] Building CXX object image/CMakeFiles/images.dir/bmp.cpp.o
[  4%] Building CXX object image/CMakeFiles/images.dir/yuv.cpp.o
[  6%] Building CXX object image/CMakeFiles/images.dir/jpeg.cpp.o
[  8%] Building CXX object image/CMakeFiles/images.dir/png.cpp.o
[ 10%] Building CXX object image/CMakeFiles/images.dir/dng.cpp.o
[ 12%] Linking CXX shared library libimages.so
[ 12%] Built target images
[ 14%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/post_processing_stage.cpp.o
[ 16%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/negate_stage.cpp.o
[ 18%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/hdr_stage.cpp.o
[ 20%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/pwl.cpp.o
[ 22%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/histogram.cpp.o
[ 24%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/motion_detect_stage.cpp.o
[ 26%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/sobel_cv_stage.cpp.o
[ 28%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/face_detect_cv_stage.cpp.o
[ 30%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/annotate_cv_stage.cpp.o
[ 32%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/plot_pose_cv_stage.cpp.o
[ 34%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/object_detect_draw_cv_stage.cpp.o
[ 36%] Linking CXX shared library libpost_processing_stages.so
[ 36%] Built target post_processing_stages
Generating version string: 3dcda2bacfab-intree-dirty 25-07-2022 (16:50:09)
[ 36%] Built target VersionCpp
[ 38%] Building CXX object preview/CMakeFiles/preview.dir/null_preview.cpp.o
[ 40%] Building CXX object preview/CMakeFiles/preview.dir/preview.cpp.o
[ 42%] Building CXX object preview/CMakeFiles/preview.dir/drm_preview.cpp.o
[ 44%] Building CXX object preview/CMakeFiles/preview.dir/egl_preview.cpp.o
[ 46%] Building CXX object preview/CMakeFiles/preview.dir/qt_preview.cpp.o
[ 48%] Linking CXX shared library libpreview.so
[ 48%] Built target preview
[ 50%] Building CXX object core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o
In file included from /usr/include/c++/12.1.0/functional:49,
                 from /home/marc/libcamera-apps/preview/preview.hpp:10,
                 from /home/marc/libcamera-apps/core/libcamera_app.cpp:8:
In member function ‘constexpr bool std::less<_Tp*>::operator()(_Tp*, _Tp*) const [with _Tp = CompletedRequest]’,
    inlined from ‘std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::iterator std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_lower_bound(_Link_type, _Base_ptr, const _Key&) [with _Key = CompletedRequest*; _Val = CompletedRequest*; _KeyOfValue = std::_Identity<CompletedRequest*>; _Compare = std::less<CompletedRequest*>; _Alloc = std::allocator<CompletedRequest*>]’ at /usr/include/c++/12.1.0/bits/stl_tree.h:1951:29,
    inlined from ‘std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::iterator std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::find(const _Key&) [with _Key = CompletedRequest*; _Val = CompletedRequest*; _KeyOfValue = std::_Identity<CompletedRequest*>; _Compare = std::less<CompletedRequest*>; _Alloc = std::allocator<CompletedRequest*>]’ at /usr/include/c++/12.1.0/bits/stl_tree.h:2531:36,
    inlined from ‘std::set<_Key, _Compare, _Alloc>::iterator std::set<_Key, _Compare, _Alloc>::find(const key_type&) [with _Key = CompletedRequest*; _Compare = std::less<CompletedRequest*>; _Alloc = std::allocator<CompletedRequest*>]’ at /usr/include/c++/12.1.0/bits/stl_set.h:795:25,
    inlined from ‘void LibcameraApp::queueRequest(CompletedRequest*)’ at /home/marc/libcamera-apps/core/libcamera_app.cpp:517:37:
/usr/include/c++/12.1.0/bits/stl_function.h:457:38: error: pointer ‘completed_request’ may be used after ‘void operator delete(void*, std::size_t)’ [-Werror=use-after-free]
  457 |         return (__UINTPTR_TYPE__)__x < (__UINTPTR_TYPE__)__y;
      |                ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
make[2]: *** [core/CMakeFiles/libcamera_app.dir/build.make:76: core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:262: core/CMakeFiles/libcamera_app.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
[marc@scan3d64 build]$ uname -a
Linux scan3d64 5.15.52-1-MANJARO-ARM-RPI #1 SMP PREEMPT Tue Jul 5 12:49:52 UTC 2022 aarch64 GNU/Linux

**Additional context**
libcamera was built on same platform today without error...

naushir, 2022-07-26:
@davidplowman - looking at [this](https://github.com/raspberrypi/libcamera-apps/blob/3dcda2bacfab88bd652a0b4fce7789ebb6b6dfee/core/libcamera_app.cpp#L517) code, I'm not sure what it's trying to do, and can see why the compiler might be complaining.  Any thoughts?

davidplowman, 2022-07-26:
It seems to be warning that we're using a pointer to an object after deleting it. In fact, we are just looking for the pointer and won't attempt to dereference it, but the compiler is evidently nervous. We could re-order the code a little bit and place that block further up. Obviously we wouldn't want to return from the function at that earlier location, we'd have to set a flag so that we could "erase or return" later (in fact, where it happens now).

naushir, 2022-07-26:
https://github.com/raspberrypi/libcamera-apps/pull/327 ought to solve this warning.

naushir, 2022-07-27:
Resolving...

closed 2022-07-27 by naushir