raspberrypi/rpicam-apps #328: Make fails on libav_encoder.cpp on Manjaro GCC12 [BUG]

m3cinc, 2022-07-26:
**Describe the bug**
When building on GCC12 (latest Manjaro release), an invalid conversion is reported in libav_encoder.cpp.

Perhaps a compiler flag would prevent???

**Bug report**
[marc@scan3d64 build]$ uname -a
Linux scan3d64 5.15.52-1-MANJARO-ARM-RPI #1 SMP PREEMPT Tue Jul 5 12:49:52 UTC 2022 aarch64 GNU/Linux
[marc@scan3d64 build]$ make
[ 12%] Built target images
[ 36%] Built target post_processing_stages
Generating version string: 3dcda2bacfab-intree-dirty 26-07-2022 (11:17:05)
[ 36%] Built target VersionCpp
[ 48%] Built target preview
Consolidate compiler generated dependencies of target libcamera_app
[ 50%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
[ 52%] Linking CXX shared library libcamera_app.so
[ 58%] Built target libcamera_app
[ 60%] Building CXX object encoder/CMakeFiles/encoders.dir/libav_encoder.cpp.o
/home/marc/libcamera-apps/encoder/libav_encoder.cpp: In member function ‘void LibAvEncoder::initAudioInCodec(const VideoOptions*, const StreamInfo&)’:
/home/marc/libcamera-apps/encoder/libav_encoder.cpp:132:56: error: invalid conversion from ‘const AVInputFormat*’ to ‘AVInputFormat*’ [-fpermissive]
  132 |         AVInputFormat *input_fmt = av_find_input_format("pulse");
      |                                    ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
      |                                                        |
      |                                                        const AVInputFormat*
compilation terminated due to -Wfatal-errors.
make[2]: *** [encoder/CMakeFiles/encoders.dir/build.make:132: encoder/CMakeFiles/encoders.dir/libav_encoder.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:288: encoder/CMakeFiles/encoders.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

naushir, 2022-08-02:
This should be fixed with https://github.com/raspberrypi/libcamera-apps/pull/333.

naushir, 2022-08-02:
Resolving...

closed 2022-08-02 by naushir