raspberrypi/rpicam-apps #329: Unusual that `--help` and `--version` print to stderr

weslord, 2022-07-26:
I tried to pipe the help text to less (`libcamera-vid --help | less`) so that I could scroll and search the text like I do with manpages, and was surprised & confused when the text disappeared. It took me a little while to realize the text was printing to `stderr` and not `stdout`. It looks like `--version` & `--list-cameras` do the same.  Most other command line utilities print help and version to `stdout` so it's not very intuitive, especially when you're explicitly requesting that information.

I know I can use a redirect (`libcamera-vid --help 2>&1 | less`), but it might be hard for a beginner to figure out why their pipe to grep or file redirect or shell script isn't working as expected.

naushir, 2022-07-27:
https://github.com/raspberrypi/libcamera-apps/pull/330 ought to fix this problem.

naushir, 2022-07-28:
This is now merged.

closed 2022-07-28 by naushir