raspberrypi/rpicam-apps #331: [BUG] Unable to Build libcamera-apps, error: compilation terminated due to -Wfatal-errors.

heya-naohiro, 2022-07-31:
**Describe the bug**
Unable to Build libcamera-apps, 

1. Build and Install libcamera latest 
```

 $ git clone git://linuxtv.org/libcamera.git
 $ git log -1
commit 69ae75b0cc211f82665b3e92fb3de64a9852b403 (HEAD -> master, origin/master, origin/HEAD)
 $ meson build --buildtype=release -Dpipelines=raspberrypi -Dipas=raspberrypi -Dv4l2=true -Dgstreamer=disabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=enabled -Ddocumentation=disabled
The Meson build system
Version: 0.63.0
Source dir: /home/heyanaohiro/libcamera
Build dir: /home/heyanaohiro/libcamera/build
Build type: native build
Project name: libcamera
Project version: 0.0.0
C compiler for the host machine: cc (gcc 10.2.1 "cc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
C linker for the host machine: cc ld.bfd 2.35.2
C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
C++ linker for the host machine: c++ ld.bfd 2.35.2
Host machine cpu family: arm
Host machine cpu: armv7l
Header "unistd.h" has symbol "issetugid" : NO 
Header "stdlib.h" has symbol "secure_getenv" : YES 
Compiler for C supports arguments -Wno-c99-designator: NO 
Library lttng-ust found: NO
...
Found ninja-1.10.1 at /usr/bin/ninja
$ ninja -C build   # use -j 2 on Raspberry Pi 3 or earlier devices
$ sudo ninja -C build install
```
2. Build  libcamera-app latest 
```
$ git log -1
commit 67d4e9294d61233fcdc882a6dc6e788ee97c5b10 (HEAD -> main, origin/main, origin/HEAD)
$ cmake .. -DENABLE_DRM=1 -DENABLE_X11=1 -DENABLE_QT=1 -DENABLE_OPENCV=0 -DENABLE_TFLITE=1
$ make -j4 
-- Platform: armhf
-- libcamera library found:
--     version: 0.0.0
--     libraries: /usr/local/lib/arm-linux-gnueabihf/libcamera.so;/usr/local/lib/arm-linux-gnueabihf/libcamera-base.so
--     include path: /usr/local/include/libcamera
-- Checking for libavcodec
-- LIBDRM_LINK_LIBRARIES=/usr/lib/arm-linux-gnueabihf/libdrm.so
-- LIBDRM display mode enabled
-- EGL display mode will be unavailable!
-- QTCORE_LINK_LIBRARIES=/usr/lib/arm-linux-gnueabihf/libQt5Core.so
-- QTCORE_INCLUDE_DIRS=/usr/include/arm-linux-gnueabihf/qt5/QtCore;/usr/include/arm-linux-gnueabihf/qt5
-- QT display mode enabled
-- Omitting check for OpenCV
-- OpenCV support not being included
-- Adding TFLite support
-- Configuring done
-- Generating done
-- Build files have been written to: /home/heyanaohiro/libcamera-apps/build
[  2%] Building CXX object image/CMakeFiles/images.dir/bmp.cpp.o
[  4%] Building CXX object output/CMakeFiles/outputs.dir/output.cpp.o
[  6%] Building CXX object preview/CMakeFiles/preview.dir/null_preview.cpp.o
Generating version string: 67d4e9294d61-intree 31-07-2022 (11:01:01)
[  6%] Built target VersionCpp
[  8%] Building CXX object image/CMakeFiles/images.dir/yuv.cpp.o
[ 12%] Building CXX object preview/CMakeFiles/preview.dir/preview.cpp.o
[ 12%] Building CXX object image/CMakeFiles/images.dir/jpeg.cpp.o
[ 14%] Building CXX object image/CMakeFiles/images.dir/png.cpp.o
[ 16%] Building CXX object output/CMakeFiles/outputs.dir/file_output.cpp.o
[ 18%] Building CXX object output/CMakeFiles/outputs.dir/net_output.cpp.o
[ 20%] Building CXX object output/CMakeFiles/outputs.dir/circular_output.cpp.o
[ 22%] Linking CXX shared library liboutputs.so
[ 22%] Built target outputs
[ 24%] Building CXX object image/CMakeFiles/images.dir/dng.cpp.o
/home/heyanaohiro/libcamera-apps/image/dng.cpp: In function ‘void dng_save(const std::vector<libcamera::Span<unsigned char> >&, const StreamInfo&, const libcamera::ControlList&, const string&, const string&, const StillOptions*)’:
/home/heyanaohiro/libcamera-apps/image/dng.cpp:154:6: error: could not convert ‘bl’ from ‘libcamera::Span<const int>’ to ‘boo’
  154 |  if (bl)
      |      ^~
      |      |
      |      libcamera::Span<const int>
compilation terminated due to -Wfatal-errors.
make[2]: *** [image/CMakeFiles/images.dir/build.make:134: image/CMakeFiles/images.dir/dng.cpp.o] エラー 1
make[2]: *** 未完了のジョブを待っています....
[ 26%] Building CXX object preview/CMakeFiles/preview.dir/drm_preview.cpp.o
[ 28%] Building CXX object preview/CMakeFiles/preview.dir/qt_preview.cpp.o
In file included from /home/heyanaohiro/libcamera-apps/core/options.hpp:21,
                 from /home/heyanaohiro/libcamera-apps/core/still_options.hpp:12,
                 from /home/heyanaohiro/libcamera-apps/image/jpeg.cpp:23:
/home/heyanaohiro/libcamera-apps/image/jpeg.cpp: In function ‘void create_exif_data(const std::vector<libcamera::Span<unsigned char> >&, const StreamInfo&, const libcamera::ControlList&, const string&, const StillOptions*, uint8_t*&, unsigned int&, uint8_t*&, jpeg_mem_len_t&)’:
/home/heyanaohiro/libcamera-apps/image/jpeg.cpp:477:32: error: invalid type argument of unary ‘*’ (have ‘int’)
  477 |    LOG(2, "Exposure time: " << *exposure_time);
      |                                ^~~~~~~~~~~~~~
/home/heyanaohiro/libcamera-apps/core/logging.hpp:7:17: note: in definition of macro ‘LOG’
    7 |    std::cerr << text << std::endl;                                                                            \
      |                 ^~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [image/CMakeFiles/images.dir/build.make:108: image/CMakeFiles/images.dir/jpeg.cpp.o] エラー 1
[ 30%] Linking CXX shared library libpreview.so
[ 30%] Built target preview
make[1]: *** [CMakeFiles/Makefile2:355: image/CMakeFiles/images.dir/all] エラー 2
make: *** [Makefile:171: all] エラー 2
```

**Additional context**
 It appears to be related to the issue( https://github.com/raspberrypi/libcamera-apps/issues/325 ).  I am using the latest version of both libcamera and libcameraapp, but I am still getting this error. 　Do you have any advice?

naushir, 2022-08-01:
It seems as if the libcamera-apps build is not finding the latest libcamera header files.

You need to make sure you install the libcamera library before you build libcamera apps.  I think you might already be doing this.  However, after doing that, try running ``sudo ldconfg`` to ensure the new headers are correctly cached. Then try a clean rebuild of libcamera-apps.

eli-schwartz, 2022-08-01:
ldconfig only rebuilds the index of shared libraries, it has absolutely nothing to do with headers...

(And anyway that's a matter for the runtime loader, not the build-time linker.)

heya-naohiro, 2022-08-01:
I now understand what I did wrong: I had previously tried the procedure to build libcamera-app without building libcamera, so the libcamera-dev installed with apt was left in the environment.
```
sudo apt purge libcamera-dev libepoxy-dev libjpeg-dev libtiff5-dev
```
and reinstalled libcamera from source and it worked. Thanks for the reply.

closed 2022-08-01 by heya-naohiro