raspberrypi/rpicam-apps #332: failing to compile from source

chino, 2022-08-02:
On a rpi3 attempting to build:

```
cmake ..   -DENABLE_COMPILE_FLAGS_FOR_TARGET=armv8-neon -DENABLE_X11=1  -DENABLE_OPENCV=1 -DENABLE_TFLITE=1  && make -j2 && sudo make install && sudo ldconfig

...

In file included from /home/dan/code/libcamera/libcamera-apps/core/options.hpp:21,
                 from /home/dan/code/libcamera/libcamera-apps/core/still_options.hpp:12,
                 from /home/dan/code/libcamera/libcamera-apps/image/jpeg.cpp:23:
/home/dan/code/libcamera/libcamera-apps/image/jpeg.cpp: In function ‘void create_exif_data(const std::vector<libcamera::Span<unsigned char> >&, const StreamInfo&, const libcamera::ControlList&, const string&, const StillOptions*, uint8_t*&, unsigned int&, uint8_t*&, jpeg_mem_len_t&)’:
/home/dan/code/libcamera/libcamera-apps/image/jpeg.cpp:477:32: error: invalid type argument of unary ‘*’ (have ‘int’)
  477 |    LOG(2, "Exposure time: " << *exposure_time);
      |                                ^~~~~~~~~~~~~~
/home/dan/code/libcamera/libcamera-apps/core/logging.hpp:7:17: note: in definition of macro ‘LOG’
    7 |    std::cerr << text << std::endl;                                                                            \
      |                 ^~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [image/CMakeFiles/images.dir/build.make:108: image/CMakeFiles/images.dir/jpeg.cpp.o] Error 1

```

naushir, 2022-08-02:
Please see https://github.com/raspberrypi/libcamera-apps/issues/331 and https://github.com/raspberrypi/libcamera-apps/issues/325.

You will need to update the libcamera libraries as there is a core API change.

closed 2022-08-02 by naushir

chino, 2022-08-02:
I did build/install libcamera as well, removed the apt packages, and ran ldconfig, and had an empty build directory before building.  Maybe I got something wrong along the way but I will try again.

I was a bit taken by the fact that there is no tags for stable version matching at all?

Why not include the OpenCV and Tensor support in the pre-built packages so that users can play around with it?

Also while playing with `Motion` (https://github.com/Motion-Project/motion/issues/1434) I did get it to work but for some reason it was always failing with `v4l2_fps_set: Error setting fps` even when I was trying valid settings from `--list-cameras` so I ended up having to revert back to the legacy picam support for now.

naushir, 2022-08-02:
> I was a bit taken by the fact that there is no tags for stable version matching at all?

libcamera does not yet have a stable API, so there are no version tags just yet.  It will be done in the future.

> 
> Why not include the OpenCV and Tensor support in the pre-built packages so that users can play around with it?

These libraries add substantial bloat to the images, so we have decided to not include them in our standard releases.

> 
> Also while playing with `Motion` ([Motion-Project/motion#1434](https://github.com/Motion-Project/motion/issues/1434)) I did get it to work on a lower resolution but for some reason it was always failing on the higher modes (with `v4l2_fps_set: Error setting fps`) even when I was trying valid settings from `--list-cameras` so I ended up having to revert back to the legacy picam support for now.

I am not familiar with the Motion, but I think they use the v4l2 compatibility layer of libcamera, and not the native API.  This is a work in progress as far as I know, so my not be fully functional yet.