raspberrypi/rpicam-apps #334: [BUG]

merlincom, 2022-08-02:
libcamera-hello -t 0 --qt-preview fails in Combination with putty SSH X11 forwarding an Windows10/XMing
``
Made X/EGL preview window
[0:02:02.795002742] [1595]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:02:02.827159993] [1608]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[0:02:02.827974952] [1608]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media2
[0:02:02.829004226] [1595]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[0:02:02.829444372] [1608]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 19
Abgebrochen
`
Logfile XMing:
`Welcome to the Xming X Server
Vendor: Colin Harrison
Release: 6.9.0.31
FreeType2: 2.3.4
Contact: http://sourceforge.net/forum/?group_id=156984

C:\Program Files (x86)\Xming\Xming.exe :0 -clipboard 
 -multiwindow

XdmcpRegisterConnection: newAddress 192.168.1.23
winAdjustVideoModeShadowGDI - Using Windows display depth of 32 bits per pixel
winAllocateFBShadowGDI - Creating DIB with width: 1920 height: 2280 depth: 32
winAllocateFBShadowGDI - Requested DIB (bitmap) will be larger than 16 MB.  The surface may fail to be allocated on Windows 95, 98, or Me, due to a 16 MB limit in DIB size.  This limit does not apply to Windows NT/2000, and this message may be ignored on those platforms.
winInitVisualsShadowGDI - Masks 00ff0000 0000ff00 000000ff BPRGB 8 d 24 bpp 32
glWinInitVisuals:1596: glWinInitVisuals
glWinInitVisualConfigs:1503: glWinInitVisualConfigs glWinSetVisualConfigs:1581: glWinSetVisualConfigs
init_visuals:1055: init_visuals
null screen fn ReparentWindow
null screen fn RestackWindow
InitQueue - Calling pthread_mutex_init
InitQueue - pthread_mutex_init returned
InitQueue - Calling pthread_cond_init
InitQueue - pthread_cond_init returned
winInitMultiWindowWM - Hello
winInitMultiWindowWM - Calling pthread_mutex_lock ()
winMultiWindowXMsgProc - Hello
winMultiWindowXMsgProc - Calling pthread_mutex_lock ()
glWinScreenProbe:1390: glWinScreenProbe
fixup_visuals:1303: fixup_visuals
init_screen_visuals:1336: init_screen_visuals
(--) 5 mouse buttons found
(--) Setting autorepeat to delay=500, rate=31
(--) winConfigKeyboard - Layout: "00000807" (00000807) 
(--) Using preset keyboard for "German (Switzerland)" (807), type "4"
Could not init font path element C:\Program Files (x86)\Xming/fonts/Type1/, removing from list!
Could not init font path element C:\Program Files (x86)\Xming/fonts/100dpi/, removing from list!
Could not init font path element C:\Program Files\Xming\fonts\dejavu, removing from list!
Could not init font path element C:\Program Files\Xming\fonts\cyrillic, removing from list!
Could not init font path element C:\WINDOWS\Fonts, removing from list!
winInitMultiWindowWM - pthread_mutex_lock () returned.
winInitMultiWindowWM - pthread_mutex_unlock () returned.
winInitMultiWindowWM - DISPLAY=127.0.0.1:0.0
winMultiWindowXMsgProc - pthread_mutex_lock () returned.
winMultiWindowXMsgProc - pthread_mutex_unlock () returned.
winMultiWindowXMsgProc - DISPLAY=127.0.0.1:0.0
winProcEstablishConnection - Hello
winInitClipboard ()
winProcEstablishConnection - winInitClipboard returned.
winClipboardProc - Hello
DetectUnicodeSupport - Windows Vista
winClipboardProc - DISPLAY=127.0.0.1:0.0
winInitMultiWindowWM - XOpenDisplay () returned and successfully opened the display.
winMultiWindowXMsgProc - XOpenDisplay () returned and successfully opened the display.
winClipboardProc - XOpenDisplay () returned and successfully opened the display.`

naushir, 2022-08-02:
> ```
> Made X/EGL preview window
> [0:02:02.795002742] [1595]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
> [0:02:02.827159993] [1608]  WARN RPI raspberrypi.cpp:1252 Mismatch between [...]

The first line in this log (``Made X/EGL preview window``) suggests that ``--qt-preview`` is not used.  You should see ``Made QT preview window`` if that argument was set.  Can you confirm your entire command line again?

naushir, 2022-08-02:
Can you also provide the output file generated from the ``camera-bug-report`` command?

merlincom, 2022-08-02:
[--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 02-08-2022 (15:37:55)
Command: /usr/bin/camera-bug-report -t 10000 -c libcamera-hello -t 5000

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
mypi

** cat /proc/cpuinfo **
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : c03115
Serial          : 100000005ae869d0
Model           : Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=443b34cb-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
](url)

merlincom, 2022-08-02:
> Can you also provide the output file generated from the `camera-bug-report` command?

I do not now, why it is working now (Windows 10 > XMing > Putty > X11 forwarding > on both Raspberry 4, 

1. with Pi Camera Module 2 
2. with Pi Noir  Camera Module V2
both with Bullseye Release 11 64 bit

I did'n change any  system.

```
libcamera-hello -t 0 --qt-preview
qt.qpa.xcb: X server does not support XInput 2
failed to get the current screen resources
Made QT preview window
qt.qpa.xcb: QXcbConnection: XCB error: 1 (BadRequest), sequence: 165, resource id: 90, major code: 130 (Unknown), minor code: 47
qt.qpa.xcb: QXcbConnection: XCB error: 170 (Unknown), sequence: 180, resource id: 90, major code: 146 (Unknown), minor code: 20
[3:48:11.433984782] [3274]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[3:48:11.470161725] [3290]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[3:48:11.470985666] [3290]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[3:48:11.471684627] [3274]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[3:48:11.472124440] [3290]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
```

naushir, 2022-08-02:
Good to hear it works.  It was probably something to do with the command line argument, either missing or malformed.

Feel free to close this issue if you are happy that it is resolved.

anfederman, 2022-08-02:
Hi all, I am trying to get a picam v2 to work on ARM64 Ubuntu 22.04 on a RPi 4.  in config.txt ;
camera_auto_detect=1
 
It seems to see the camera,knows that it is unicam, sees the formats,but won't stream

I am guessing a kernel issue, but I am at a loss to figure out the problem.

I am not even sure this is the same issue, or a new one.

uname -a
Linux ubiq22-desktop 5.15.0-1012-raspi #14-Ubuntu SMP PREEMPT Fri Jun 24 13:10:28 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

libcamera-hello
Preview window unavailable
[0:03:14.743136284] [2296]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3777-69ae75b0
[0:03:14.797570137] [2303]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[0:03:14.798371847] [2303]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[0:03:14.799573737] [2296]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[0:03:14.800187116] [2303]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:03:15.838061208] [2303] ERROR V4L2 v4l2_videodevice.cpp:1867 /dev/video0[12:cap]: Failed to start streaming: Input/output error
ERROR: *** failed to start camera ***

vcgencmd get_camera
supported=1 detected=0   (if #camera_auto_detect=1) it will get detected

v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media3

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media1
	/dev/media2

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/video1
	/dev/media0

ubiq22@ubiq22-desktop:~$ v4l2-compliance
v4l2-compliance 1.22.1, 64 bits, 64-bit time_t

Compliance test for unicam device /dev/video0:

Driver Info:
	Driver name      : unicam
	Card type        : unicam
	Bus info         : platform:fe801000.csi
	Driver version   : 5.15.39
	Capabilities     : 0xa5a00001
		Video Capture
		Metadata Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x25200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : unicam
	Model            : unicam
	Serial           : 
	Bus info         : platform:fe801000.csi
	Media version    : 5.15.39
	Hardware revision: 0x00000000 (0)
	Driver version   : 5.15.39
Interface Info:
	ID               : 0x03000006
	Type             : V4L Video
Entity Info:
	ID               : 0x00000004 (4)
	Name             : unicam-image
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000005   : 0: Sink
	  Link 0x02000008: from remote pad 0x1000002 of entity 'imx219 10-0010' (Camera Sensor): Data, Enabled, Immutable

Required ioctls:
	test MC information (see 'Media Driver Info' above): OK
	test VIDIOC_QUERYCAP: OK
	test invalid ioctls: OK

Allow for multiple opens:
	test second /dev/video0 open: OK
	test VIDIOC_QUERYCAP: OK
	test VIDIOC_G/S_PRIORITY: OK
	test for unlimited opens: OK

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 1 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 0 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Control ioctls (Input 0):
	test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK (Not Supported)
	test VIDIOC_QUERYCTRL: OK (Not Supported)
	test VIDIOC_G/S_CTRL: OK (Not Supported)
	test VIDIOC_G/S/TRY_EXT_CTRLS: OK (Not Supported)
	test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK (Not Supported)
	test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
	Standard Controls: 0 Private Controls: 0

Format ioctls (Input 0):
	test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK
	test VIDIOC_G/S_PARM: OK (Not Supported)
	test VIDIOC_G_FBUF: OK (Not Supported)
	test VIDIOC_G_FMT: OK
	test VIDIOC_TRY_FMT: OK
	test VIDIOC_S_FMT: OK
	test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
	test Cropping: OK (Not Supported)
	test Composing: OK (Not Supported)
	test Scaling: OK

Codec ioctls (Input 0):
	test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
	test VIDIOC_G_ENC_INDEX: OK (Not Supported)
	test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

Buffer ioctls (Input 0):
	test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: OK
	test VIDIOC_EXPBUF: OK
	test Requests: OK (Not Supported)

Total for unicam device /dev/video0: 46, Succeeded: 46, Failed: 0, Warnings: 0

merlincom, 2022-08-03:
I cannot reproduce the error even after 24 hours. Thanks for the helpe

closed 2022-08-03 by merlincom

anfederman, 2022-08-03:
Thanks, solved issue, was hardware on my end, ribbon cable was too long,
generated i2c timeouts. Short ribbon fixed.