raspberrypi/rpicam-apps #338: Can't build with the Tensorflow Lite on Raspberry pi 4B, 64 bit

DevelopDaily, 2022-08-07:
I am aware of the solution described in the issue #132.

But my problem is that I use `Pi OS Bullseye (64-bit)`, the prebuilt lib does not seem to be compatible. I read the builder's website, which states " Binaries are compatible with Raspberry `Pi OS Bullseye (32-bit)`."

Does that mean the prebuilt lib for 64-bit does not exist anywhere?

Anyway, I tried to build the latest Tensorflow Lite on the 64-bit and the build succeeded. The trouble is that the `libcamera-apps` cannot link with it because of these errors:

```
/usr/bin/ld: /usr/local/lib/libtensorflow-lite.so: undefined reference to `tensorflow::profiler::internal::g_trace_level'
/usr/bin/ld: /usr/local/lib/libtensorflow-lite.so: undefined reference to `tensorflow::profiler::TraceMeRecorder::Record(tensorflow::profiler::TraceMeRecorder::Event&&)'
/usr/bin/ld: /usr/local/lib/libtensorflow-lite.so: undefined reference to `tensorflow::profiler::GetCurrentTimeNanos()'
/usr/bin/ld: /usr/local/lib/libtensorflow-lite.so: undefined reference to `tensorflow::profiler::ScopedMemoryDebugAnnotation::ThreadMemoryDebugAnnotation()'

```
Could you please offer some suggestions?

Thanks.

naushir, 2022-08-08:
I'm afraid we only support TF on our 32-bit releases for this very reason.  There has always been a problem building/compiling TFlite in our 64-bit environment.

I don't really have any suggestions as I've not looked into this for a long time, but you might want to revert to a 32-bit build for now if you want to use this module.

DevelopDaily, 2022-08-11:
Thanks for the comments. I will keep this issue open for a while because I am trying to convince the Tensorflow people to fix their problems. This is not a `libcamera` problem.

I fixed it on my local machine. Basically, those "profiler" references, which reside on the tensorflow main branch, should never be referenced inside the TF Lite. Unfortunately, they are referenced, but the [CMakeLists.txt](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/CMakeLists.txt) of the Lite has never populated any of the code from the main branch. No wonder we see `undefined reference`.

The tricky part is that I am tempted to use `populate_tf_source_vars` to bring some code to the Lite. But, that opens a nasty can of worms because the references will escalate to reference a lot of other main branch stuff. That defeats the very motivation of the "Lite" version of Tensorflow, doesn't it?

DevelopDaily, 2022-09-02:
I can close it. The issue was filed with Tensorflow. I am not sure how and when they are going to fix it.

closed 2022-09-02 by DevelopDaily