raspberrypi/rpicam-apps #341: Unable to build libcamera-apps

dantchapman, 2022-08-10:
I have build libcamera from source as described https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-and-libcamera-apps

This is from a clean 32 bit Raspian-Lite image

When I try to build libcamera-apps I have the following error:
[ 69%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
In file included from /usr/local/include/libcamera/libcamera/control_ids.h:15,
                 from /home/cam/libcamera-apps/core/frame_info.hpp:12,
                 from /home/cam/libcamera-apps/core/libcamera_app.cpp:10:
/usr/local/include/libcamera/libcamera/controls.h: In instantiation of ‘void libcamera::ControlList::set(const libcamera::Control<T>&, const std::initializer_list<_Up>&) [with T = libcamera::Span<const long long int, 2>; V = long long int]’:
/home/cam/libcamera-apps/core/libcamera_app.cpp:411:86:   required from here
/usr/local/include/libcamera/libcamera/controls.h:403:14: error: no matching function for call to ‘libcamera::ControlValue::set<libcamera::Span<const long long int, 2> >(libcamera::Span<const long long int, 4294967295>)’
  403 |   val->set<T>(Span<const typename std::remove_cv_t<V>>{ value.begin(), value.size() });
      |   ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [core/CMakeFiles/libcamera_app.dir/build.make:82: core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:272: core/CMakeFiles/libcamera_app.dir/all] Error 2
make: *** [Makefile:171: all] Error 2

naushir, 2022-08-10:
See https://github.com/raspberrypi/libcamera-apps/issues/332.

This is due to a libcamera API change. You will have to update your libcamera libraries from source if you are building the latest version of libcamera-apps.

naushir, 2022-08-10:
Seems there's another API breakage in libcamera (see https://github.com/raspberrypi/libcamera-apps/issues/332) that we need to fixup in libcamera-apps.

dantchapman, 2022-08-10:
> This ought to be fixed with #342

I ran a build from your fork and it is now building. Looking forward to the merge so I can get back to my fork. Thanks for the (extremely!) quick response!

naushir, 2022-08-10:
Now merged...

closed 2022-08-10 by naushir