raspberrypi/rpicam-apps #344: Unable to do long exposures with IMX290

jayjar291, 2022-08-16:
When attempting to get a long exposure with ``libcamera-still -r --immediate --shutter 1000000 -o test.jpg --gain 1 --framerate 1 --awbgains 1,1`` ``exiftool`` reports that the exposure time is 1/60 instead of 1.

I have tried the following commands
``libcamera-still -r --immediate --shutter 1000000 -o test.jpg --gain 1 --framerate 1 --awbgains 1,1``
``libcamera-still -r --immediate --shutter 1000000 -o test.jpg --gain 1 --framerate 1``
``libcamera-still -r --immediate --shutter 1000000 -o test.jpg --gain 1 --framerate 1 --awbgains 1,1 --tuning-file imx290.json``
``libcamera-still -r --immediate --shutter 1000000 -o test.jpg --gain 1 --framerate 1 --tuning-file imx290.json ``

For the commands without the ``--tuning-file imx290.json`` exiftool reports an exposure time of 1/4 or 1/60 instead of the expected 1s.

For the commands with the ``--tuning-file imx290.json`` I just get a segmentation fault.

I also get a segmentation fault when attempting to run ``libcamera-hello --tuning-file imx290.json``.

The only changes to the tuning file are as follows:
```
"exposure_modes":
{
    "normal":
    {
        "shutter": [ 10, 30000, 60000, 1000000 ],
        "gain": [ 1.0, 2.0, 8.0, 8.0 ]
    },
    "sport":
    {
        "shutter": [ 10, 5000, 10000, 20000, 120000, 1000000 ],
        "gain": [ 1.0, 2.0, 4.0, 6.0, 8.0, 8.0 ]
    }
},
```

I have attached the bug reports for both commands and a GDB session for ``libcamera-hello``

[imx290.json.txt](https://github.com/raspberrypi/libcamera-apps/files/9355287/imx290.json.txt)
[IMX290_long.txt](https://github.com/raspberrypi/libcamera-apps/files/9355288/IMX290_long.txt)
[IMX290_long_with_tuning.txt](https://github.com/raspberrypi/libcamera-apps/files/9355289/IMX290_long_with_tuning.txt)
[image]
[image]

My hardware is as follows

Raspberry Pi 4 4gb
[Waveshare IMX290](https://www.waveshare.com/imx290-83-ir-cut-camera.htm)
OS versions:
I have performed the following commands to update/upgrade my system
```
apt update
apt upgrade
apt full-upgrade
rpi-update
```
[image]

6by9, 2022-08-17:
General comment with imx290/327/462, they support either a 37.125MHz or 72.25MHz reference clock, and different modules use the different frequencies. Get it wrong and framerate and exposure time corrections will be half that requested (select 72.25 when actually 37.125), or it will fail to stream (select 37.125 when actually 72.25),
The Waveshare wiki does appear to cover this by telling you to use ```dtoverlay=imx290,clock-frequency=37125000``` in config.txt. 

IMX290 is a little strange in that the datasheet tells you to change HBLANK to vary frame rate, rather than the more common VBLANK. Just changing VBLANK does work, but limits the maximum frame or exposure time.
You can manually increase HBLANK using ```v4l2-ctl -d /dev/v4l-subdev0 --set-ctrl=horizontal-blanking=X```. ```v4l2-ctl -d /dev/v4l-subdev0 --list-ctrls``` will tell you the range for each control. libcamera should automatically pick up the change in HBLANK and amend all settings to match (assuming they're in range, ie don't set HBLANK to the max, and then try getting high frame rates)

anjok, 2022-08-17:
Thanks for the explanation, but I'm not sure I follow. When exactly would you set `horizontal_blanking` and to what? The range is: `horizontal_blanking 0x009e0902 (int)    : min=280 max=63615 step=1 default=280 value=280`, the current value is 280 (during daytime, will need to put it somewhere dark).

After the start of a capture? Before? Is it persistent?

naushir, 2022-09-05:
> Thanks for the explanation, but I'm not sure I follow. When exactly would you set `horizontal_blanking` and to what? The range is: `horizontal_blanking 0x009e0902 (int) : min=280 max=63615 step=1 [...]

You can set the hblank control before starting libcamera-still. The control value is persistent, and libcamera should correctly use the value you set. The command to do this is given by @6by9 in his comment above.

6by9, 2022-09-05:
Sorry, I'd missed the followup comment.

Just testing (actually using an IMX462, but it's just the newer version of the same family). Maxing out vertical_blanking with the default horizontal_blanking appears to give 2.75fps or 364ms/frame.

Using ```v4l2-ctl -d /dev/v4l-subdev0 --set-ctrl=horizontal_blanking=64255``` to increase horizontal blanking to the max gives 0.14fps - 7.23secs/frame.

I'm testing with v4l2-ctl. Annoyingly the driver resets HBLANK and VBLANK when changing modes, so starting libcamera triggers a reset. It shouldn't really, it should only sanitise the settings so that it remains in range. I'll fix that one up now.

And actually there is a second bug in the driver. Changing VBLANK should change the range possible for exposure, however the code to do that is incorrectly after a conditional that ensures the sensor is streaming. Changing VBLANK in any state should update the exposure range.

I'll sort out a PR.

6by9, 2022-09-05:
PR opened with the fixes for IMX290 - https://github.com/raspberrypi/linux/pull/5164

naushir, 2022-09-06:
https://github.com/raspberrypi/linux/pull/5164 has now been merged, so with the ``v4l2-ctl -d /dev/v4l-subdev0 --set-ctrl=horizontal_blanking=64255`` command, you ought to get the longer exposures now.

closed 2022-09-06 by naushir