raspberrypi/rpicam-apps #345: Saving metadata with libcamera-vid

ptoews, 2022-08-23:
Is it possible to save the metadata (exposure time, gain etc. and ideally the results of the algorithms like lux level, color temperature etc.) for each frame recorded by libcamera-vid, similar to ``--save-pts``? I know that there is already an option `--metadata` for libcamera-still (#307).

naushir, 2022-08-30:
Unfortunately there is no such option at this time.  When we get a chance, we will look to add it.

naushir, 2022-09-05:
Not planned yet, but we will put this feature on our backlog.

davidplowman, 2022-09-07:
Actually William has just a little time before he leaves so we may have a quick look into this!

@will-v-pi FYI

will-v-pi, 2022-09-07:
#358 should add this option, allowing you to output the metadata to either a json file, or in plain text form like in libcamera-still.

ptoews, 2022-09-07:
Wow that's great (and quick), thank you so much!

naushir, 2022-09-15:
This has now been merged.

closed 2022-09-15 by naushir