raspberrypi/rpicam-apps #346: [BUG] Invalid metering mode (custom)

ptoews, 2022-08-23:
I'm trying to use a custom metering mode for AGC/AEC and I have defined one in a tuning file, that is based on the default one. However, when trying to use this mode (while also specifying this tuning file), I get an error

```
ERROR: *** Invalid metering mode: dashcam ***
```

At first I thought it didn't recognize/find this mode, but the error for non-existing modes is different:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Agc: no metering mode custom
Aborted
```

Here is the result of the `camera-bug-report` tool:
[debug.txt](https://github.com/raspberrypi/libcamera-apps/files/9402812/debug.txt)

Here is the used tuning file (renamed file ending because of upload restrictions):
[imx477_modified.txt](https://github.com/raspberrypi/libcamera-apps/files/9402824/imx477_modified.txt)

davidplowman, 2022-08-24:
Hi, libcamera doesn't actually use strings to identify different modes, it creates explicit enumerations which we have match. As far as I can see, "custom" should work.You'll need to call it "custom" in the tuning file and use "custom" on the command line. What happens if you try that?

ptoews, 2022-08-25:
Yes, that works. Thank you!
It was just a misunderstanding of the documentation then. Reading it again it seems more obvious now.

closed 2022-08-25 by ptoews