raspberrypi/rpicam-apps #349: --gain ISO variable

Bra1nsen, 2022-08-26:
I thought I would be able to set ISO with --gain according to the documentation.

libcamera-jpeg -o vgl2.jpg --shutter 133 --gain 0 -n 

```
pi@allskyLA:~ $ exiftool vgl2.jpg
ExifTool Version Number         : 12.16
File Name                       : vgl2.jpg
Directory                       : .
File Size                       : 2.5 MiB
File Modification Date/Time     : 2022:08:26 18:55:07+02:00
File Access Date/Time           : 2022:08:26 18:55:07+02:00
File Inode Change Date/Time     : 2022:08:26 18:55:07+02:00
File Permissions                : rw-r--r--
File Type                       : JPEG
File Type Extension             : jpg
MIME Type                       : image/jpeg
Exif Byte Order                 : Little-endian (Intel, II)
Make                            : Raspberry Pi
Camera Model Name               : /base/soc/i2c0mux/i2c@1/imx477@1a
Software                        : libcamera-still
Modify Date                     : 2022:08:26 18:55:07
Exposure Time                   : 1/8772
ISO                             : 931
Date/Time Original              : 2022:08:26 18:55:07
Create Date                     : 2022:08:26 18:55:07
Compression                     : JPEG (old-style)
Thumbnail Offset                : 336
Thumbnail Length                : 7442
Image Width                     : 4056
Image Height                    : 3040
Encoding Process                : Baseline DCT, Huffman coding
Bits Per Sample                 : 8
Color Components                : 3
Y Cb Cr Sub Sampling            : YCbCr4:2:0 (2 2)
Image Size                      : 4056x3040
Megapixels                      : 12.3
Shutter Speed                   : 1/8772
Thumbnail Image                 : (Binary data 7442 bytes, use -b option to extract)
```

pi@allskyLA:~ $ libcamera-jpeg -o vgl22.jpg --shutter 143 --gain 0 -n

```
pi@allskyLA:~ $ exiftool vgl22.jpg
ExifTool Version Number         : 12.16
File Name                       : vgl22.jpg
Directory                       : .
File Size                       : 2.3 MiB
File Modification Date/Time     : 2022:08:26 18:56:34+02:00
File Access Date/Time           : 2022:08:26 18:56:34+02:00
File Inode Change Date/Time     : 2022:08:26 18:56:34+02:00
File Permissions                : rw-r--r--
File Type                       : JPEG
File Type Extension             : jpg
MIME Type                       : image/jpeg
Exif Byte Order                 : Little-endian (Intel, II)
Make                            : Raspberry Pi
Camera Model Name               : /base/soc/i2c0mux/i2c@1/imx477@1a
Software                        : libcamera-still
Modify Date                     : 2022:08:26 18:56:34
Exposure Time                   : 1/7042
ISO                             : 800
Date/Time Original              : 2022:08:26 18:56:34
Create Date                     : 2022:08:26 18:56:34
Compression                     : JPEG (old-style)
Thumbnail Offset                : 336
Thumbnail Length                : 7291
Image Width                     : 4056
Image Height                    : 3040
Encoding Process                : Baseline DCT, Huffman coding
Bits Per Sample                 : 8
Color Components                : 3
Y Cb Cr Sub Sampling            : YCbCr4:2:0 (2 2)
Image Size                      : 4056x3040
Megapixels                      : 12.3
Shutter Speed                   : 1/7042
Thumbnail Image                 : (Binary data 7291 bytes, use -b option to extract)
```

how can I set specific ISO values?

thank you guys

Bra1nsen, 2022-08-26:
[image]

python:

```
with picamera2.Picamera2() as camera:
   
        camera.resolution = (4056,3040)
        #camera.framerate = Fraction(1, 2)
        camera.iso = 100
        camera.exposure_mode = 'off'
        camera.awb_mode = 'off'
        camera.awb_gains = (1.8,1.8)
        
        config = camera.create_still_configuration()
        camera.configure(config)
 
     
		
        camera.set_controls({"ExposureTime": 150, "AnalogueGain": 1})
        camera.start()
        camera.capture_file('ttx0.jpg')
        print(camera.capture_metadata())
		
        camera.stop()
```

libapp

`libcamera-jpeg --shutter 150 --gain 0 -o schenll.jpg`

naushir, 2022-08-30:
The ``--gain`` argument specifies the analogue system gain to use for the capture.  You can deduce ISO from that by multiplying the value by 100.  So ``--gain 1`` would give an ISO equivalent of 100.  Specifying a value of 0 is incorrect for this argument.

naushir, 2022-09-02:
Is there anything more to be done for this issue?  If not, I'll close this soon.

Bra1nsen, 2022-09-02:
sorry ! youre absolut right, forgot to answer appropriately. thank you naushir

closed 2022-09-02 by Bra1nsen