raspberrypi/rpicam-apps #350: [HELP WANTED] libcamera-detect object variable

tmsd2001, 2022-08-28:
Is it possible to deposit the recognized object in the saved image?
could it be possible in the Filename or exif
or draw the object as in the preview with opencv?
In which file should changes be made?

naushir, 2022-08-30:
> Is it possible to deposit the recognized object in the saved image? could it be possible in the Filename or exif or draw the object as in the preview with opencv? In which file should changes be made?

You can probably do this fairly easily by modifying [libcamera_detect.cpp](https://github.com/raspberrypi/libcamera-apps/blob/main/apps/libcamera_detect.cpp).  If you have something that seems useful, feel free to open a PR to merge it in!

> 
> Is it possible to recompile only the libcamera-detekt?

Yes, you can simply run ``make libcamera_detect`` to only build the libcamera_detect application.

tmsd2001, 2022-09-01:
I found the variable I was looking for and added it to libcamara-detect.
Can I also open a pull request if the changes are not perfect yet? So that others can improve on this?

naushir, 2022-09-02:
> I found the variable I was looking for and added it to libcamara-detect.
> Can I also open a pull request if the changes are not perfect yet? So that others can improve on this?

Yes, we will be happy to review your changes!

naushir, 2022-09-15:
With https://github.com/raspberrypi/libcamera-apps/pull/362 now merged, presumably this can be now closed?

closed 2022-09-15 by tmsd2001