raspberrypi/rpicam-apps #354: [question] Problem with OV5647

tmsd2001, 2022-09-01:
sometimes when using the libcamera-detect app I get a Segmentation fault.
With the imx477 I don't get any errors.

I think it might be because I'm using a longer than normal cable, but I can't test it because the camera is difficult to access.

The OV5647 shows frequently this info, the lower the frame rate, the more common.
```
couch[62] (0.64) @ 331,349 357x330
TfStage: Inference time: 86655.5 ms
Stream configuration adjusted
[13:49:42.010022542] [1542966]  INFO Camera camera.cpp:1035 configuring streams: (0) 2592x1944-YUV420 (1) 2592x1944-SGRBG10_CSI2P
[13:49:42.011959138] [1542967]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGRBG10_1X10 - Selected unicam format: 2592x1944-pgAA
TfStage: no low resolution stream
TfStage: Main stream is 2592x1944
 detected
[13:49:42.535174483] [1542966]  INFO Camera camera.cpp:1035 configuring streams: (0) 1296x972-YUV420 (1) 450x300-YUV420
[13:49:42.537029766] [1542967]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGRBG10_1X10 - Selected unicam format: 1296x972-pgAA
TfStage: Low resolution stream is 450x300
TfStage: Main stream is 1296x972
couch[62] (0.64) @ 331,349 357x327
```

What is your opinion, it is because of the cable and can I change it with software.
I need only a slow Framerate 0.5 - 5 fps
For example slow transfer rate

tmsd2001, 2022-09-01:
looks like the problem is with libcamera and not here.

naushir, 2022-09-05:
> What is your opinion, it is because of the cable and can I change it with software.

I doubt a faulty cable would give you an exception in software.  Instead, you would get a timeout message on screen.

As @kbingham mentioned, a call-stack (though gdb possibly) would be extremely useful to help debug this and point to where this error occurs.  I have a feeling it might be in libcamera-apps rather than libcamera itself.

tmsd2001, 2022-09-07:
Here the verbose print:
`
Available controls:
    AeEnable : [false..true]
    ExposureTime : [130..1064891]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [63965..1065021]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    Saturation : [0.000000..32.000000]
    AeMeteringMode : [0..3]
    Contrast : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    AwbMode : [0..7]
    AeConstraintMode : [0..3]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/2592x1944]
    AnalogueGain : [1.000000..63.937500]
    ColourGains : [0.000000..32.000000]
    Brightness : [-1.000000..1.000000]
    AeExposureMode : [0..3]
    AwbEnable : [false..true]
    Sharpness : [0.000000..16.000000]
Buffers allocated and mapped
Still capture setup complete
Requests created
Camera started!
person detected
[57:41:46.254456103] [2033006]  WARN V4L2 v4l2_videodevice.cpp:1810 /dev/video0[15:cap]: Zero sequence expected for first frame (got 1)

`
And then the program stops, all process still running with low cpu use.
top output:
``
thomasd  2033005 2025267  2 00:22 pts/5    00:00:33          |       |   \_ libcamera-detect --post-process-file object_detect_tf.json --verbose 2 --lores-width 400 --lores-height 300 -t 0 --object person -o person%03d.jpg
``

naushir, 2022-09-09:
What Raspberry Pi platform are you running this on? I suspect the Zero variants are going to struggle running libcamera + tensorflow lite, but that should not cause an actual crash.

tmsd2001, 2022-09-09:
It is a Raspberry Pi 4 2 GB with Rasbian OS
Uname output: Linux camerapi8 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux.
But i have test the Camera with a Rpi3 32bit, with the same result.
A reverse test with a Rpi Camera V3 showed no problems on Both systems.

tmsd2001, 2022-09-10:
I noticed that the error occurs when switching from still mode to viewfinder after saving a picture.

naushir, 2022-09-16:
I've added some timeout handling code to reset the device here: https://github.com/raspberrypi/libcamera-apps/pull/363.  Hope this should help with this issue.  Note that there needs to be an accompanying libcamera change built to trigger this reset mechanism.  That change has been posted for review [here](https://patchwork.libcamera.org/project/libcamera/list/?series=3488).

naushir, 2022-09-16:
@tmsd2001 can you please provide the output of ``libcamera-hello --version``?

naushir, 2022-10-03:
@tmsd2001, presumably you are still seeing this issue? Are you able to run your application under ``gdb`` and provide a call stack when it crashes?

tmsd2001, 2022-10-04:
Overvolt 6 has no effect, by chance I tried Overvolt 2 and as little CPU load as possible and low image resolution, I haven't had a failure since yesterday, but that can also be a coincidence.  I'll keep testing and looking at gdb.

naushir, 2022-10-04:
@tmsd2001 thanks for the update.  Do let me know the results.

naushir, 2022-10-07:
If this does not seem to go wrong any more, feel free to resolve this issue.

tmsd2001, 2022-10-09:
After a few days of testing, I think the best solution to the problem is overvolt=2 in the config.txt

closed 2022-10-09 by tmsd2001