raspberrypi/rpicam-apps #355: Building libcamera-apps with ubuntu 22 server on RPI4

altineller, 2022-09-02:
Firstly thank you very much for the hard work,

I am trying to build libcamera-apps on ubuntu 22 running on my rpi4. When I issue `cmake ..` under build directory, I get the following error:

-- Platform: arm64
-- Checking for module 'libcamera'
--   No package 'libcamera' found
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:48 (pkg_check_modules)

Appearently it complains about not finding libcamera. But however I did install libcamera0 and libcamera-dev packages that comes with the apt-get.

Any ideas on how to point the cmake script in the right direction?

I also made a dpkg query to see which files were installed by libcamera0 and libcamera-dev packages.

Best Regards,
Can Altineller

`$ dpkg-query -L libcamera-dev
/.
/usr
/usr/include
/usr/include/libcamera
/usr/include/libcamera/ipa
/usr/include/libcamera/ipa/ipa_controls.h
/usr/include/libcamera/ipa/ipa_interface.h
/usr/include/libcamera/ipa/ipa_module_info.h
/usr/include/libcamera/libcamera
/usr/include/libcamera/libcamera/bound_method.h
/usr/include/libcamera/libcamera/buffer.h
/usr/include/libcamera/libcamera/camera.h
/usr/include/libcamera/libcamera/camera_manager.h
/usr/include/libcamera/libcamera/control_ids.h
/usr/include/libcamera/libcamera/controls.h
/usr/include/libcamera/libcamera/event_dispatcher.h
/usr/include/libcamera/libcamera/event_notifier.h
/usr/include/libcamera/libcamera/file_descriptor.h
/usr/include/libcamera/libcamera/formats.h
/usr/include/libcamera/libcamera/framebuffer_allocator.h
/usr/include/libcamera/libcamera/geometry.h
/usr/include/libcamera/libcamera/libcamera.h
/usr/include/libcamera/libcamera/logging.h
/usr/include/libcamera/libcamera/object.h
/usr/include/libcamera/libcamera/pixel_format.h
/usr/include/libcamera/libcamera/property_ids.h
/usr/include/libcamera/libcamera/request.h
/usr/include/libcamera/libcamera/signal.h
/usr/include/libcamera/libcamera/span.h
/usr/include/libcamera/libcamera/stream.h
/usr/include/libcamera/libcamera/timer.h
/usr/include/libcamera/libcamera/version.h
/usr/lib
/usr/lib/aarch64-linux-gnu
/usr/lib/aarch64-linux-gnu/pkgconfig
/usr/lib/aarch64-linux-gnu/pkgconfig/camera.pc
/usr/lib/aarch64-linux-gnu/v4l2-compat.so
`

also:

`$ dpkg-query -L libcamera0 | grep -v usr\/share
/.
/usr
/usr/lib
/usr/lib/aarch64-linux-gnu
/usr/lib/aarch64-linux-gnu/gstreamer-1.0
/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
/usr/lib/aarch64-linux-gnu/libcamera
/usr/lib/aarch64-linux-gnu/libcamera/ipa_proxy_linux
/usr/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so
/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so
/usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so
/usr/lib/aarch64-linux-gnu/libcamera.so.0.1.0
/usr/lib/aarch64-linux-gnu/libcamera.so.0`

tmsd2001, 2022-09-02:
I can be wrong too, but when compiling libcamera-apps you should also use libcamera from soure compil.
The source code is much further.

naushir, 2022-09-05:
From the [documentation](https://cmake.org/cmake/help/latest/command/find_package.html), you could try setting ``CMAKE_MODULE_PATH`` to /usr/lib/aarch64-linux-gnu/libcamera/ explicitly.  I'm not sure why find-package does not locate the library, it could be something specific in the Ubuntu distribution.

naushir, 2022-09-05:
You might get a more detailed answer if you post this query in their support forums as well.

altineller, 2022-09-09:
I did compile, and installed (as root) libcamera, and now I am able to compile - somewhat - libcamera-apps. 

Thank you.

closed 2022-09-09 by altineller