raspberrypi/rpicam-apps #360: libcamera-apps - does not display on monitor, preview window not available.

altineller, 2022-09-09:
Hello,

After fiddling around, and installing required libraries and components, I am able to build and run libcamera-apps. However, I am running on a RPI, with no X installed. I would like to see the camera output on my monitor, without no x windows, just like in raspbian bullseye.

I did enable egl display (I had to install libepoxy) and compiled with it, and I still get `preview window unavailable` on the top when I run it.

Here is my `cmake ..` output:

> -- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No previous build - default to Release build
-- Platform: arm64
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libcamera'
--   Found libcamera, version 0.0.0
-- libcamera library found:
--     version: 0.0.0
--     libraries: /usr/local/lib/aarch64-linux-gnu/libcamera.so;/usr/local/lib/aarch64-linux-gnu/libcamera-base.so
--     include path: /usr/local/include/libcamera
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: program_options 
-- Checking for libavcodec
-- LIBDRM_LINK_LIBRARIES=/usr/lib/aarch64-linux-gnu/libdrm.so
-- LIBDRM display mode enabled
-- X11_LINK_LIBRARIES=/usr/lib/aarch64-linux-gnu/libX11.so
-- EPOXY_LINK_LIBRARIES=/usr/lib/aarch64-linux-gnu/libepoxy.so
-- EGL display mode enabled
-- ENABLE_QT not specified - set to 1
-- QT display mode will be unavailable!
-- Checking for OpenCV
-- OpenCV library found:
--     version: 4.5.4
--     libraries: -lopencv_core;-lopencv_imgproc;-lopencv_objdetect
--     include path: /usr/include/opencv4
-- OpenCV support is included
-- TFLite support not being included
-- Configuring done
-- Generating done
-- Build files have been written to: /home/can/libcamera-apps/build

Both libdrm and egl libraries are enabled as seen on above.

And here is my `make` output:

> [  2%] Building CXX object image/CMakeFiles/images.dir/bmp.cpp.o
[  4%] Building CXX object image/CMakeFiles/images.dir/yuv.cpp.o
[  6%] Building CXX object image/CMakeFiles/images.dir/jpeg.cpp.o
[  8%] Building CXX object image/CMakeFiles/images.dir/png.cpp.o
[ 10%] Building CXX object image/CMakeFiles/images.dir/dng.cpp.o
[ 12%] Linking CXX shared library libimages.so
[ 12%] Built target images
[ 14%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/post_processing_stage.cpp.o
[ 16%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/negate_stage.cpp.o
[ 18%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/hdr_stage.cpp.o
[ 20%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/pwl.cpp.o
[ 22%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/histogram.cpp.o
[ 25%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/motion_detect_stage.cpp.o
[ 27%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/sobel_cv_stage.cpp.o
[ 29%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/face_detect_cv_stage.cpp.o
[ 31%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/annotate_cv_stage.cpp.o
[ 33%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/plot_pose_cv_stage.cpp.o
[ 35%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/object_detect_draw_cv_stage.cpp.o
[ 37%] Linking CXX shared library libpost_processing_stages.so
[ 37%] Built target post_processing_stages
Generating version string: 1bf0ccacf2ef-intree 09-09-2022 (16:41:10)
[ 37%] Built target VersionCpp
[ 39%] Building CXX object preview/CMakeFiles/preview.dir/null_preview.cpp.o
[ 41%] Building CXX object preview/CMakeFiles/preview.dir/preview.cpp.o
[ 43%] Building CXX object preview/CMakeFiles/preview.dir/drm_preview.cpp.o
[ 45%] Building CXX object preview/CMakeFiles/preview.dir/egl_preview.cpp.o
[ 47%] Linking CXX shared library libpreview.so
[ 47%] Built target preview
[ 50%] Building CXX object core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o
[ 52%] Building CXX object core/CMakeFiles/libcamera_app.dir/post_processor.cpp.o
[ 54%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
[ 56%] Building CXX object core/CMakeFiles/libcamera_app.dir/options.cpp.o
[ 58%] Linking CXX shared library libcamera_app.so
[ 58%] Built target libcamera_app
[ 60%] Building CXX object encoder/CMakeFiles/encoders.dir/encoder.cpp.o
[ 62%] Building CXX object encoder/CMakeFiles/encoders.dir/null_encoder.cpp.o
[ 64%] Building CXX object encoder/CMakeFiles/encoders.dir/h264_encoder.cpp.o
[ 66%] Building CXX object encoder/CMakeFiles/encoders.dir/mjpeg_encoder.cpp.o
[ 68%] Linking CXX shared library libencoders.so
[ 68%] Built target encoders
[ 70%] Building CXX object output/CMakeFiles/outputs.dir/output.cpp.o
[ 72%] Building CXX object output/CMakeFiles/outputs.dir/file_output.cpp.o
[ 75%] Building CXX object output/CMakeFiles/outputs.dir/net_output.cpp.o
[ 77%] Building CXX object output/CMakeFiles/outputs.dir/circular_output.cpp.o
[ 79%] Linking CXX shared library liboutputs.so
[ 79%] Built target outputs
[ 81%] Building CXX object apps/CMakeFiles/libcamera-still.dir/libcamera_still.cpp.o
[ 83%] Linking CXX executable ../libcamera-still
[ 83%] Built target libcamera-still
[ 85%] Building CXX object apps/CMakeFiles/libcamera-vid.dir/libcamera_vid.cpp.o
[ 87%] Linking CXX executable ../libcamera-vid
[ 87%] Built target libcamera-vid
[ 89%] Building CXX object apps/CMakeFiles/libcamera-hello.dir/libcamera_hello.cpp.o
[ 91%] Linking CXX executable ../libcamera-hello
[ 91%] Built target libcamera-hello
[ 93%] Building CXX object apps/CMakeFiles/libcamera-raw.dir/libcamera_raw.cpp.o
[ 95%] Linking CXX executable ../libcamera-raw
[ 95%] Built target libcamera-raw
[ 97%] Building CXX object apps/CMakeFiles/libcamera-jpeg.dir/libcamera_jpeg.cpp.o
[100%] Linking CXX executable ../libcamera-jpeg
[100%] Built target libcamera-jpeg

As seen it is building preview, drm_preview and egl_preview and linking the libpreview.so

I am not sure which library is responsible of displaying images on console, so I enabled all the previews, with no luck.

Also, on ubuntu22, I have been unable to get a image display using cam application from libcamera package, so I am thinking it is a distribution specific problem, but I don't know what it is.

Any ideas/help/recommendations greatly appreciated.

Best Regards,
Can

naushir, 2022-09-10:
On the command line, you will need to use the DRM preview - which according to your build logs, is correctly enabled.
What distribution are you running?  Do you have FKMS enabled in your config.txt file?

naushir, 2022-09-15:
Is there anything more to look at here?  If not, I will close this down.

altineller, 2022-09-21:
@naushir libcamera-apps does compile on ubuntu 22 on RPI4 with success. Thank you.

closed 2022-09-21 by altineller