raspberrypi/rpicam-apps #361: libcamera-still not exiting after long timeout [BUG]

wcbonner, 2022-09-10:
I'm running the program from another program with the environment variable set:
```
LIBCAMERA_LOG_LEVELS=*:4
```
and command line:
```
libcamera-still --nopreview --hflip --vflip --thumb none --width 1920 --height 1080 --timeout 50100000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220909/0909%04d.jpg --framestart 1 --verbose 0
```

The timeout is calculated daily based on the time of sunset, assuming it starts at sunrise. When it's been running all day long it stops taking pictures at the correct time, but the program doesn't exit and the calling program remains waiting indefinitely. 

I've tried the following to duplicate in a short amount of time, but didn't see the same issue. 

```
wim@WimPiZeroW-Wim:~ $ libcamera-still --nopreview --hflip --vflip --thumb none --width 1920 --height 1080 --timeout 120010 --timelapse 60000 --output 0909%04d.jpg  --verbose 0
wim@WimPiZeroW-Wim:~ $ sudo systemctl --full --force edit wimsconstructioncam.service 
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=*:4 libcamera-still --nopreview --hflip --vflip --thumb none --width 1920 --height 1080 --timeout 120010 --timelapse 60000 --output 0909%04d.jpg  --verbose 0
wim@WimPiZeroW-Wim:~ $ libcamera-still --version
libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:16:31)
libcamera build: v0.0.0+3866-0c55e522
wim@WimPiZeroW-Wim:~ $ libcamera-still --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
wim@WimPiZeroW-Wim:~ $ camera-bug-report -t 5 -o bug.txt -c "libcamera-still -t 1000 -o test.jpg"
```

Is there anything else I can test without running the camera-bug-report tool with the libcamera-still timeout also set to 50100000?

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/9539633/bug.txt)

naushir, 2022-09-10:
Nothing obvious why this is not working.  All timeout values are 64bit, so should have not wrap-around problems.  Are you able to run libcamera-still with ``-v 2`` to display some verbose output and we can see what state it may be in when it is meant to exit?

wcbonner, 2022-09-10:
I'll let the program run today with its current settings to see the if it continues with the same problem, and assuming it does will run it with the --verbose 2 option afterwards. 

Do you want me to run my entire command as the argument to the bug-report program?

naushir, 2022-09-10:
No need to run through the bug-report program, just a copy/past of the output will do.  Thanks!

wcbonner, 2022-09-10:
I'll get it checked this evening. 

I should mention that I wasn't having the issue I'm currently having before the latest update arrived from the apt update in the Raspian feed. That's the only reason I think it may be a bug as opposed to something I'm doing wrong.

wcbonner, 2022-09-11:
I think things froze much quicker when I was running overnight. It crashed after 100 images but didn't exit the app. 

I ran it with the command:
```
sudo libcamera-still --nopreview --hflip --vflip --thumb none --width 1920 --height 1080 --timeout 49920000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220910-2/0910%04d.jpg --framestart 1 --verbose 2
```
Here's the final output section with most of the Viewfinder frame messages removed:
```
Viewfinder frame 120814
Viewfinder frame 120815
Camera stopped!
Tearing down requests, buffers and configuration
Configuring still capture...
Stream configuration adjusted
[105:54:24.878350592] [27328]  INFO Camera camera.cpp:1035 configuring streams: (0) 1920x1080-YUV420 (1) 2592x1944-SGRBG10_CSI2P
[105:54:24.888041388] [27329]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGRBG10_1X10 - Selected unicam format: 2592x1944-pgAA
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [130..1064891]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/2592x1944]
    AnalogueGain : [1.000000..63.937500]
    AeMeteringMode : [0..3]
    ColourGains : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [63965..1065021]
    Brightness : [-1.000000..1.000000]
    AwbMode : [0..7]
    AwbEnable : [false..true]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
    Saturation : [0.000000..32.000000]
    AeConstraintMode : [0..3]
Buffers allocated and mapped
Still capture setup complete
Requests created
Camera started!
Camera stopped!
Still capture image received
Exposure time: 59967
Ag 4.375 Dg 1.01214 Total 4.42809
JPEG size is 318271
EXIF data len 262
Saved image 1920 x 1080 to file /media/WimPiZeroW-Wim/DCIM/20220910-2/09100099.jpg
Tearing down requests, buffers and configuration
Configuring viewfinder...
Viewfinder size chosen is 1296x728
[105:54:25.820995051] [27328]  INFO Camera camera.cpp:1035 configuring streams: (0) 1296x728-YUV420
[105:54:25.832620405] [27329]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGRBG10_1X10 - Selected unicam format: 1920x1080-pgAA
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [118..969249]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/1928x1080]
    AnalogueGain : [1.000000..63.937500]
    AeMeteringMode : [0..3]
    ColourGains : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [32660..969367]
    Brightness : [-1.000000..1.000000]
    AwbMode : [0..7]
    AwbEnable : [false..true]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
[... 109 lines cut ...]
    AwbMode : [0..7]
    AwbEnable : [false..true]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
    Saturation : [0.000000..32.000000]
    AeConstraintMode : [0..3]
Buffers allocated and mapped
Still capture setup complete
Requests created
Camera started!
[105:56:25.258394729] [27329]  WARN V4L2 v4l2_videodevice.cpp:1803 /dev/video0[12:cap]: Zero sequence expected for first frame (got 1)
```
I had to press Ctrl-C to exit the program this morning. and then looked at the output directory:
```
wim@WimPiZeroW-Wim:~ $ ls /media/WimPiZeroW-Wim/DCIM/20220910-2/
09100001.jpg  09100009.jpg  09100017.jpg  09100025.jpg  09100033.jpg  09100041.jpg  09100049.jpg  09100057.jpg  09100065.jpg  09100073.jpg  09100081.jpg  09100089.jpg  09100097.jpg
09100002.jpg  09100010.jpg  09100018.jpg  09100026.jpg  09100034.jpg  09100042.jpg  09100050.jpg  09100058.jpg  09100066.jpg  09100074.jpg  09100082.jpg  09100090.jpg  09100098.jpg
09100003.jpg  09100011.jpg  09100019.jpg  09100027.jpg  09100035.jpg  09100043.jpg  09100051.jpg  09100059.jpg  09100067.jpg  09100075.jpg  09100083.jpg  09100091.jpg  09100099.jpg
09100004.jpg  09100012.jpg  09100020.jpg  09100028.jpg  09100036.jpg  09100044.jpg  09100052.jpg  09100060.jpg  09100068.jpg  09100076.jpg  09100084.jpg  09100092.jpg  09100100.jpg
09100005.jpg  09100013.jpg  09100021.jpg  09100029.jpg  09100037.jpg  09100045.jpg  09100053.jpg  09100061.jpg  09100069.jpg  09100077.jpg  09100085.jpg  09100093.jpg
09100006.jpg  09100014.jpg  09100022.jpg  09100030.jpg  09100038.jpg  09100046.jpg  09100054.jpg  09100062.jpg  09100070.jpg  09100078.jpg  09100086.jpg  09100094.jpg
09100007.jpg  09100015.jpg  09100023.jpg  09100031.jpg  09100039.jpg  09100047.jpg  09100055.jpg  09100063.jpg  09100071.jpg  09100079.jpg  09100087.jpg  09100095.jpg
09100008.jpg  09100016.jpg  09100024.jpg  09100032.jpg  09100040.jpg  09100048.jpg  09100056.jpg  09100064.jpg  09100072.jpg  09100080.jpg  09100088.jpg  09100096.jpg
```
You can see that it only ran for 100 captures before it froze up.

naushir, 2022-09-12:
Are you able to capture the kernel logs with ``dmesg`` once it crashes?  I've left a board running to see if I can reproduce this as well.

tmsd2001, 2022-09-12:
This looks like my error.
[https://github.com/raspberrypi/libcamera-apps/issues/354](url)

wcbonner, 2022-09-12:
I've changed my calling program to use the entire sensor in it's image capture by not specifying the image size. This morning it started and only ran for 21 minutes before freezing.  There were a bunch of CIFS messages at the end of the dmesg output, but nothing obvious from the camera system.  The camera program is writing to the local sd card so shouldn't be interacting with any of the CIFS mount points. 
```
Sep 12 06:12:16 WimPiZeroW-Wim wimsconstructioncam[27762]:  libcamera-still --nopreview --hflip --vflip --thumb none --timeout 49500000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220912/0912%04d.jpg --framestart 1 --verbose 0
wim@WimPiZeroW-Wim:~ $ ls /media/WimPiZeroW-Wim/DCIM/20220912/     
09120001.jpg  09120003.jpg  09120005.jpg  09120007.jpg  09120009.jpg  09120011.jpg  09120013.jpg  09120015.jpg  09120017.jpg  09120019.jpg  09120021.jpg
09120002.jpg  09120004.jpg  09120006.jpg  09120008.jpg  09120010.jpg  09120012.jpg  09120014.jpg  09120016.jpg  09120018.jpg  09120020.jpg
wim@WimPiZeroW-Wim:~ $ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.61+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1579 Fri Aug 26 11:08:59 BST 2022
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x0bc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i smsc95xx.macaddr=B8:27:EB:4C:4A:87 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=a8861cdf-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 177188K/458752K available (9029K kernel code, 1335K rwdata, 2944K rodata, 436K init, 546K bss, 19420K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 31886 entries in 94 pages
[    0.000000] ftrace: allocated 94 pages with 5 groups
[    0.000000] trace event string verifier disabled
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000097] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000207] bcm2835: system timer (irq = 27)
[    0.001126] Console: colour dummy device 80x30
[    0.001805] printk: console [tty1] enabled
[    0.001916] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060381] pid_max: default: 32768 minimum: 301
[    0.060650] LSM: Security Framework initializing
[    0.061029] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.061122] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.063520] cgroup: Disabling memory control group subsystem
[    0.064028] CPU: Testing write buffer coherency: ok
[    0.066436] Setting up static identity map for 0x8220 - 0x8258
[    0.068017] devtmpfs: initialized
[    0.082307] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.082768] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.082875] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.125462] pinctrl core: initialized pinctrl subsystem
[    0.127907] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.132843] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.139605] audit: initializing netlink subsys (disabled)
[    0.140316] audit: type=2000 audit(0.130:1): state=initialized audit_enabled=0 res=1
[    0.141712] thermal_sys: Registered thermal governor 'step_wise'
[    0.142776] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.142885] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.143228] Serial: AMBA PL011 UART driver
[    0.157507] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.191081] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-08-26T14:04:36, variant start
[    0.201118] raspberrypi-firmware soc:firmware: Firmware hash is 102f1e848393c2112206fadffaaf86db04e98326
[    0.262129] Kprobes globally optimized
[... cut ...]
[   44.898031] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   49.768875] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[  326.417586] FS-Cache: Netfs 'cifs' registered for caching
[  326.508656] Key type cifs.spnego registered
[  326.508714] Key type cifs.idmap registered
```
I'll be running the command directly and seeing how long it goes:
```
libcamera-still --nopreview --hflip --vflip --thumb none --timeout 40800000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220912/0912%04d.jpg --framestart 23 --verbose 2
```

naushir, 2022-09-12:
@wcbonner, thanks for the results.  Given there is nothing in the libcamera logs nor the dmesg logs, I'm not really sure what could be going wrong.

One suspicion would be the ov5647 sensor that sporadically stops working.  We do have a timeout mechanism that ought to catch this occurrence and throw an error message in the logs, but that seems to be missing as well!  Are you able to try with another sensor like the imx219 Camera v2?

One thing to note is the log message
```
[105:56:25.258394729] [27329]  WARN V4L2 v4l2_videodevice.cpp:1803 /dev/video0[12:cap]: Zero sequence expected for first frame (got 1)
```
is not expected to happen, so maybe there is something to look at here...

wcbonner, 2022-09-12:
@naushir Unfortunately, the two other cameras I've got aren't simple to swap out with this one. My current issues are running on a PiZeroW using this camera: https://www.amazon.com/gp/product/B07TB3CHZ3/ 

I've got an Arducam 64MP camera on one of my Pi4 running 64bit raspian https://www.arducam.com/64mp-ultra-high-res-camera-raspberry-pi/ but it requires running their custom fork of libcamera-still to support focusing. 

I've also got https://www.amazon.com/gp/product/B09VSVB4DT/ that I bought before realizing it needed custom kernel drivers running on Bullseye. I wasn't able to get it running quickly on my PiZeroW and my other Pi4 is still running Buster so I haven't been able to get it running yet. 

My current test on the PiZero with --verbose 2 is still running and I hope will provide further information. It will either stall or run until a but past sunset tonight before I get more data.

wcbonner, 2022-09-12:
I happened to look up and realize the screen had stopped scrolling again..  

```
Viewfinder frame 548568
Viewfinder frame 548569
Viewfinder frame 548570
Viewfinder frame 548571
Viewfinder frame 548572
Viewfinder frame 548573
Camera stopped!
Tearing down requests, buffers and configuration
Configuring still capture...
Stream configuration adjusted
[144:51:16.294316491] [8181]  INFO Camera camera.cpp:1035 configuring streams: (0) 2592x1944-YUV420 (1) 2592x1944-SGRBG10_CSI2P
[144:51:16.309587822] [8183]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGRBG10_1X10 - Selected unicam format: 2592x1944-pgAA
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [130..1064891]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/2592x1944]
    AnalogueGain : [1.000000..63.937500]
    AeMeteringMode : [0..3]
    ColourGains : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [63965..1065021]
    Brightness : [-1.000000..1.000000]
    AwbMode : [0..7]
    AwbEnable : [false..true]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
    Saturation : [0.000000..32.000000]
    AeConstraintMode : [0..3]
Buffers allocated and mapped
Still capture setup complete
Requests created
Camera started!
[144:51:16.698597164] [8183]  WARN V4L2 v4l2_videodevice.cpp:1803 /dev/video0[12:cap]: Zero sequence expected for first frame (got 1)
^C
```
Unfortunately the dmesg command didn't show anything recent, There were 340 files in my output directory now, so with the starting sequence of 23 that should be about five hours and 17 minutes, which isn't really helpful. 

Scrolling back to the previous capture, which I guess would have been the last successful image capture, the line with the warning would have been:
```
Camera stopped!
Still capture image received
Exposure time: 1755
Ag 1 Dg 1.01109 Total 1.01109
JPEG size is 967999
EXIF data len 262
Saved image 2592 x 1944 to file /media/WimPiZeroW-Wim/DCIM/20220912/09120340.jpg
Tearing down requests, buffers and configuration
Configuring viewfinder...
Viewfinder size chosen is 1296x972
[144:50:17.938214081] [8181]  INFO Camera camera.cpp:1035 configuring streams: (0) 1296x972-YUV420
[144:50:17.953041841] [8183]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGRBG10_1X10 - Selected unicam format: 1296x972-pgAA
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [92..760636]
    ScalerCrop : [(0, 0)/128x128..(0, 0)/2592x1944]
    AnalogueGain : [1.000000..63.937500]
    AeMeteringMode : [0..3]
    ColourGains : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
[... 111 lines cut ...]
Viewfinder frame 12
Viewfinder frame 13
Viewfinder frame 14
Viewfinder frame 15
Viewfinder frame 16
Viewfinder frame 17
Viewfinder frame 18
Viewfinder frame 19
Viewfinder frame 20
Viewfinder frame 21
Viewfinder frame 22
Viewfinder frame 23
Viewfinder frame 24
Viewfinder frame 25
Viewfinder frame 26
Viewfinder frame 27
Viewfinder frame 28
Viewfinder frame 29
Viewfinder frame 30
Viewfinder frame 31
Viewfinder frame 32
Viewfinder frame 33
Viewfinder frame 34
Viewfinder frame 35
```

tmsd2001, 2022-09-13:
My test with the imx477 worked without problems.
We need some kind of reset.
Because if I kill the program and start it again, it works right away.

naushir, 2022-09-13:
> My test with the imx477 worked without problems. We need some kind of reset. Because if I kill the program and start it again, it works right away.

This does add evidence that it may be a sensor related issue.  However, I would expect an ERROR message in the log stating as such.

In our old firmware camera stack we had a sensor reset mechanism if a lockup was detected.  Unfortunately with the new libcamera stack, there is no direct mechanism to trigger a reset on the kernel driver :(

naushir, 2022-09-13:
@wcbonner just to confirm - you are not running out of disk space during your capture run are you?

davidplowman, 2022-09-14:
I notice that the "Zero sequence expected for first frame (got 1)" message is the same as is reported in https://github.com/raspberrypi/libcamera-apps/issues/354 (this was commented on earlier, though the link is broken).

Can we establish what length cables are being used in both these cases? Of course that doesn't mean there can't be some kind of recovery mechanism, but finding an explanation for the trouble would be helpful.

naushir, 2022-09-14:
> I notice that the "Zero sequence expected for first frame (got 1)" message is the same as is reported in https://github.com/raspberrypi/libcamera-apps/issues/354 (this was commented on earlier, though the link is broken).

From a brief code inspection, we can only get this if the sensor produces a frame end interrupt before frame start or line interrupt on the very first frame.  Why this happens, I cannot tell but likely to be something specific to the ov5647.  I still don't see how this could cause the lockup though, it's only a log message.  Unless it indicates that the sensor is not behaving correctly and will soon lockup.  But again, we really ought to see a timeout error message!

naushir, 2022-09-14:
This is the timeout message you ought to see if the sensor has stopped streaming:

```
[43:44:16.063219438] [9354]  WARN V4L2 v4l2_videodevice.cpp:1997 /dev/video0[12:cap]: Dequeue timer of 10000.00us has expired!
[43:44:16.063350492] [9354] ERROR RPI raspberrypi.cpp:1819 Unicam has timed out!
[43:44:16.063420954] [9354] ERROR RPI raspberrypi.cpp:1820 Please check that your camera sensor connector is attached securely.
[43:44:16.063486750] [9354] ERROR RPI raspberrypi.cpp:1821 Alternatively, try another cable and/or sensor
```

naushir, 2022-09-16:
I've added some timeout handling code to reset the device here: https://github.com/raspberrypi/libcamera-apps/pull/363.  Hope this should help with this issue.  Note that there needs to be an accompanying libcamera change built to trigger this reset mechanism.  That change has been posted for review [here](https://patchwork.libcamera.org/project/libcamera/list/?series=3488).

naushir, 2022-09-16:
>```
> wim@WimPiZeroW-Wim:~ $ libcamera-still --version
> libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:16:31)
> libcamera build: v0.0.0+3866-0c55e522
>```

Rewinding back a bit - I'm trying to identify the build commit for libcamera given above.  0c55e522 does not seem to exist in the libcamera upstream tree at all.  Can you point me to the source of your libcamera tree - assuming you did a manual build of the libraries?

wcbonner, 2022-09-16:
> Rewinding back a bit - I'm trying to identify the build commit for libcamera given above.  0c55e522 does not seem to exist in the libcamera upstream tree at all.  Can you point me to the source of [...]

This was on a Raspberry Pi Zero W with Bullseye Lite initially installed from Pi Imager, then updated with the standard apt update commands. 

I don't have access to the machine right now, but should be able to get to it in a few hours if information on the apt packages is useful.

naushir, 2022-09-16:
It seems like this may be an issue with our packaging - it does not pick up the top libcamera source commit id when generating the version string.

wcbonner, 2022-09-17:
Don't know if this information is helpful at all:
```
wim@WimPiZeroW-Wim:~ $ apt info libcamera-apps               
Package: libcamera-apps
Version: 0~git20220830+1bf0cca-1
Priority: optional
Section: video
Maintainer: Serge Schneider <serge@raspberrypi.com>
Installed-Size: 1,281 kB
Depends: libavcodec58 (>= 7:4.2), libavdevice58 (>= 7:4.0), libavformat58 (>= 7:4.2), libboost-program-options1.74.0 (>= 1.74.0), libc6 (>= 2.29), libcamera0 (>= 0~git20220826+3fad116f), libdrm2 (>= 2.4.99), libepoxy0 (>= 1.0), libexif12 (>= 0.6.21-1~), libgcc-s1 (>= 3.5), libjpeg62-turbo (>= 1.3.1), libpng16-16 (>= 1.6.2-1), libqt5core5a (>= 5.0.2), libqt5gui5 (>= 5.0.2) | libqt5gui5-gles (>= 5.0.2), libqt5widgets5 (>= 5.0.2), libstdc++6 (>= 7), libswresample3 (>= 7:4.0), libtiff5 (>= 4.0.3), libx11-6
Conflicts: libcamera-apps-lite
Replaces: libcamera-apps-lite
Homepage: https://github.com/raspberrypi/libcamera-apps/
Download-Size: 267 kB
APT-Sources: http://archive.raspberrypi.org/debian bullseye/main armhf Packages
Description: libcamera-apps
 Small suite of libcamera-based apps that aim to copy the functionality of the
 existing "raspicam" apps

wim@WimPiZeroW-Wim:~ $ apt info libcamera0
Package: libcamera0
Version: 0~git20220826+3fad116f-1
Priority: optional
Section: libs
Source: libcamera
Maintainer: Debian Multimedia Maintainers <debian-multimedia@lists.debian.org>
Installed-Size: 2,184 kB
Depends: libatomic1 (>= 4.8), libc6 (>= 2.30), libdw1 (>= 0.158), libgcc-s1 (>= 3.5), libglib2.0-0 (>= 2.37.3), libgnutls30 (>= 3.7.0), libgstreamer-plugins-base1.0-0 (>= 1.14.0), libgstreamer1.0-0 (>= 1.18.0), libstdc++6 (>= 6), libudev1 (>= 183), libunwind8, libyaml-0-2, raspberrypi-kernel (>= 1:1.20211201~)
Homepage: https://libcamera.org/
Download-Size: 543 kB
APT-Manual-Installed: no
APT-Sources: http://archive.raspberrypi.org/debian bullseye/main armhf Packages
Description: complex camera support library
 libcamera is a complex camera support library which handles low-level
 control of the camera devices, providing a unified higher-level
 programming interface to the applications.

N: There is 1 additional record. Please use the '-a' switch to see it
wim@WimPiZeroW-Wim:~ $ libcamera-still --version
libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:16:31)
libcamera build: v0.0.0+3866-0c55e522
```

naushir, 2022-09-23:
Given that the sensor timeout handling code does not seem to fix the issue, I wonder if this could possibly be a libcamera-apps bug where a request message may get lost.  @tmsd2001 and @wcbonner would you be able to run the libcamera-still timeout command with the following patch added to libcamera-apps:
```
diff --git a/core/libcamera_app.hpp b/core/libcamera_app.hpp
index ef976ac0aee8..e488dcfc3e25 100644
--- a/core/libcamera_app.hpp
+++ b/core/libcamera_app.hpp
@@ -19,6 +19,9 @@
 #include <thread>
 #include <variant>

+#include <chrono>
+using namespace std::chrono_literals;
+
 #include <libcamera/base/span.h>
 #include <libcamera/camera.h>
 #include <libcamera/camera_manager.h>
@@ -144,10 +147,17 @@ private:
                T Wait()
                {
                        std::unique_lock<std::mutex> lock(mutex_);
-                       cond_.wait(lock, [this] { return !queue_.empty(); });
-                       T msg = std::move(queue_.front());
-                       queue_.pop();
-                       return msg;
+                       if (cond_.wait_for(lock, 5s, [this] { return !queue_.empty(); }))
+                       {
+                               T msg = std::move(queue_.front());
+                               queue_.pop();
+                               return msg;
+                       }
+                       else
+                       {
+                               std::cout << "QUEUE TIMEOUT!!!" << std::endl;
+                               return T(MsgType::Timeout);
+                       }
                }
```

Please also append LIBCAMERA_LOG_LEVELS=RPI:0 to the command, e.g:
```
LIBCAMERA_LOG_LEVELS=RPI:0 libcamera-still --nopreview --hflip --vflip --thumb none --timeout 40800000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220912/0912%04d.jpg --verbose 2  
```

Once you see the lockup, please post the console output and we can see if my above error message gets displayed.  Thanks!!

tmsd2001, 2022-09-23:
Is this Patch in the main brunch?

naushir, 2022-09-23:
No this patch is purely for debug purposes, so you will have to put it manually in your build tree.

Alternatively I could make a new tree with the battery included for you to check out and build?

tmsd2001, 2022-09-23:
That would be nice.

wcbonner, 2022-09-23:
I am traveling today, but should be able to build the code with the patch and test it tonight or tomorrow.

wcbonner, 2022-09-25:
I seem to have broken something. I followed https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps and here's the commands I ran to build the code locally:
```
git clone git@github.com:raspberrypi/libcamera-apps.git
cd libcamera-apps/
mkdir build
cd build/
sudo apt install -y cmake libboost-program-options-dev libdrm-dev libexif-dev
sudo apt install -y libcamera-dev libjpeg-dev libtiff5-dev
cmake .. -DENABLE_DRM=1 -DENABLE_X11=0 -DENABLE_QT=0 -DENABLE_OPENCV=0 -DENABLE_TFLITE=0
make -j1
```
That built properly, but I didn't think to test to see if it was running properly. 

Then I manually added your changes to core/libcamera_app.hpp because I couldn't remember the command to add the patch automatically and ran the make -j1 command again.  I confirmed that the debian package command is still in the path and can run the locally built version with the path included. I seem to have broken something and neither the freshly built program or the debian package can get images now. Any suggestion short of a fresh sd card format to diagnose this?

```
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=RPI:0 libcamera-still --version                       
libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:16:31)
libcamera build: v0.0.0+3866-0c55e522
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=RPI:0 ~/libcamera-apps/build/libcamera-still --version
libcamera-apps build: dd97618a2552-intree-dirty 25-09-2022 (10:50:46)
libcamera build: v0.0.0+3866-0c55e522
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=RPI:0 ~/libcamera-apps/build/libcamera-still --list-cameras --verbose 2
Available cameras
-----------------
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [30.00 fps - (0, 0)/0x0 crop]
                             1296x972 [30.00 fps - (0, 0)/0x0 crop]
                             1920x1080 [30.00 fps - (0, 0)/0x0 crop]
                             2592x1944 [30.00 fps - (0, 0)/0x0 crop]
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=RPI:0 ~/libcamera-apps/build/libcamera-still --nopreview --hflip --vflip --thumb none --timeout 28920000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220925/0925%04d.jpg --framestart 1 --verbose 2
Options:
    verbose: 2
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 28920000
    width: 0
    height: 0
    output: /media/WimPiZeroW-Wim/DCIM/20220925/0925%04d.jpg
    post_process_file: 
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: hvflip
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
    metadata: 
    metadata-format: json
    encoding: jpg
    quality: 93
    raw: 0
    restart: 0
    timelapse: 60000
    framestart: 1
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
    thumbnail width: 0
    thumbnail height: 0
    thumbnail quality: 0
    latest: 
    immediate 0
Running without preview window
Opening camera...
[0:20:21.537121132] [1240]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:20:21.692859917] [1241]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[0:20:21.694447874] [1241] DEBUG RPI raspberrypi.cpp:1160 Unable to acquire a Unicam instance
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
Configuring viewfinder...
[0:20:21.702091667] [1240] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
Viewfinder size chosen is 1296x972
[0:20:21.709023479] [1240] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
[0:20:21.712059397] [1240] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
[0:20:21.714604328] [1240]  INFO Camera camera.cpp:1035 configuring streams: (0) 1296x972-YUV420
[0:20:21.717143259] [1241] ERROR V4L2 v4l2_videodevice.cpp:1234 /dev/video16[16:cap]: Unable to request 0 buffers: Device or resource busy
[0:20:21.719091207] [1241] DEBUG RPI raspberrypi.cpp:169 Format: 640x480 fmt SGBRG10 Score: 3296 (best 3296)
[0:20:21.720422171] [1241] DEBUG RPI raspberrypi.cpp:169 Format: 1296x972 fmt SGBRG10 Score: 1000 (best 1000)
[0:20:21.723616084] [1241] DEBUG RPI raspberrypi.cpp:169 Format: 1920x1080 fmt SGBRG10 Score: 1349.67 (best 1000)
[0:20:21.724070072] [1241] DEBUG RPI raspberrypi.cpp:169 Format: 2592x1944 fmt SGBRG10 Score: 1567 (best 1000)
[0:20:21.726239013] [1241]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGRBG10_1X10 - Selected unicam format: 1296x972-pgAA
[0:20:21.728529951] [1241] DEBUG RPI raspberrypi.cpp:793 Setting ISP Output0 to 1296x972-YU12
[0:20:21.730112908] [1241] DEBUG RPI raspberrypi.cpp:807 Stream ISP Output0 has color space sYCC
[0:20:21.733316821] [1241] DEBUG RPI raspberrypi.cpp:873 Setting ISP Output1 (internal) to 648x486-YU12
[0:20:21.735903751] [1241] ERROR V4L2 v4l2_videodevice.cpp:897 /dev/video16[16:cap]: Unable to set format: Invalid argument
[0:20:21.737582706] [1241] ERROR RPI raspberrypi.cpp:889 Failed to set format on ISP stats stream: 0x0-BSTA
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
[0:20:21.745562490] [1241] ERROR V4L2 v4l2_videodevice.cpp:1234 /dev/video16[16:cap]: Unable to request 0 buffers: Device or resource busy
[0:20:21.748061422] [1241] ERROR V4L2 v4l2_videodevice.cpp:1234 /dev/video16[16:cap]: Unable to request 0 buffers: Device or resource busy
Camera closed
ERROR: *** failed to configure streams ***
```

wcbonner, 2022-09-25:
Ok. It looks like my previous program was hanging on to something after stopping, even though it didn't show in ps output. I disabled my service and rebooted. Here's a verbose output with the lockup at the end
```
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=RPI:0 ~/libcamera-apps/build/libcamera-still --nopreview --hflip --vflip --thumb none --timeout 28920000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220925/0925%04d.jpg --framestart 1 --verbose 2
Options:
    verbose: 2
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 28920000
    width: 0
    height: 0
    output: /media/WimPiZeroW-Wim/DCIM/20220925/0925%04d.jpg
    post_process_file:
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: hvflip
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
    metadata:
    metadata-format: json
    encoding: jpg
    quality: 93
    raw: 0
    restart: 0
    timelapse: 60000
    framestart: 1
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
    thumbnail width: 0
    thumbnail height: 0
    thumbnail quality: 0
    latest:
    immediate 0
Running without preview window
Opening camera...
[0:01:54.410750406] [793]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:01:54.598790865] [794]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:01:54.600400869] [794] DEBUG RPI raspberrypi.cpp:1160 Unable to acquire a Unicam instance
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
Configuring viewfinder...
[0:01:54.606847884] [793] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
Viewfinder size chosen is 1296x972
[0:01:54.613736901] [793] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
[0:01:54.615978907] [793] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
[0:01:54.618540913] [793]  INFO Camera camera.cpp:1035 configuring streams: (0) 1296x972-YUV420
[0:01:54.621311920] [794] DEBUG RPI raspberrypi.cpp:169 Format: 640x480 fmt SGBRG10 Score: 3296 (best 3296)
[0:01:54.623172924] [794] DEBUG RPI raspberrypi.cpp:169 Format: 1296x972 fmt SGBRG10 Score: 1000 (best 1000)
[0:01:54.625019929] [794] DEBUG RPI raspberrypi.cpp:169 Format: 1920x1080 fmt SGBRG10 Score: 1349.67 (best 1000)
[0:01:54.626811933] [794] DEBUG RPI raspberrypi.cpp:169 Format: 2592x1944 fmt SGBRG10 Score: 1567 (best 1000)
[0:01:54.628857938] [794]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGRBG10_1X10 - Selected unicam format: 1296x972-pgAA
[0:01:54.631569945] [794] DEBUG RPI raspberrypi.cpp:793 Setting ISP Output0 to 1296x972-YU12
[... 1932 lines cut ...]
[0:02:01.439910650] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 0 for ISP Output0
[0:02:01.443249655] [794] DEBUG RPI raspberrypi.cpp:1752 Input re-queue to ISP, buffer id 1, timestamp: 121245540000
[0:02:01.445035658] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer -1 for ISP Input
[0:02:01.447238661] [794] DEBUG RPI raspberrypi.cpp:1524 frame start 197
[0:02:01.453895672] [794] DEBUG RPI raspberrypi.cpp:1524 frame start 198
[0:02:01.456385676] [794] DEBUG RPI raspberrypi.cpp:1870 Stream ISP Input buffer complete, buffer id 1, timestamp: 121245540000
[0:02:01.458152678] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 1 for Unicam Image
[0:02:01.460225682] [794] DEBUG RPI raspberrypi.cpp:1898 Stream ISP Output0 buffer complete, buffer id 1, timestamp: 121245540000
[0:02:01.462159685] [794] DEBUG RPI raspberrypi.cpp:1898 Stream ISP Output1 buffer complete, buffer id 0, timestamp: 121245540000
[0:02:01.464492688] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 0 for ISP Output1
[0:02:01.465889690] [794] DEBUG RPI raspberrypi.cpp:1898 Stream ISP Stats buffer complete, buffer id 0, timestamp: 121245540000
[0:02:01.472306700] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 0 for ISP Stats
Viewfinder frame 113
[0:02:01.479021711] [794] DEBUG RPI raspberrypi.cpp:2145 Signalling signalIspPrepare: Bayer buffer id: 2
[0:02:01.484303719] [794] DEBUG RPI raspberrypi.cpp:1115 queueRequestDevice: New request.
[0:02:01.486092722] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 1 for ISP Output0
[0:02:01.489424727] [794] DEBUG RPI raspberrypi.cpp:1752 Input re-queue to ISP, buffer id 2, timestamp: 121312171000
[0:02:01.491348730] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer -1 for ISP Input
[0:02:01.494480735] [794] DEBUG RPI raspberrypi.cpp:1842 Stream Unicam Image buffer dequeue, buffer id 0, timestamp: 121445438000
[0:02:01.497508739] [794] DEBUG RPI raspberrypi.cpp:1524 frame start 199
[0:02:01.501935746] [794] DEBUG RPI raspberrypi.cpp:1842 Stream Unicam Image buffer dequeue, buffer id 1, timestamp: 121478751000
[0:02:01.504864751] [794] DEBUG RPI raspberrypi.cpp:1870 Stream ISP Input buffer complete, buffer id 2, timestamp: 121312171000
[0:02:01.507247755] [794] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 2 for Unicam Image
[0:02:01.509989759] [794] DEBUG RPI raspberrypi.cpp:1898 Stream ISP Output0 buffer complete, buffer id 2, timestamp: 121312171000
[0:02:

wcbonner, 2022-09-25:
After waiting several minutes with nothing further displaying on the screen, I pressed ^C and got a prompt back, then reran the previous command. It seemed to have the same problem I was having earlier of a resource being locked out.
```
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=RPI:0 ~/libcamera-apps/build/libcamera-still --nopreview --hflip --vflip --thumb none --timeout 28920000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220925/0925%04d.jpg --framestart 1 --verbose 2
Options:
    verbose: 2
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 28920000
    width: 0
    height: 0
    output: /media/WimPiZeroW-Wim/DCIM/20220925/0925%04d.jpg
    post_process_file:
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: hvflip
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
    metadata:
    metadata-format: json
    encoding: jpg
    quality: 93
    raw: 0
    restart: 0
    timelapse: 60000
    framestart: 1
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
    thumbnail width: 0
    thumbnail height: 0
    thumbnail quality: 0
    latest:
    immediate 0
Running without preview window
Opening camera...
[0:09:05.173337344] [1036]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:09:05.328052040] [1037]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:09:05.329572880] [1037] DEBUG RPI raspberrypi.cpp:1160 Unable to acquire a Unicam instance
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
Configuring viewfinder...
[0:09:05.337188077] [1036] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
Viewfinder size chosen is 1296x972
[0:09:05.343377425] [1036] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
[0:09:05.346198128] [1036] DEBUG RPI raspberrypi.cpp:531 Try color space sYCC
[0:09:05.348722862] [1036]  INFO Camera camera.cpp:1035 configuring streams: (0) 1296x972-YUV420
[0:09:05.350915631] [1037] ERROR V4L2 v4l2_videodevice.cpp:1234 /dev/video13[13:out]: Unable to request 0 buffers: Device or resource busy
[0:09:05.353287381] [1037] DEBUG RPI raspberrypi.cpp:169 Format: 640x480 fmt SGBRG10 Score: 3296 (best 3296)
[0:09:05.355121187] [1037] DEBUG RPI raspberrypi.cpp:169 Format: 1296x972 fmt SGBRG10 Score: 1000 (best 1000)
[0:09:05.357606925] [1037] DEBUG RPI raspberrypi.cpp:169 Format: 1920x1080 fmt SGBRG10 Score: 1349.67 (best 1000)
[0:09:05.358737806] [1037] DEBUG RPI raspberrypi.cpp:169 Format: 2592x1944 fmt SGBRG10 Score: 1567 (best 1000)
[0:09:05.360281644] [1037]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGRBG10_1X10 - Selected unicam format: 1296x972-pgAA
[0:09:05.362471413] [1037] ERROR V4L2 v4l2_videodevice.cpp:1040 /dev/video13[13:out]: Unable to set format: Invalid argument
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
[0:09:05.374178179] [1037] ERROR V4L2 v4l2_videodevice.cpp:1234 /dev/video13[13:out]: Unable to request 0 buffers: Device or resource busy
[0:09:05.376599924] [1037] ERROR V4L2 v4l2_videodevice.cpp:1234 /dev/video13[13:out]: Unable to request 0 buffers: Device or resource busy
Camera closed
ERROR: *** failed to configure streams ***
wim@WimPiZeroW-Wim:~ $
```

naushir, 2022-09-26:
@wcbonner thanks for sharing those results. Looks like you have a kernel crash this time round.  Did you see this before when libcamera-still locked up?  In you dmesg log posted earlier, I did not see any evidence of a kernel crash.

naushir, 2022-09-26:
Another useful test would be to try adding ``over_voltage=6`` to your ``/boot/config.txt`` file, reboot and run your command again.

naushir, 2022-09-26:
> That would be nice.

@tmsd2001 you can find the patch merged on my test tree at https://github.com/naushir/libcamera-apps/tree/issue_361

wcbonner, 2022-09-26:
> Another useful test would be to try adding `over_voltage=6` to your `/boot/config.txt` file, reboot and run your command again.

@naushir I just now added that in the line after the camera_detect line, rebooted my pi, and restarted the command that crashed. It's streaming debug data on my console, and I've seen it pause once a second to record pictures. I'll let this run until it either crashes or hits the timeout value, which might be close to sunset this evening.

wcbonner, 2022-09-27:
@naushir I think the code just exited normally after my timeout value.  I hit the up arrow in my output below to see what the command I'd started this run with was. 

```
[8:04:06.581451663] [783] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 0 for ISP Stats
Viewfinder frame 513417
[8:04:06.588426367] [783] DEBUG RPI raspberrypi.cpp:2145 Signalling signalIspPrepare: Bayer buffer id: 1
[8:04:06.593111168] [783] DEBUG RPI raspberrypi.cpp:1115 queueRequestDevice: New request.
[8:04:06.594919091] [783] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 0 for ISP Output0
[8:04:06.598265949] [783] DEBUG RPI raspberrypi.cpp:1752 Input re-queue to ISP, buffer id 1, timestamp: 29046504454000
[8:04:06.600153869] [783] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer -1 for ISP Input
[8:04:06.603097744] [783] DEBUG RPI raspberrypi.cpp:1842 Stream Unicam Image buffer dequeue, buffer id 2, timestamp: 29046571081000
[8:04:06.605454644] [783] DEBUG RPI raspberrypi.cpp:1524 frame start 1278
[8:04:06.607968537] [783] DEBUG RPI raspberrypi.cpp:1524 frame start 1279
[8:04:06.611497387] [783] DEBUG RPI raspberrypi.cpp:1870 Stream ISP Input buffer complete, buffer id 1, timestamp: 29046504454000
[8:04:06.613313310] [783] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 1 for Unicam Image
[8:04:06.615965197] [783] DEBUG RPI raspberrypi.cpp:1898 Stream ISP Output0 buffer complete, buffer id 1, timestamp: 29046504454000
[8:04:06.618558087] [783] DEBUG RPI raspberrypi.cpp:1898 Stream ISP Output1 buffer complete, buffer id 0, timestamp: 29046504454000
[8:04:06.620367010] [783] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 0 for ISP Output1
[8:04:06.621894945] [783] DEBUG RPI raspberrypi.cpp:1898 Stream ISP Stats buffer complete, buffer id 0, timestamp: 29046504454000
[8:04:06.627174721] [783] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 0 for ISP Stats
Viewfinder frame 513418
Closing Libcamera application(frames displayed 512456, dropped 0)
[8:04:06.638182254] [783] DEBUG RPI raspberrypi.cpp:2145 Signalling signalIspPrepare: Bayer buffer id: 2
[8:04:06.642874054] [783] DEBUG RPI raspberrypi.cpp:1115 queueRequestDevice: New request.
[8:04:06.644654979] [783] DEBUG RPISTREAM rpi_stream.cpp:240 Queuing buffer 1 for ISP Output0
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
wim@WimPiZeroW-Wim:~ $ LIBCAMERA_LOG_LEVELS=RPI:0 ~/libcamera-apps/build/libcamera-still --nopreview --hflip --vflip --thumb none --timeout 28920000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20220925/0925%04d.jpg --framestart 1 --verbose 2
```

I'm going to run the default distribution package for the rest of the day with the changes to the /boot/config.txt and see if it runs until sunset without problems. (it should be about two and a half hours)

wcbonner, 2022-09-28:
@naushir Today I let my service start the raspian package version of libcamera-still at sunrise and it ran uninterrupted to sunset, producing 773 images. The only change I know of from before is that I've got the `over_voltage=6` line in my `/boot/confit.txt` file.

naushir, 2022-09-28:
@wcbonner That's encouraging to hear. There has been a similar issue raised at https://github.com/raspberrypi/libcamera-apps/issues/246, where ``over_voltage`` seems to certainly help with stability on Pi Zero devices.

wcbonner, 2022-09-28:
> @wcbonner That's encouraging to hear. There has been a similar issue raised at #246, where `over_voltage` seems to certainly help with stability on Pi Zero devices.

@naushir It's certainly encouraging to hear that the camera locking up on PiZero devices may be a hardware problem with voltage dropping that may have a workaround. Hopefully the software can be figured out to recognize that fault and exit with an error code instead of waiting forever on the non-responsive device. 

```
wim@WimPiZeroW-Wim:~ $ vcgencmd get_config int
aphy_params_current=547
arm_freq=1000
arm_freq_min=700
audio_pwm_mode=514
camera_auto_detect=1
config_hdmi_boost=5
core_freq=400
disable_auto_turbo=1
disable_commandline_tags=2
disable_overscan=1
display_auto_detect=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dphy_params_current=547
dvfs=3
enable_tvout=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=300
ignore_lcd=1
init_uart_clock=0x2dc6c00
mask_gpu_interrupt0=3072
mask_gpu_interrupt1=26498
max_framebuffers=2
over_voltage=6
over_voltage_avs=0x249f0
pause_burst_frames=1
program_serial_random=1
sdram_freq=450
total_mem=512
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x9a7ec80
wim@WimPiZeroW-Wim:~ $ vcgencmd measure_volts
volt=1.4000V
```

closed 2022-09-30 by wcbonner