raspberrypi/rpicam-apps #364: [BUG] undefined symbol: _ZN9libcamera10ColorSpace4SyccE

vitasimek, 2022-09-19:
**Describe the bug**
After update libcamera does not work.

**Bug report**
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 19-09-2022 (13:48:55)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000003d2ad277
Model		: Raspberry Pi 3 Model B Rev 1.2

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=76775905-02 rootfstype=ext4 fsck.repair=yes rootwait

** cat /boot/config.txt **
# For more options and information see
[... 806 lines cut ...]
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Raspbian GNU/Linux bookworm/sid"
NAME="Raspbian GNU/Linux"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

** vcgencmd version **
Aug 26 2022 14:04:36 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

** libcamera-hello --version **
libcamera-hello: symbol lookup error: /lib/arm-linux-gnueabihf/libpreview.so: undefined symbol: _ZN9libcamera10ColorSpace4SyccE

**Additional context**
Firstly just an ordinary upgrade has been run for bullseye, then I've changed to bookworm, and full-upgraded, but the result is the same.

Thanks

naushir, 2022-09-20:
This has already been fixed.  You need to ensure you have built the latest version of the libcamera libraries.  Alternatively you could run ``sudo apt update`` and ``sudo apt upgrade`` to pull the latest libraries from apt.

closed 2022-09-20 by naushir

vitasimek, 2022-09-20:
> This has already been fixed. You need to ensure you have built the latest version of the libcamera libraries. Alternatively you could run `sudo apt update` and `sudo apt upgrade` to pull the latest libraries from apt.

Thank you for answer.
I am using bookworm version, and the libcamera-apps comes from bullseye (archive.raspberrypi.org/debian bullseye/main) repo. It is not available there in bookworm repo.
Let's build it localy from github.

pi@raspberrypi:~/repos/libcamera $ apt show libcamera-apps
Package: libcamera-apps
Version: 0~git20220830+1bf0cca-1
Priority: optional
Section: video
Maintainer: Serge Schneider <serge@raspberrypi.com>
Installed-Size: 1 281 kB
Depends: libavcodec58 (>= 7:4.2), libavdevice58 (>= 7:4.0), libavformat58 (>= 7:4.2), libboost-program-options1.74.0 (>= 1.74.0), libc6 (>= 2.29), libcamera0 (>= 0~git20220826+3fad116f), libdrm2 (>= 2.4.99), libepoxy0 (>= 1.0), libexif12 (>= 0.6.21-1~), libgcc-s1 (>= 3.5), libjpeg62-turbo (>= 1.3.1), libpng16-16 (>= 1.6.2-1), libqt5core5a (>= 5.0.2), libqt5gui5 (>= 5.0.2) | libqt5gui5-gles (>= 5.0.2), libqt5widgets5 (>= 5.0.2), libstdc++6 (>= 7), libswresample3 (>= 7:4.0), libtiff5 (>= 4.0.3), libx11-6
Conflicts: libcamera-apps-lite
Replaces: libcamera-apps-lite
Homepage: https://github.com/raspberrypi/libcamera-apps/
Download-Size: 267 kB
APT-Manual-Installed: yes
APT-Sources: http://archive.raspberrypi.org/debian bullseye/main armhf Packages
Description: libcamera-apps
 Small suite of libcamera-based apps that aim to copy the functionality of the
 existing "raspicam" apps