raspberrypi/rpicam-apps #365: framerate parameter behaivor

altineller, 2022-09-21:
Hello,

I am using libcamera-apps, more specifically libcamera-vid with ov9281, on the latest buster based raspbian. When I ask libcamera to list cameras found it responds with the following:

>0 : ov9281 [1280x800] (/base/soc/i2c0mux/i2c@1/ov9281@60)
    Modes: 'R8' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                  1280x720 [144.40 fps - (0, 40)/1280x720 crop]
                  1280x800 [120.76 fps - (0, 0)/1280x800 crop]
           'R10_CSI2P' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                         1280x720 [144.40 fps - (0, 40)/1280x720 crop]
                         1280x800 [120.76 fps - (0, 0)/1280x800 crop]

Which confirms the specs of the camera: I should be able to do 1280x800 at 120fps or 1280x720 at 140fps.

But however when I give the --framerate 120 as below, I get the following output:

> libcamera-vid -t 0 --tuning-file /usr/share/libcamera/ipa/raspberrypi/ov9281.json --height 720 --width 1280 --framerate 120

Made DRM preview window
[1:54:27.237858739] [1509]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[1:54:27.256852831] [1510]  WARN CameraSensorProperties camera_sensor_properties.cpp:174 No static properties available for 'ov9281'
[1:54:27.256952609] [1510]  WARN CameraSensorProperties camera_sensor_properties.cpp:176 Please consider updating the camera sensor properties database
[1:54:27.266763961] [1510]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 to Unicam device /dev/media0 and ISP device /dev/media1
[1:54:27.267539257] [1509]  INFO Camera camera.cpp:1035 configuring streams: (0) 1280x720-YUV420
[1:54:27.268399591] [1510]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov9281@60 - Selected sensor format: 1280x720-Y10_1X10 - Selected unicam format: 1280x720-Y10P
ERROR: *** failed to start output streaming ***

So, no matter what the resolution is it allows max 60fps, any higher than that will result in "failed to start output streaming* - and also it opens the sensor with Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 - I dont know if this 60 on the end depicts framerate.

Any ideas, recommendations, help greatly appreciated.

Best Regards,
Can

6by9, 2022-09-21:
H264 levels - https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels

The codec default is level 4.0, which is max 245,760 macroblocks (16x16 blocks) per second. That is good up to 1080p30 or 720p60.
720p120 needs at least level 4.2.
```libcamera-vid --level 4.2 ......```

Note that the H264 encoder was only originally designed for level 4.0, so there are no guarantees that level 4.2 will be achievable in real time.
I'm just looking at ov9281 for other reasons - it is possible that there are issues as those frame rates for R8 modes look wrong (the pixel rate increases in 8 bit mode, so the frame rate should go 10/8 times higher).

> /i2c@1/ov9281@60

No, it's the I2C address of the sensor.

6by9, 2022-09-21:
If it reports that it failed to start streaming, have a look in the kernel log (```dmesg```) to see if there is any further information as to why.

6by9, 2022-09-22:
I had an ov9281 up and running for other reasons.
```libcamera-vid --width 1280 --height 800 -o foo.h264 --framerate 120 --save-pts foo.txt -t 10000``` worked perfectly. Timestamps of all frames are 8.335ms apart.

Using Bullseye is strongly recommended for using libcamera.
Buster is the legacy distribution and won't be updated beyond those updates made to the upstream kernel, or with updates from Debian.

altineller, 2022-09-23:
I mistakenly wrote Buster, I am using Bullseye.

The dmesg outputs the following:

[  320.602339] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -3

This is also the default libcamera that comes with Raspbian Bullseye. I also built libcamera apps on an ubuntu 22, and I will try that one to see if it acts any differently.

Best Regards,
C.