raspberrypi/rpicam-apps #367: libcamera-hello Error

E-man-dev, 2022-09-22:
I am using a Picamera v2 on a Pi 3B+, and am getting the following output:
`
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[0:25:27.600453980] [2480]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:25:27.715137256] [2493]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:25:27.717717083] [2493]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:25:27.718354683] [2480]  INFO Camera camera.cpp:848 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx219@10 ***
`
It is running a fresh 32 bit Raspberry Pi OS Full image. I am not using "pi" as the username. A window comes up for a brief moment, but is only a black screen. Picamera2 also fails.

davidplowman, 2022-09-23:
Hi, thanks for the question. First of all, can I assume you're using the latest Bullseye image that was released this month? Can you also say whether you're connected directly to the Pi with a screen and keyboard, or are logging in via ssh with X-forwarding?

1. If you're connected directly to the Pi, can you confirm that you've re-enabled Glamor (use `sudo raspi-config`, it's in the Advanced Options, I think).
2. If you're using ssh and X-forwarding then the GPU accelerated preview can't work, so you would have to add the parameter `--qt-preview` to your command.

E-man-dev, 2022-09-23:
Thanks, that solves that issue! Will this advice be included in the official Picamera2 guide, or will there be no need to go through this step in the future?
That being said, I ran into a different issue. I am getting a `libEGL warning: DRI2: failed to authenticate` message coupled with `ERROR: *** no cameras available ***`. After a little research, it sounds like the libEGL might be due to my 3.2" Waveshare RPi LCD, which uses xserver-xorg. Is my display the cause of my new problem? I'm not well versed in the workings of libcamera.

davidplowman, 2022-09-27:
Hi again, so there is some discussion of this issue in the [Picamera2 manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) already, in section 3.4.

On your second point, there is sometimes a conflict between the camera's I2C and LCD displays, but I believe it should work so long as you're using the kms rather than the fkms display driver (so your /boot/config.txt should contain `dtoverlay=vc4-kms-v3d`).

naushir, 2022-10-07:
Is there anything more to do here?  If not, I will resolve this issue.

E-man-dev, 2022-10-07:
I haven't yet tried the solution proposed, but if it doesn't work then I will just buy a different display. You can close the thread if you want.

closed 2022-10-08 by naushir