raspberrypi/rpicam-apps #370: Compile libcamera-hello Error[BUG]

Robin329, 2022-09-27:
[main] Building folder: libcamera-apps libcamera-hello
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /home/pi/workspace/libcamera-apps/build --config Debug --target libcamera-hello -j 6 --
[build] Scanning dependencies of target images
[build] [ 13%] Built target preview
[build] [ 17%] Building CXX object image/CMakeFiles/images.dir/jpeg.cpp.o
[build] [ 21%] Building CXX object image/CMakeFiles/images.dir/dng.cpp.o
[build] Generating version string: 22a52590c33a-intree 28-09-2022 (00:33:10)
[build] [ 21%] Built target VersionCpp
[build] /home/pi/workspace/libcamera-apps/image/dng.cpp: In function ‘void dng_save(const std::vector<libcamera::Span<unsigned char> >&, const StreamInfo&, const libcamera::ControlList&, const string&, const string&, const StillOptions*)’:
[build] /home/pi/workspace/libcamera-apps/image/dng.cpp:154:6: error: could not convert ‘bl’ from ‘libcamera::Span<const int>’ to ‘bool’
[build]   154 |  if (bl)
[build]       |      ^~
[build]       |      |
[build]       |      libcamera::Span<const int>
[build] compilation terminated due to -Wfatal-errors.
[build] gmake[3]: *** [image/CMakeFiles/images.dir/build.make:134: image/CMakeFiles/images.dir/dng.cpp.o] Error 1
[build] gmake[3]: *** Waiting for unfinished jobs....
[build] In file included from /home/pi/workspace/libcamera-apps/core/options.hpp:21,
[build]                  from /home/pi/workspace/libcamera-apps/core/still_options.hpp:12,
[build]                  from /home/pi/workspace/libcamera-apps/image/jpeg.cpp:23:
[build] /home/pi/workspace/libcamera-apps/image/jpeg.cpp: In function ‘void create_exif_data(const std::vector<libcamera::Span<unsigned char> >&, const StreamInfo&, const libcamera::ControlList&, const string&, const StillOptions*, uint8_t*&, unsigned int&, uint8_t*&, jpeg_mem_len_t&)’:
[build] /home/pi/workspace/libcamera-apps/image/jpeg.cpp:477:32: error: invalid type argument of unary ‘*’ (have ‘int’)
[build]   477 |    LOG(2, "Exposure time: " << *exposure_time);
[build]       |                                ^~~~~~~~~~~~~~
[build] /home/pi/workspace/libcamera-apps/core/logging.hpp:7:17: note: in definition of macro ‘LOG’
[build]     7 |    std::cerr << text << std::endl;                                                                            \
[build]       |                 ^~~~
[build] compilation terminated due to -Wfatal-errors.
[build] gmake[3]: *** [image/CMakeFiles/images.dir/build.make:108: image/CMakeFiles/images.dir/jpeg.cpp.o] Error 1
[build] gmake[2]: *** [CMakeFiles/Makefile2:353: image/CMakeFiles/images.dir/all] Error 2
[build] gmake[1]: *** [CMakeFiles/Makefile2:533: apps/CMakeFiles/libcamera-hello.dir/rule] Error 2
[build] gmake: *** [Makefile:322: libcamera-hello] Error 2
[proc] The command: /usr/bin/cmake --build /home/pi/workspace/libcamera-apps/build --config Debug --target libcamera-hello -j 6 -- exited with code: 2 and signal: null
[build] Build finished with exit code 2

naushir, 2022-09-28:
You have an API incompatibility between libcamera and libcamera-apps.

Please ensure you have rebuilt and install the libcamera libraries from the latest sources.

Robin329, 2022-09-28:
Let me try，I compiled on Raspberry Pi 4 Model B, maybe libcamera is not updated.

Thanks.

naushir, 2022-09-30:
If there is nothing more to be done, I will close this issue now.  Feel free to re-open if you are still having problems.

closed 2022-09-30 by naushir