raspberrypi/rpicam-apps #372: --text-info not displaying in preview window on Rapi OS Bullseye lite

shantigilbert, 2022-09-29:
Hello, first of all let me apologize if I missed something obvious, but I did search all around and nobody seems to either have the same issue or using my same setup. 

I made a very simple but effective timelapse camera setup that uses a Raspberry Pi Zero W and small hdmi monitor to do a preview setup. 

When I was using Buster (with the old camera stack) I could use `--focus` to show a small window with focus information and it worked fine. Now with libcamera, documentation says I can use `--text-info "%focus"` to do something similar, the issue is that since I am using the lite version of Raspi OS with no window manager the preview image has no title bar, thus no focus info or no info at all when using `--text-info "whatever"` is this a limitation or is it something that I am missing? 

Is there a way to maybe force it to be displayed inside the preview window? 

Here is an example when using `--verbose=2`
```
 libcamera-hello --info-text "%focus" --verbose=2
Options:
    verbose: 2
    info_text:%focus
    timeout: 5000
    width: 0
    height: 0
    output:
    post_process_file:
    rawfull: 0
    preview: default
    qt-preview: 0
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
No connector ID specified.  Choosing default from list:
Connector 32 (crtc 89): type 11, 1280x800 (chosen)
Made DRM preview window
Opening camera...
[0:51:50.124743166] [1033]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:51:50.298528667] [1034]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
Acquired camera /base/soc/i2c0mux/i2c@1/imx477@1a
Configuring viewfinder...
Viewfinder size chosen is 2028x1520
Final viewfinder size is 2028x1520
[0:51:50.314258531] [1033]  INFO Camera camera.cpp:1035 configuring streams: (0) 2028x1520-YUV420
[0:51:50.320353478] [1034]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
Camera streams configured
Available controls:
    AeEnable : [false..true]
    ExposureTime : [60..127156999]
    ScalerCrop : [(0, 0)/128x128..(0, 0)/4056x3040]
    AnalogueGain : [1.000000..22.260870]
    AeMeteringMode : [0..3]
    ColourGains : [0.000000..32.000000]
    ExposureValue : [-8.000000..8.000000]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [24994..127157333]
    Brightness : [-1.000000..1.000000]
    AwbMode : [0..7]
    AwbEnable : [false..true]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
    Saturation : [0.000000..32.000000]
    AeConstraintMode : [0..3]
Buffers allocated and mapped
Viewfinder setup complete
Requests created
Camera started!
Viewfinder frame 0
Viewfinder frame 1
...
Viewfinder frame 123
Closing Libcamera application(frames displayed 123, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed

```

Thanks!

naushir, 2022-09-30:
> Now with libcamera, documentation says I can use `--info-text "%focus"` to do something similar, the issue is that since I am using the lite version of Raspi OS with no window manager the preview image has no title bar, thus no focus info or no info at all when using `--info-text "whatever"` is this a limitation or is it something that I am missing?

This is correct.  Since you are using the DRM preview mode when no window manager is present, the info text will not display.  An alternative would be to setup a remote display (with ``ssh -X``) and use the ``--qt-preview`` option.  This will allow the remote display to show the titlebar along with the focus values.

shantigilbert, 2022-09-30:
Ah that is unfortunate, due to my setup I do not have access to SSH the device easily. 

Is adding the information to the DRM preview something that is considered for the future?

naushir, 2022-10-03:
This is not something we plan on adding anytime soon unfortunately.  However, you might be able to use the annotate_cv software postprocessing stage to write info-text directly onto the image, if that is a suitable thing to do.  See [here](https://www.raspberrypi.com/documentation/accessories/camera.html#annotate_cv-stage) for more details.

6by9, 2022-10-03:
For DRM preview the libcamera app is the master, and therefore could add an additional layer to take info text. Doing so shouldn't be too tricky, but will require a modicum of coding. If we already have code that writes the text to a layer for GL to render, then that bit can probably be common.

naushir, 2022-10-03:
> If we already have code that writes the text to a layer for GL to render, then that bit can probably be common.

The text is displayed on the titlebar of the GL window, so we do not have any code to write text to a separate GL layer.

6by9, 2022-10-03:
> The text is displayed on the titlebar of the GL window, so we do not have any code to write text to a separate GL layer.

OK, text rendering isn't beyond the wit of man to achieve.

shantigilbert, 2022-10-03:
> This is not something we plan on adding anytime soon unfortunately. However, you might be able to use the annotate_cv software postprocessing stage to write info-text directly onto the image, if [...]

Thank you, but if I understand correctly, this is not what I need. This will write the text on the images directly  
` This allows other post-processing stages to pass it text strings to be written onto the top of the images.`
What I need is to use it in real time before taking any photo so I can adjust the focus

> This is not something we plan on adding anytime soon unfortunately.

Thank you, I appreciate your time! I guess my only option is to keep using the old stack.

naushir, 2022-10-04:
I'll put DRM preview overlays on the backlog, but it will be some time before I get to it.

closed 2022-10-04 by naushir