raspberrypi/rpicam-apps #373: [PB] Very long delay on stream with libav

tygerlord, 2022-09-30:
Hello,

Sorry, I hope to be on the right place for this question, I use 

>libcamera-vid -t 0 --codec libav --libav-format mpegts --libav-audio -o "tcp://0.0.0.0:8000?listen=1"

on raspberry pi3 B+ 

>Linux doorbell 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

to stream Raspberry pi caméra v1.3 and vlc to play the stream tcp://x.x.x.x:8000

I see the camera images, but they are up to **10 or 12 seconds between the reality and the image**, very annoying to make real time conversation.

Need some help to understand if it's normal or not, please ?
If someone successfully streams audio and video on Raspberry pi with libcamera-vid, I would like known how to do, thank you.

naushir, 2022-10-03:
Can you let us know how you are viewing the stream on the client side?  I would recommend trying ffplay with a command line like:

```
ffplay tcp://<ip-addr-of-server>:<port> -vf "setpts=N/30" -fflags nobuffer -flags low_delay -framedrop
```

This ought to give you minimal latency across the network link.

tygerlord, 2022-10-03:
I use vlc.

I've tried your command with same result, up to 11 seconds between event and result on image player.

I don't know very well how it's working, does the coder need to write file to disk before send, if so like raspberry pi use sd card it will explain the delay.  If not, I'm unable to explain why this take so long.

naushir, 2022-10-04:
An 11 second delay is quite large indeed.  The server does not buffer anything on disk as far as I know, buffering all happens on the client side.

Are you able to try running the client (ffplay) command I mentioned earlier on the same device as the server to eliminate any actual network interference?

tygerlord, 2022-10-05:
Not easy, I use ssh to access my pi and if I run the command to stream on pi I fail on error with libEGL

pi@doorbell:~ $ libEGL warning: DRI2: failed to authenticate

pi@doorbell:~ $ X Error of failed request:  BadRequest (invalid request code or no such operation)
  Major opcode of failed request:  153 ()
  Minor opcode of failed request:  1
  Serial number of failed request:  13
  Current serial number in output stream:  13

[1]+  Exit 1                  libcamera-vid -t 0 --codec libav --libav-format mpegts --libav-audio -o "tcp://0.0.0.0:8000?listen=1"

tygerlord, 2022-10-05:
I've done it, and got a delay a little smaller, up to 9 seconds but with remote screen.

I do **a first ssh** to raspberry pi with command (**no flags -X** I fail in error with it : libEGL warning: DRI2: failed to authenticate):

`libcamera-vid -t 0 --codec libav --libav-format mpegts --libav-audio -o "tcp://0.0.0.0:8000?listen=1"&`

 And a second **ssh -X to raspberry** with command

`ffplay tcp://localhost:8000 -vf "setpts=N/30" -fflags nobuffer -flags low_delay -framedrop`

so I guess that network interference isn't totally eliminated this way ?

naushir, 2022-10-06:
Could you try using the native (i.e. non libav) method of streaming:

```
libcamera-vid -t 0 --inline --listen -o tcp://0.0.0.0:<port>
```

and see if that improves things?

tygerlord, 2022-10-06:
I don't understand anything.
I connect a screen and a keyboard to my raspberrypi and I successfully stream with your commands 
```
libcamera-vid -t 0 -n --inline --listen -o tcp://0.0.0.0:<port> &
ffplay tcp://localhost:8000 -vf "setpts=N/30" -fflags nobuffer -flags low_delay -framedrop
```
But if I try libav in this configuration 
```
libcamera-vid -t 0 -n --codec libav --libav-format mpegts --libav-audio -o "tcp://0.0.0.0:8000?listen=1" &
ffplay tcp://localhost:8000 -vf "setpts=N/30" -fflags nobuffer -flags low_delay -framedrop
```
The up to 10 seconds delay come back and I have no sound, I need to investigate if it's the cause of delay

naushir, 2022-10-07:
Could you try this command:
```
libcamera-vid -t 0 -n --codec libav --libav-format h264 -o "tcp://0.0.0.0:8000?listen=1"
```
and report the latency?  It removes the audio stream from libav.

naushir, 2022-10-10:
I've just tried this again on my setup, I see minimal network latency in my network environment.

Could you run ``camera-bug-report`` and attach the generated ``bug-report.txt`` file so I can have a look and see if there may be any configuration issue with your setup?

naushir, 2022-10-14:
I cannot see anything obviously wrong with your setup.  Are you able to create a fresh 32-bit Raspberry Pi OS image and try the same commands out again - without anything else installed?

naushir, 2022-10-21:
Any updates on this issue? If not, I will close this down shortly.

tygerlord, 2022-10-21:
Sorry, I don't have time to switch to 32-bit image yet. I will do, and since there are no much complains about trouble with it, I have to look for something wrong in my side more probably, so I will close issue for now. Thank you for your time.

naushir, 2022-10-21:
No problem.  Feel free to re-open a new issue if you are still having problems.