raspberrypi/rpicam-apps #374: run libcamera-hello cannot preview camera video[BUG]

Robin329, 2022-10-01:
**Describe the bug**
After running libcamera-hello, the preview window does not appear on the screen.

**Bug report**

`camera-bug-report -t 5 -o bug.txt -c "libcamera-hello -v"`

```
pi@raspberrypi:~/Downloads $ libcamera-hello --version
libcamera-apps build: 22a52590c33a-intree 29-09-2022 (23:33:16)
libcamera build: v0.0.0+3916-a64d17b9-dirty (2022-10-01T13:37:13+08:00)
```

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/9689636/bug.txt)

Robin329, 2022-10-01:
I have run `export DISPLAY=:0` in current shell terminal.

naushir, 2022-10-03:
From the bug-report output, it looks like you have built libcamera-apps and libcamera by yourself:

```
** libcamera-hello --version **
libcamera-apps build: 22a52590c33a-intree 29-09-2022 (23:33:16)
libcamera build: v0.0.0+3916-a64d17b9-dirty (2022-10-01T13:37:13+08:00)
```

Did you ensure that you installed all the display dependencies (e.g. egl, libepoxy, drm) before running your build?  If not, the preview display will not be built.  The cmake command output will tell you if the EGL display is enabled.  You can find further instructions on the build and installing dependencies [here](uhttps://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-and-libcamera-apps).

Robin329, 2022-10-04:
Thank you.
Install some dependent libraries solved.

closed 2022-10-04 by Robin329