raspberrypi/rpicam-apps #375: [BUG] libcamera-jpeg prints out of memory error on Raspberry Pi Zero W

tewarid, 2022-10-01:
**Describe the bug**
libcamera-jpeg prints out of memory error on Raspberry Pi Zero W at full resolution jpeg capture with an imx219. Lowering the resolution (e.g. width 640 and height 480) works without error.

**Bug report**
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 30-09-2022 (20:00:54)
Command: /usr/bin/camera-bug-report -c libcamera-jpeg -o test.jpg

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi0-wifi

** cat /proc/cpuinfo **
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000002fe9032e
Model		: Raspberry Pi Zero W Rev 1.1

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait cma=64M

** cat /boot/config.txt **
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
enable_uart=1
dtoverlay=vc4-kms-v3d
dtoverlay=imx219

dtoverlay=seeed-2mic-voicecard

dtoverlay=waveshare35a
dtoverlay=dwc2,dr_mode=otg

--------------------------------------------------------------------------------
Logs
--------------------------------------------------------------------------------
** dmesg **
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.56 (oe-user@oe-host) (arm-poky-linux-musleabi-gcc (GCC) 12.2.0, GNU ld (GNU Binutils) 2.39.0.20220819) #1 Fri Jul 22 13:23:23 UTC 2022
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: bypass linux,cma node, using cmdline CMA params instead
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    0.000000] cma: Reserved 64 MiB at 0x17c00000
[    0.000000] Zone ranges:
[    0.000000] Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000] node   0: [mem 0x0000000000000000-0x000000001bffffff]
[... 723 lines cut ...]
VERSION="4.1+snapshot-e2ead243cdedfd34793e3a40b9af0116a7b0b0cd (langdale)"
VERSION_ID=4.1-snapshot-e2ead243cdedfd34793e3a40b9af0116a7b0b0cd
PRETTY_NAME="Poky (Yocto Project Reference Distro) 4.1+snapshot-e2ead243cdedfd34793e3a40b9af0116a7b0b0cd (langdale)"
DISTRO_CODENAME="langdale"

** vcgencmd version **
/bin/sh: line 1: vcgencmd: command not found

** libcamera-jpeg --version **
libcamera-apps build: 22a52590c33a-intree-dirty 29-09-2022 (17:20:32)
libcamera build: v0.0.0+3949-ed591e70-dirty (2022-09-29T17:20:12+00:00)

--------------------------------------------------------------------------------
libcamera-jpeg -o test.jpg
--------------------------------------------------------------------------------
Preview window unavailable
[0:06:13.395242840] [380]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3949-ed591e70-dirty (2022-09-29T17:20:12+00:00)
[0:06:14.117302242] [381]  WARN RPI raspberrypi.cpp:1263 Mismatch between Unicam and CamHelper for embedded data usage!
[0:06:14.145219103] [381]  INFO RPI raspberrypi.cpp:1379 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[0:06:14.150357077] [380]  INFO Camera camera.cpp:1026 configuring streams: (0) 1640x1232-YUV420
[0:06:14.154018059] [381]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:06:19.432056443] [380]  INFO Camera camera.cpp:1026 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-SBGGR10_CSI2P
[0:06:19.464309303] [381]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[0:06:19.770165972] [381] ERROR V4L2 v4l2_videodevice.cpp:1241 /dev/video14[14:cap]: Unable to request 1 buffers: Out of memory
[0:06:19.770520970] [381] ERROR RPI raspberrypi.cpp:1030 Failed to allocate buffers

naushir, 2022-10-03:
You have manually set a CMA of 64MB:
```
**cat /proc/meminfo**
...
CmaTotal: 65536 kB
CmaFree: 57980 kB
```

```
** cat /boot/cmdline.txt **
dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait cma=64M
```

Please remove this and revert to the system default.  That ought to fix your out of memory issues.

tewarid, 2022-10-03:
Thanks - that did it!

closed 2022-10-03 by tewarid