raspberrypi/rpicam-apps #376: [BUG]  Libcamera-vid Only 1st segment playable?

Mattncb, 2022-10-01:
I'm having a strange problem with segmented videos

libcamera-vid -t 30000 --segment 10000 -o video%04d.h264

Run the command without error and successfully creates 3 files all of similar file size. Unfortunately, only the first file is playable and the following 2 seem to be corrupted.

Can anyone offer any suggestions on how to resolve?

Linux raspberrypi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux
libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:16:34)
libcamera build: v0.0.0+3866-0c55e522

Pi Zero W 2 
HQ Camera Module

naushir, 2022-10-03:
What are you using to play back the files?  Could you attach one good and one bad file for us to have a look at please?

Additionally, you might want to use ``--inline`` option when using segments to ensure frame headers are written appropriately.

6by9, 2022-10-03:
Raspivid automatically enabled inline headers if segmentation was selected. Does libcamera-vid not do the same?
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVid.c#L757

naushir, 2022-10-03:
No it does not, but probably makes sense if it does...

Mattncb, 2022-10-03:
Inline solved it thanks. Agree with the above or maybe at least documentation on segment should be amended to show inline is needed.

thanks!

naushir, 2022-10-04:
Looking at the code here:
https://github.com/raspberrypi/libcamera-apps/blob/22a52590c33a813743b4e6337478c208201c77b1/core/video_options.hpp#L134
you should have seen the error message "WARNING: consider inline headers with 'pause'/split/segment/circular" printed out.

naushir, 2022-10-07:
Resolving...

closed 2022-10-07 by naushir