raspberrypi/rpicam-apps #377: [BUG]Unable to run libcamera-hello

shaja-asm, 2022-10-03:
Whenever I try to run lbcamera-hello I get the following error.

```
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[1:30:13.893349420] [6492]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3949-ed591e70-dirty (2022-10-03T10:13:27+05:30)
[1:30:14.114115059] [6508]  WARN RPI raspberrypi.cpp:1263 Mismatch between Unicam and CamHelper for embedded data usage!
[1:30:14.120827671] [6508]  INFO RPI raspberrypi.cpp:1379 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[1:30:14.125738785] [6492]  INFO Camera camera.cpp:1026 configuring streams: (0) 1640x1232-YUV420
[1:30:14.128591527] [6508]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 21
Aborted

```
Here is the bug report for the 
```
camera-bug-report -t 5 -o bug.txt -c "libcamera-hello -v"
```

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/9695377/bug.txt)

naushir, 2022-10-03:
Are you running this over a remote ssh session?  If so, you will need to forward the display (using ``ssh -X``) as well as use the qt preview (``--qt-preview``) window to allow a remote preview.

shaja-asm, 2022-10-03:
No. I'm not running over ssh. I have a separate monitor for the Pi.

naushir, 2022-10-03:
Are you running libcamera-hello through a windowing manager?  If so can you try adding ``--qt-preview`` to your command line and see if that works?

Just to confirm, are you using a custom build of ``libcamera-apps``?

shaja-asm, 2022-10-04:
- I'm able to capture camera output using qcam app `sudo qcam`

- I'm unable to run any of the basic libcamera functions. Here is the camera bug report for the `camera-bug-report -t 5 -o bug.txt -c "libcamera-hello -v"
[bug.txt](https://github.com/raspberrypi/picamera2/files/9695594/bug.txt)

- `libcamera-hello -t 0 --qt-preview` gives a preview window. But `libcamera-hello -t 0` gives the following error
```libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[2:14:44.519406501] [7770]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3949-ed591e70-dirty (2022-10-03T10:13:27+05:30)
[2:14:44.738688278] [7785]  WARN RPI raspberrypi.cpp:1263 Mismatch between Unicam and CamHelper for embedded data usage!
[2:14:44.745319937] [7785]  INFO RPI raspberrypi.cpp:1379 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[2:14:44.750219008] [7770]  INFO Camera camera.cpp:1026 configuring streams: (0) 1640x1232-YUV420
[2:14:44.753027108] [7785]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 21
Aborted

```

naushir, 2022-10-04:
If you built libcamera-apps yourself, and the EGL preview is unavailable, this likely means you are missing some library dependencies so EGL preview is disabled.  The ``cmake`` command output will tell you if anything related to EGL display is missing or not.  You can find further instructions on what dependencies are needed [here](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-and-libcamera-apps).

To start with, you might want to revert to the apt version of libcamera-apps and confirm EGL display still works for you.

naushir, 2022-10-07:
Any update on this issue?  If not, I will resolve it.

naushir, 2022-10-10:
Resolving due to inactivity.  If you are still having problems, feel free to reopen.

closed 2022-10-10 by naushir