raspberrypi/rpicam-apps #378: [BUG] Cannot output to STDOUT when using libav codec with libcamera-vid

pakal, 2022-10-03:
When using libcamera-vid with output "-", it normally prints binary data to the console:

`libcamera-vid  --nopreview  -o -`

However when using libav codec, instead, it writes data to a file named "-"

`libcamera-vid  --nopreview --codec libav --libav-format mp4 -o -`

A similar problem has been solved for images (https://github.com/raspberrypi/libcamera-apps/issues/139), but this bug here seems quite different

naushir, 2022-10-04:
stdout from libav is not currently supported in the software.  However, it ought to be easy enough to add in.  I will look to do this when I get a chance.

naushir, 2022-10-04:
As a workaround you can use ``-o pipe:`` to pipe to stdout.  Note that you cannot pipe to stdout with ``--libav-format mp4``.  This is probably because the mp4 format needs to seek back to the start of the stream to write headers and other things.  Instead, you can use ``--libav-format h264`` or ``--libav-format mpegts``.

naushir, 2022-10-04:
https://github.com/raspberrypi/libcamera-apps/pull/380 ought to fix this.

closed 2022-10-05 by naushir

pakal, 2022-10-06:
Thanks a lot, "pipe:" works indeed fine with mpegts, in the meantime   :)